theyosh/TerrariumPI #168: Offline Issue

TuxujPes, 2018-07-06:
Hi,

I am always offline.
Have tried to update, restart, clean db but nothing helps.

What could be the issue?

<details><summary>Log</summary>

```bash
2018-07-06 20:43:00,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.1.106:8090 ...
2018-07-06 20:43:26,514 - WARNING - terrariumSensor      - Measured value None% from humidity sensor 'hum1' is outside valid range 0.00% - 100.00% in 14.74961 seconds.
2018-07-06 20:43:41,253 - WARNING - terrariumSensor      - Measured value NoneC from temperature sensor 'temp1' is outside valid range 0.00C - 100.00C in 14.73582 seconds.
2018-07-06 20:43:41,000 - INFO    - terrariumWebserver   - Running webserver at :::8090
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 862, in _handle
    return route.call(**args)
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 1740, in wrapper
    rv = callback(*a, **ka)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 43, in webserver_headers
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/bottle_websocket/plugin.py", line 5, in wrapper
    callback(request.environ.get('wsgi.websocket'), *args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 435, in handle_websocket
    terrariumWebserver.app.terrarium.get_power_usage_water_flow(socket=True)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 914, in get_power_usage_water_flow
    totaldata = self.__get_total_power_usage_water_flow()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 376, in __get_total_power_usage_water_flow
    totals = self.collector.get_total_power_water_usage()
  File "/home/pi/TerrariumPI/terrariumCollector.py", line 292, in get_total_power_water_usage
    totals = {'power_wattage' : {'duration' : int(row['TotalTime']) , 'wattage' : float(row['Watt'])},
TypeError: int() argument must be a string or a number, not 'NoneType'
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 862, in _handle
    return route.call(**args)
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 1740, in wrapper
    rv = callback(*a, **ka)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 43, in webserver_headers
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/bottle_websocket/plugin.py", line 5, in wrapper
    callback(request.environ.get('wsgi.websocket'), *args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 435, in handle_websocket
    terrariumWebserver.app.terrarium.get_power_usage_water_flow(socket=True)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 914, in get_power_usage_water_flow
    totaldata = self.__get_total_power_usage_water_flow()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 376, in __get_total_power_usage_water_flow
    totals = self.collector.get_total_power_water_usage()
  File "/home/pi/TerrariumPI/terrariumCollector.py", line 292, in get_total_power_water_usage
    totals = {'power_wattage' : {'duration' : int(row['TotalTime']) , 'wattage' : float(row['Watt'])},
TypeError: int() argument must be a string or a number, not 'NoneType'
2018-07-06 20:43:56,150 - WARNING - terrariumSensor      - Measured value None% from humidity sensor 'hum1' is outside valid range 0.00% - 100.00% in 14.73936 seconds.
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 862, in _handle
    return route.call(**args)
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 1740, in wrapper
    rv = callback(*a, **ka)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 43, in webserver_headers
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/bottle_websocket/plugin.py", line 5, in wrapper
    callback(request.environ.get('wsgi.websocket'), *args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 435, in handle_websocket
    terrariumWebserver.app.terrarium.get_power_usage_water_flow(socket=True)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 914, in get_power_usage_water_flow
    totaldata = self.__get_total_power_usage_water_flow()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 376, in __get_total_power_usage_water_flow
    totals = self.collector.get_total_power_water_usage()
  File "/home/pi/TerrariumPI/terrariumCollector.py", line 292, in get_total_power_water_usage
    totals = {'power_wattage' : {'duration' : int(row['TotalTime']) , 'wattage' : float(row['Watt'])},
TypeError: int() argument must be a string or a number, not 'NoneType'
2018-07-06 20:44:11,008 - WARNING - terrariumSensor      - Measured value NoneC from temperature sensor 'temp1' is outside valid range 0.00C - 40.00C in 14.73542 seconds.
int() argument must be a string or a number, not 'NoneType'
Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 716, in gevent._greenlet.Greenlet.run
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 442, in __engine_loop
    lcd_message = ['%s %s' % (_('Uptime'),terrariumUtils.format_uptime(system_data['uptime']),),
UnboundLocalError: local variable 'system_data' referenced before assignment
2018-07-06T17:44:11Z <Greenlet "Greenlet-0" at 0x6bdac1c8: <bound method terrariumEngine.__engine_loop of <terrariumEngine.terrariumEngine object at 0x76a349b0>>> failed with UnboundLocalError

Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 862, in _handle
    return route.call(**args)
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 1740, in wrapper
    rv = callback(*a, **ka)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 43, in webserver_headers
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/bottle_websocket/plugin.py", line 5, in wrapper
    callback(request.environ.get('wsgi.websocket'), *args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 435, in handle_websocket
    terrariumWebserver.app.terrarium.get_power_usage_water_flow(socket=True)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 914, in get_power_usage_water_flow
    totaldata = self.__get_total_power_usage_water_flow()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 376, in __get_total_power_usage_water_flow
    totals = self.collector.get_total_power_water_usage()
  File "/home/pi/TerrariumPI/terrariumCollector.py", line 292, in get_total_power_water_usage
    totals = {'power_wattage' : {'duration' : int(row['TotalTime']) , 'wattage' : float(row['Watt'])},
TypeError: int() argument must be a string or a number, not 'NoneType'
```
</details>

kaju666, 2018-07-06:
Me too from today.

theyosh, 2018-07-06:
Damm, that is not good. I am currently investigating it, and with a clean system, I do have this issue now also. So I will try to fix this asap. Hope withing 1 a 2 hours.

theyosh, 2018-07-06:
My fault. Did not test the new collector code with an empty database. So it is fixed now.

Do a `git pull` and restart. Should all work again.

theyosh, 2018-07-06:
Closing because problem solved and confirmed by @kaju666

closed 2018-07-06 by theyosh

TuxujPes, 2018-07-07:
Yeah, got fixed. Thanks!