theyosh/TerrariumPI #169: Little bug with weather location setting

kaju666, 2018-07-06:
Hello again :)

Default location is:
https://www.yr.no/place/Netherlands/Groningen/Groningen
And is working fine when saving.
But going to yr.no and having own location link:
https://www.yr.no/sted/Polen/Masovia/Warszawa
And "data could not be saved"
https://openweathermap.org/city/756135
And also "data could not be saved"
https://weather.com/weather/today/l/PLXX0028:1:PL
And also "data could not be saved"

Only works when i edit by hand default location to look like this:
https://www.yr.no/place/Polen/Masovia/Warszawa

theyosh, 2018-07-06:
Hi :)

This is also missing some documentation. But if you open the help section, it does shows the valid url formats. And yes, for yr.no I need English urls. See also: #154 

So the other formats are described in the help section which is by default closed (right top down carrot)

Dave0408, 2020-07-09:
Hello, 

after the support for yr.no was dropped [#417] i wanted to switch to openweathermap. 
I get the error message: "Error: Data could not be saved"
In the GUI in Help -> Usage -> Weather -> Settings I can't find any helpful documentation about the URL format. 
The log also remains empty.
The url i tried was: https://openweathermap.org/city/3393008

Where can I find more information?
Or is your recommandation using http://api.wunderground.com/ ? 
I didn't try that due to need of an API key. 

Thanks a lot.

theyosh, 2020-07-09:
Hi, added some more info at #417 I hope this will help you out