theyosh/TerrariumPI #170: Sending commands via Telegram

kaju666, 2018-07-07:
Hi

I have a system called Octoprint on another raspberry. It's for controling 3d printer. It also have very nice plugin which can send me notifications about print status in telegram messeger also using Botfather like in your TerrariumPi. But i can also write a message to that bot, for example /status. 
Then this bot is sending me actual info about my printer and a webcam image of actual print.

It will be very nice feature if i can also write here in your script to a bot and get actual information about envoirment and a picture form webcam in terrarium :). Or even send a command to change some variables :P 

Just a idea for you if you have too much free time :P

theyosh, 2018-07-07:
Yeah, I have had this discussion before, but short answer: I have a webinterface for this. And I do not need Telegram for it.

But you are not the first, so look at this: https://github.com/theyosh/TerrariumPI/tree/master/contrib/telegram_bot

That is doing what you want. But it is not created by me, so I will not update that part of the code. Do with it what you want.

BashSer, 2018-07-08:
Yeees, this beautiful project has many interesting features and one of them is API !

For example, I do toggle switch in bot so:
`requests.post(url = str("http://localhost:8090/api/switch/toggle/"+dict_switches_id.get("Switch1")), auth=(login_user_Terrarium_Pi, login_password_Terrarium_Pi))`
You can see this on link above from @theyosh.
Just insert your username(string 21), token (string 24), user/pass for web-interface (strings 26,27) and install missing packages from "import" (requests, import).
I running them over ssh, just type `nohup python bot_menu_api.py &` and they working in background with logging.

This bot can send alarm, info about sensors, system, toggle switches, send photos and etc.
I newbie in python and dont end this, but you can try use this ;)
Before in my country block telegram i use this everyday :D