theyosh/TerrariumPI #171: Switch, EG-PM-2 doesnt work

dasolan, 2018-07-11:
Since the notification update, the timer wasn't working correctly. 
Since update to 3.8.1, the switch panel wasn't displayed anymore, after a fresh install, settings couldn't be saved :(
The circle is always spinning.

theyosh, 2018-07-11:
This sounds like a browser caching issue. Have you cleared browser cache?

theyosh, 2018-07-13:
Any updates? Because I cannot reproduce

dasolan, 2018-07-15:
The settings thing was gone with clear browser cache. 
But the timer isn't working, a light should be turned on at 2 PM, now its 2:17 and nothing has happend. 
If I do it manualy, the switch is turned off again. 

When I'm submit the settings page with timer settings again, its turned on :/ 

Do you need a log sth?

theyosh, 2018-07-17:
This sounds like you have configured the same power switch somewhere in the environment settings. If this is the case, remove the power switch from the environment part.

I have tested it now couple of days, and it all works like expected.

You could add the logfile here and I can have a look at it.

theyosh, 2018-07-20:
Any updates? As stated before, I cannot reproduce and for me it is working

dasolan, 2018-07-21:
Same even with 3.8.2, the "shutdown" works perfect but startup is hangig nearly every time. 
Two terrariums are depending, one starts at 2 PM, the 2nd at 8 PM. 
The first is hangig, my workaround is to submit settings page once again with exact same settings. 
2 PM starts, 2nd too, everthing shuts down in time. 
Its only the startup :/

dasolan, 2018-07-21:
``

2018-07-21 16:16:18,333 - INFO    - terrariumEnvironment - Update done in 0.00183 seconds. Waiting for 14.99817 seconds for next update
2018-07-21 16:16:17,088 - INFO    - terrariumEngine      - Update done in 0.47584 seconds. Waiting for 29.52416 seconds for next update
2018-07-21 16:16:16,960 - INFO    - terrariumSwitch      - Timer action is done for switch LED. State not changed.
2018-07-21 16:16:16,858 - INFO    - terrariumSwitch      - Timer action is done for switch FarRed. State not changed.
2018-07-21 16:16:16,756 - INFO    - terrariumSwitch      - Timer action is done for switch LED. State not changed.
2018-07-21 16:16:16,654 - INFO    - terrariumSwitch      - Timer action is done for switch LED1. State not changed.
2018-07-21 16:16:03,331 - INFO    - terrariumEnvironment - Update done in 0.00181 seconds. Waiting for 14.99819 seconds for next update
2018-07-21 16:15:48,315 - INFO    - terrariumEnvironment - Update done in 0.00178 seconds. Waiting for 14.99822 seconds for next update
2018-07-21 16:15:47,217 - INFO    - terrariumEngine      - Update done in 0.61762 seconds. Waiting for 29.38238 seconds for next update
2018-07-21 16:15:47,097 - INFO    - terrariumSwitch      - Timer action is done for switch LED. Switched to state on.
2018-07-21 16:15:46,957 - INFO    - terrariumSwitch      - Toggle switch 'LED' from off to on
2018-07-21 16:15:46,831 - INFO    - terrariumSwitch      - Timer action is done for switch FarRed. State not changed.
2018-07-21 16:15:46,730 - INFO    - terrariumSwitch      - Timer action is done for switch LED. State not changed.
2018-07-21 16:15:47,097 - INFO    - terrariumSwitch      - Timer action is done for switch LED. Switched to state on.
2018-07-21 16:15:46,957 - INFO    - terrariumSwitch      - Toggle switch 'LED' from off to on
2018-07-21 16:15:46,831 - INFO    - terrariumSwitch      - Timer action is done for switch FarRed. State not changed.
2018-07-21 16:15:46,730 - INFO    - terrariumSwitch      - Timer action is done for switch LED. State not changed.
2018-07-21 16:15:46,628 - INFO    - terrariumSwitch      - Timer action is done for switch LED1. State not changed.
2018-07-21 16:15:39,005 - INFO    - terrariumEngine      - Done reloading terrariumPI switches. Found 4 switches in 0.047 seconds
2018-07-21 16:15:39,002 - INFO    - terrariumSwitch      - Timer time table loaded for switch 'LED1' with 1 entries.
2018-07-21 16:15:39,001 - INFO    - terrariumSwitch      - Calculating timer 'LED1' with timer data: enabled = True, start = 20:00, stop = 08:00, on duration = 720, off duration = 720
2018-07-21 16:15:39,000 - INFO    - terrariumSwitch      - Timer time table loaded for switch 'LED1' with 1 entries.
2018-07-21 16:15:38,999 - INFO    - terrariumSwitch      - Calculating timer 'LED1' with timer data: enabled = True, start = 20:00, stop = 08:00, on duration = 720, off duration = 720
2018-07-21 16:15:38,998 - INFO    - terrariumSwitch      - Timer time table loaded for switch 'LED1' with 1 entries.
2018-07-21 16:15:38,996 - INFO    - terrariumSwitch      - Calculating timer 'LED1' with timer data: enabled = True, start = 20:00, stop = 08:00, on duration = 720, off duration = 720
2018-07-21 16:15:38,996 - INFO    - terrariumSwitch      - Timer time table loaded for switch 'LED1' with 1 entries.
2018-07-21 16:15:38,994 - INFO    - terrariumSwitch      - Calculating timer 'LED1' with timer data: enabled = True, start = 20:00, stop = 08:00, on duration = 720, off duration = 720
2018-07-21 16:15:38,993 - INFO    - terrariumSwitch      - Timer time table loaded for switch 'LED1' with 1 entries.
2018-07-21 16:15:38,992 - INFO    - terrariumSwitch      - Calculating timer 'LED1' with timer data: enabled = True, start = 20:00, stop = 08:00, on duration = 720, off duration = 720
2018-07-21 16:15:38,991 - INFO    - terrariumSwitch      - Timer time table loaded for switch 'LED' with 1 entries.
2018-07-21 16:15:38,990 - INFO    - terrariumSwitch      - Calculating timer 'LED' with timer data: enabled = True, start = 14:00, stop = 08:00, on duration = 1080, off duration = 360
2018-07-21 16:15:38,989 - INFO    - terrariumSwitch      - Timer time table loaded for switch 'LED' with 1 entries.
2018-07-21 16:15:38,988 - INFO    - terrariumSwitch      - Calculating timer 'LED' with timer data: enabled = True, start = 14:00, stop = 08:00, on duration = 1080, off duration = 360
2018-07-21 16:15:38,987 - INFO    - terrariumSwitch      - Timer time table loaded for switch 'LED' with 1 entries.
2018-07-21 16:15:38,985 - INFO    - terrariumSwitch      - Calculating timer 'LED' with timer data: enabled = True, start = 14:00, stop = 08:00, on duration = 1080, off duration = 360
2018-07-21 16:15:38,985 - INFO    - terrariumSwitch      - Timer time table loaded for switch 'LED' with 1 entries.
2018-07-21 16:15:38,983 - INFO    - terrariumSwitch      - Calculating timer 'LED' with timer data: enabled = True, start = 14:00, stop = 08:00, on duration = 1080, off duration = 360
2018-07-21 16:15:38,982 - INFO    - terrariumSwitch      - Timer time table loaded for switch 'LED' with 1 entries.
2018-07-21 16:15:38,981 - INFO    - terrariumSwitch      - Calculating timer 'LED' with timer data: enabled = True, start = 14:00, stop = 08:00, on duration = 1080, off duration = 360
2018-07-21 16:15:38,980 - INFO    - terrariumSwitch      - Timer time table loaded for switch 'LED' with 1 entries.
[... cut ...]
2018-07-21 16:09:16,383 - INFO    - terrariumSwitch      - Timer action is done for switch LED1. State not changed.
2018-07-21 16:09:03,134 - INFO    - terrariumEnvironment - Update done in 0.00180 seconds. Waiting for 14.99820 seconds for next update
2018-07-21 16:08:48,118 - INFO    - terrariumEnvironment - Update done in 0.00179 seconds. Waiting for 14.99821 seconds for next update
2018-07-21 16:08:46,813 - INFO    - terrariumEngine      - Update done in 0.47730 seconds. Waiting for 29.52270 seconds for next update``

theyosh, 2018-07-22:
Not sure if you just played with the timer values but that could be part of the problem.

I also found a code issue, so you should also do a `git pull` to get the latest code which should make your power switch fully working as expected.

theyosh, 2018-07-24:
Have you tried the latest code already? I think your issue is fixed

theyosh, 2018-07-27:
Closing due to code has been fixed and tested and no response for 6 days

closed 2018-07-27 by theyosh