theyosh/TerrariumPI #172: ERROR Getting online data - terrariumWeather

TwoCH, 2018-07-11:
Hello theyosh,

I keep getting errors from weather.
I tested with openweathermap.org and yr.no (for Groningen and Madagaskar)
When I enter a different URL it says Data saved, log see below.
```
2018-07-11 20:30:25,783 - INFO    - terrariumWeather     - Done refreshing weather data in 0.16859 seconds. Next refresh after: 2018-07-12 00:30:25
2018-07-11 20:30:25,782 - INFO    - terrariumWeather     - Done loading data in 0.166903018951 seconds. Hour forcecast items: 0, week forecast items: 0.
2018-07-11 20:30:25,780 - ERROR   - terrariumWeather     - Error getting online data from yr.no
2018-07-11 20:30:25,616 - INFO    - terrariumWeather     - Update YR.no data from ONLINE refreshing cache.
2018-07-11 20:30:25,614 - INFO    - terrariumWeather     - Refreshing 'yr.no' weather data from source 'https://www.yr.no/place/Madagascar/Diana/Ambilobe/'
2018-07-11 20:30:25,610 - INFO    - terrariumWeather     - Set weather to type 'yr.no' based on source to 'https://www.yr.no/place/Madagascar/Diana/Ambilobe/'
```

theyosh, 2018-07-11:
This is probably a hickup. Just loaded: https://www.yr.no/place/Madagascar/Diana/Ambilobe/ without any problems in my test setup.

Restarting the Pi could help. It is able to get data from online? And is the Pi running in the correct timezone?

TwoCH, 2018-07-11:
TZ was correct, was tested directly after fresh install.
However, restarting as shown in wiki solved it, weather data is now shown correct.
Thank you 👍

closed 2018-07-11 by TwoCH