theyosh/TerrariumPI #173: Error: database disk image is malformed

TuxujPes, 2018-07-19:
Not sure what has caused an issue, maybe some of my local changes.

```
TerrariumPI Collecter exception! (DatabaseError('database disk image is malformed',),)
```

theyosh, 2018-07-20:
This is an error that happens when the historical data can not be written or red correctly to the sqlite database. In the past this has happened a lot at my system, but for the last couple of months, I did not have this issue anymore. It could happen that the SD card has problems. Does `dmesg` show sd card errors?
Or was there a power failure which shutdown the Pi without proper shutdown?

If you did not do anything, it should already be fixed. The code is capable of detecting this and should start a recovery action. This recovery will dump the sqlite database to a plain file entry for entry. And the entries that are corrupt will be ignored. Mostly it is only one record.

After dumping, the sql is read in again and used as new database. So look in the terrariumlog file if there are messages about recovery. That should give you information if the database is recovered correctly.

theyosh, 2018-07-20:
There should be a file called `.recovery.sql` file that should grow. `ls -la` should show it

TuxujPes, 2018-07-21:
The file is in there and grows.

And app seems to work now (at least server is running) but very slowly.
I won't lose much by checking out from scratch.

Can I simply hard reset repository it and then install anew?

Offtopic: If I want to turn ON one pin do I go to Environments? (E.g. if Humidity > 95% then turn on GPIO 36)

theyosh, 2018-07-21:
You can just remove the history.db file following these steps:
1. Stop the software: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-stop-terrariumpi
2. Remove history.db file `rm history.db`
3. Start the software: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-start-terrariumpi
This should create a new database for historical data

For your humidity question:
1. Create a humidity sensor with the wanted alarm high and alarm low values. In this case a alarm max of 95
2. Create a power switch that operates through the GPIO pin. Make sure it is working by clicking the power switch in the status page.
3. Go the environment, and select `sensor mode`, choice the created Humidity sensor at the senors part, and then open the alarm high option to select the created power switch.
4. Save it all

This will trigger the environment part to power on the power switch at GPIO 36 when the humidity gets higher then 95%

TuxujPes, 2018-07-21:
@theyosh steps helped to restart stuff!
At least app is responsive now and I can see both temp and humidity on the charts.

`screen -r` though, logs some odd messages:
```
TerrariumPI Collecter exception! (OperationalError('attempt to write a readonly database',),)
```

I don't think I've seen this before

theyosh, 2018-07-21:
Euh, this is new. I get the feeling that there is an SD card dying somehow... I will never put the database in readonly.

Maybe to be sure use a new SD card?

TuxujPes, 2018-07-21:
The SD card is new.  There probably would not be ways for me to kill it in 2 weeks, would it?

Could it be related with me `sudo ./start.sh` and it creating `history.db` ?

TuxujPes, 2018-07-21:
@theyosh I've redone db deletion steps and everything seems to work now as expected!
I think, previously, instead of shutting down, I restarted app with only on ctrl+c, thus could be a problem with access.

closed 2018-07-23 by TuxujPes