theyosh/TerrariumPI #175: Startup error

Natrixz, 2018-07-26:
Traceback (most recent call last):
  File "/home/user/TerrariumPI/terrariumPI.py", line 17, in <module>
    terrariumEngine = terrariumEngine()
  File "/home/user/TerrariumPI/terrariumEngine.py", line 115, in __init__
    self.__load_sensors()
  File "/home/user/TerrariumPI/terrariumEngine.py", line 156, in __load_sensors
    for sensor in terrariumSensor.scan(self.config.get_owfs_port(), self.__unit_type):
  File "/home/user/TerrariumPI/terrariumSensor.py", line 290, in scan
    for (sensor,sensortype) in terrariumMiFloraSensor.scan():
  File "/home/user/TerrariumPI/terrariumBluetoothSensor.py", line 160, in scan
    for device in Scanner().scan(SCANTIME):
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 734, in scan
    self.start(passive=passive)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 672, in start
    self._mgmtCmd("le on")
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 283, in _mgmtCmd
    "Failed to execute mgmt cmd '%s'" % (cmd))
bluepy.btle.BTLEException: Failed to execute mgmt cmd 'le on'

This is since the newest update

theyosh, 2018-07-26:
Euh, did you run the installer by any change? If not, could you try it? I think I have to update the version number to trigger the installer. That is needed for the new bluetooth devices.

Sorry for not taking care of this.

Natrixz, 2018-07-26:
Yes I ran the installer, I could just look up what package it needs and manually install it

Natrixz, 2018-07-26:
Nope, doesn't fix it :(

theyosh, 2018-07-26:
Ok, this is strange. For me it is working. To manually test it you can do the following:

1. So first you need to check if you have the problem as sudo-user
    `sudo /usr/local/lib/python2.7/dist-packages/bluepy/bluepy-helper 0`
    `le on`
2. If it works with sudo, and you want to be able to run it as a normal user, you need to set capabilities on the executable. try:
    `sudo setcap 'cap_net_raw,cap_net_admin+eip' /usr/local/lib/python2.7/dist-packages/bluepy/bluepy-helper`
    I don't know exactly what this does, but it allows me to run bluepy-helper without root privileges (sudo)
3. Now you can run it as a normal user with
    `/usr/local/lib/python2.7/dist-packages/bluepy/bluepy-helper 0`
    `le on`

It should produce the output: rsp=$mgmt code=$success

And with the following command you can check the settings:

`getcap /usr/local/lib/python2.7/dist-packages/bluepy/bluepy-helper`

Should produce:
`/usr/local/lib/python2.7/dist-packages/bluepy/bluepy-helper = cap_net_admin,cap_net_raw+eip`

Natrixz, 2018-07-26:
le on
rsp=$mgmt code=$protoerr

Well that's not good....lol

theyosh, 2018-07-26:
When did you get this output? As normal user? Could you try a reboot then?

Natrixz, 2018-07-26:
As both root, and as sudo
Rebooted a few times ;)

theyosh, 2018-07-26:
Euh, I need to now if it works for normal user without sudo. If not, then the rights are not correct.

what is the output of `getcap /usr/local/lib/python2.7/dist-packages/bluepy/bluepy-helper`

Natrixz, 2018-07-26:
/usr/local/lib/python2.7/dist-packages/bluepy/bluepy-helper = cap_net_admin,cap_net_raw+eip

theyosh, 2018-07-26:
do you use the user pi or do you have a different user. If you have your own user, could your post the output of /etc/group ? Maybe that the user PI is in more groups then a custom user. Else I do not know why...

Natrixz, 2018-07-26:
I use a different user :)

root:x:0:
daemon:x:1:
bin:x:2:
sys:x:3:
adm:x:4:pi,natrixz
tty:x:5:
disk:x:6:
lp:x:7:
mail:x:8:
news:x:9:
uucp:x:10:
man:x:12:
proxy:x:13:
kmem:x:15:
dialout:x:20:pi,natrixz
fax:x:21:
voice:x:22:
cdrom:x:24:pi,natrixz
floppy:x:25:
tape:x:26:
sudo:x:27:pi,natrixz
audio:x:29:pi,natrixz
dip:x:30:
www-data:x:33:
backup:x:34:
operator:x:37:
list:x:38:
irc:x:39:
src:x:40:
gnats:x:41:
shadow:x:42:
utmp:x:43:
video:x:44:pi,natrixz
sasl:x:45:
plugdev:x:46:pi,natrixz
staff:x:50:
games:x:60:pi,natrixz
users:x:100:pi,natrixz
nogroup:x:65534:
input:x:101:pi,natrixz
systemd-journal:x:102:
systemd-timesync:x:103:
systemd-network:x:104:
systemd-resolve:x:105:
systemd-bus-proxy:x:106:
crontab:x:107:
netdev:x:108:pi,natrixz
pi:x:1000:
messagebus:x:109:
ssh:x:110:
bluetooth:x:111:
avahi:x:112:
spi:x:999:pi,natrixz
i2c:x:998:pi,natrixz
gpio:x:997:pi,natrixz
natrixz:x:1001:
sispmctl:x:1002:natrixz
ntp:x:113:

Natrixz, 2018-07-26:
How about it throws this error because its a raspberry 2b without BlueTooth? =)

theyosh, 2018-07-26:
https://i.kym-cdn.com/photos/images/original/000/925/410/4cc.jpg

theyosh, 2018-07-26:
Ok, the latest code will at least continue now. Just added an exception to continue

Natrixz, 2018-07-26:
Ps;

Found some more issues;

HTTPSConnectionPool(host='www.yr.no', port=443): Max retries exceeded with url: /place/Netherlands/Overijssel/Deventer/forecast_hour_by_hour.xml (Caused by SSLError(SSLError("bad handshake: Error([('', 'osrandom_rand_bytes', 'getrandom() initialization failed.')],)",),))

SMTP AUTH extension not supported by server.
2018-07-26 22:59:54,260 - ERROR  - terrariumNotificatio - Mailserver login credentials are invalid. Cannot sent mail!

theyosh, 2018-07-26:
Yesh, I have them also after  reboot. Somehow the number generator of the OS is not done at that moment. Just keep it running. It will restart at my place only once and then it will all continue. Will add some code to fix this also...

Natrixz, 2018-07-26:
Alright:)

And i can conform it starts again now

theyosh, 2018-07-26:
Cool. Damm... did not thought about bluetooth less Pi's.... ah well.. I will blame the weather :)

theyosh, 2018-07-26:
Ok... then I close this issue.. Thanks for debugging

closed 2018-07-26 by theyosh