theyosh/TerrariumPI #176: Can not hide graph legend

TuxujPes, 2018-07-29:
Not able to switch "Horizontal graph legend" in the settings.

Also, tried, blindly,  add "horizontal_graph_legend=false" to "defaults.cfg" but with no results.

[image]

P.S Legend on the right is annoying since it covers the most recent values :)

theyosh, 2018-07-29:
Legend can not be hidden. But, if yo change it to Yes, save the settings and reload the hole page, it should be better.

This setting will only work after reloading the hole web page. It is a cached value.

TuxujPes, 2018-07-29:
It worked!
Damn... I thought that blue means "selected".
And that switcher didn't work.

Very off topic but not sure what I read on forums corresponds to what I did.

Can you blow raspberry by touching 5v and 3.3v pins together?
Generalized question is, can you fry raspberry by the means it gives you without any other external sources.

One of my rasps went on "red" led only :(

theyosh, 2018-07-30:
Hmm, not sure. I have had shortage in some sensors and start smelling burning plastic. And the wires did get warm. But after disconnecting, everything is stil working. But I am not a big hardware technician.
I do not think that you are able to blow it up with wrong cabling.

Only a red led is mostly a boot issue with the SD card. Only way to know is to hook it up to a monitor. HDMI to the television will work.

TuxujPes, 2018-07-30:
The SD card works just fine on another raspberry :(
That's a pity it lasted so short, but whatever.

Thanks for the help!

closed 2018-07-30 by TuxujPes