theyosh/TerrariumPI #177: Co2 sensor

5point0, 2018-08-07:
Would it be possible to add support for a Co2 sensor or should I try add this as a remote sensor?

I have found this sensor and it looks rather promising: https://www.co2meter.com/products/cozir-10000-ppm-co2-temperature-humidity-sensor?variant=878558412820

They also have some info using this sensor with raspberry PI and python:
http://www.co2meters.com/Documentation/AppNotes/AN127-COZIR-sensor-raspberry-pi-uart.pdf

theyosh, 2018-08-08:
As the sensor is a bit pricey for just playing around. So what we could do, is that I add support for CO2 sensors through remote data. Because that is the fasted way to do without having actual sensors.

So if you are handy with Python code, you could make a small script that is requesting the sensors data every minute (cronjob) and write the output in the folder: /home/pi/TerrariumPI/static/extern so that you ca use the own TerrariumPI webserver as remote server for the sensor data.

Ex: /home/pi/TerrariumPI/static/extern/CO2.json is the file you write with your script. Then the remote data will be: http://terrariumpi:8090/static/extern/CO2.json#sensors1

If your script is working, you could mail it to me, so that I can try to add it so that there is also native support. But then we need be sure that your code is stable enough to be added without the ability for me to test it.

5point0, 2018-08-13:
Thank you for adding this. 

I still have to order the sensor but I will test when it arrives. Ill close this ticket for now.

theyosh, 2018-08-16:
Hmm, I misted that apparently

So I merged development and master, as the only difference is the new DC-dimmer. So now it is a bit living on the edge :)

So with a git pull you will get the latest version and should contain the missing CO2 page.

@AndrejWeber for your DC-dimmer test, you can now also use the master (normal) branch. No switch to development is needed anymore.

rrgrow, 2018-08-16:
works nicely .. I'll tick to dev for now. Cheers

theyosh, 2018-08-16:
That is fine.... dev and master at the same code base currently. So I suggest that you switch back when a new release is made... Have fun with it!

5point0, 2019-11-11:
Hi,

I have connected a K30 co2meter to a PI.
So far I am able to output the current level(code below) but I'm not sure how to add this as a remote sensor with TerrariumPI.

    import serial
    import time

    ser = serial.Serial("/dev/ttyS0",baudrate =9600,timeout = .5)
    print ( "  AN-137: Raspberry Pi3 to K-30 Via UART\n")
    ser.flushInput()
    time.sleep(1)

    for i in range(1):
        
    
    ser.flushInput()
    ser.write("\xFE\x44\x00\x08\x02\x9F\x25")
    time.sleep(.5)
    resp = ser.read(7)
    high = ord(resp[3])
    low = ord(resp[4])
    co2 = (high*256) + low
    print "current: " +str(co2)
    time.sleep(.1)

Maybe @rrgrow has gotten this working?
Otherwise @theyosh could you please help me out here?

5point0, 2019-11-11:
I have added a co2 sensor in sensor options and tried to add an example value but I get the following error:

    TypeError:` '<=' not supported between instances of 'float' and 'NoneType'
    if current is None or not (self.get_limit_min() <=            terrariumUtils.conver_to_value(current,self.get_indicator()) <= self.get_limit_max()):
    File "/home/pi/TerrariumPI/terrariumSensor.py", line 114, in update
    self.sensors[sensorid].update()
    File "/home/pi/TerrariumPI/terrariumEngine.py", line 570, in __engine_loop
    Traceback (most recent call last):
    2019-11-11 13:07:22,817 - ERROR   - terrariumEngine      - Engine loop: Sensor has problems: '<='     not supported between instances of 'float' and 'NoneType'

theyosh, 2019-11-11:
@5point0 There are two ways to do this.

1. You use the remote sensor option, and then you need to output some JSON data to a text file. Someting like:

```
with open('/home/pi/Terrariumpi/static/extern/custom_co2.json', 'wb') as output:
  output.write('{ "co2" :  ' + co2 + ' }')
```
Then load that json file as a remote source with path: http://localhost:8090/static/extern/custom_co2.json

2. Use the sensor script option. Then you need to specifiy your script, and make sure that there is only one number as output. So no extra print lines. And only, only the value

5point0, 2019-11-12:
I have used the script option and it seems to be working well.

Do you thank you could add this? Or do you need the JSON data?

theyosh, 2020-01-28:
@5point0 Not sure what you mean with: Do you thank you could add this? Or do you need the JSON data?

The idea is that with the custom script option, you have more flexibility. Or you can use it through a JSON trick what I meantioned earlier.

But if you have the CO2 sensor on the same Pi as my software, I would use the 'custom script' option. Make sure your output is only the value. No extra text. That will cause errors.

But is your sensor an USB device? Because, your code is not that hard to add... I think.

theyosh, 2020-01-28:
Added direct support for the K30 CO2 sensor. Do a `git pull` and do a restart. Clear the browser cache, and you should be able to add a K30 CO2 sensor.

You have the enter as address: '/dev/ttySX' where X is the number in the /dev folder.

5point0, 2020-01-28:
Hi @theyosh, apologies I meant can you add direct support. But you did so thank you for that!

I have done a git pull and installed but I get this when I try to start and open screen.

2020-01-28 21:57:01,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.88.178:8090 ...
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 21, in <module>
    from terrariumEngine import terrariumEngine
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 782
    os.chmod('motd.sh', 0755)
                           ^
SyntaxError: invalid token
2020-01-28 21:57:04,000 - INFO    - terrariumWrapper     - Restart counter: 1/5
2020-01-28 21:57:04,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 3 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5

Keep in mind I use `sudo /home/pi/TerrariumPI/start.sh 3` as I need Python 3.

theyosh, 2020-01-28:
my mistake. Made a fix. Try a new `git pull` and restart.

5point0, 2020-01-28:
Okay great it's starting again.

And to answer your question, I use the UART TXD‐RXD Pins and not USB.

I need to connect the sensor back to the PI as I was having an issue where it doesn't seem to work with Buster but for some reason the same code works on stretch. 

So I'm not sure if thats still going to be a problem but will let you know.

5point0, 2020-01-29:
Hi @theyosh,

So this is weird, I tried a fresh buster install with a Pi3B and I was able to get an output with the below code. However I tried the same process on a PI3B+ and it gives me a 0 value regardless if the sensor is connected or not. 

I have now tried on a Pi4B and this also seems to work. So I have decided to stick with the Pi4B+.

Now the results.

Code I run to test the sensor that just gives me the value and nothing else.(Custom script I use in TerrariumPI too that works):

    import serial
    import time

     ser = serial.Serial("/dev/ttyS0",baudrate =9600,timeout = .5)
     #print ( "  AN-137: Raspberry Pi3 to K-30 Via UART\n")
     ser.flushInput()
     time.sleep(1)
   
     for i in range(1):

     ser.flushInput()
     ser.write("\xFE\x44\x00\x08\x02\x9F\x25")
     time.sleep(.5)
     resp = ser.read(7)
     high = ord(resp[3])
     low = ord(resp[4])
     co2 = (high*256) + low
     print str(co2)
     time.sleep(.1)`

I have tried to add the sensor via the direct support you added but I get this error:
`2020-01-29 18:23:11,796 - WARNING - terrariumSensor      - Measured value Noneppm from k30co2 sensor 'Test Co2' is outside valid range 0.00ppm - 10000.00ppm in 1.00204 seconds.`

I have tried /dev/ttyS0 and /dev/serial0.

5point0, 2020-01-30:
Maybe this permission denied is the issue?

    2020-01-30 09:23:49,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.88.140:8090 ...
    2020-01-30 09:23:57,289 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device! If this is not correct, try 'i2cdetect -y 1' to see if device is connected.
    2020-01-30 09:24:02,326 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
    [Errno 13] could not open port /dev/ttyS0: [Errno 13] Permission denied: '/dev/ttyS0'
    2020-01-30 09:24:14,753 - WARNING - terrariumSensor      - Measured value Noneppm from k30co2 sensor 'Co2' is outside valid range 0.00ppm - 100.00ppm in 0.00116 seconds.
    2020-01-30 09:24:19,000 - INFO    - terrariumWebserver   - Running webserver at :::8090
    [Errno 13] could not open port /dev/ttyS0: [Errno 13] Permission denied: '/dev/ttyS0'
    2020-01-30 09:24:20,109 - WARNING - terrariumSensor      - Measured value Noneppm from k30co2 sensor 'Co2' is outside valid range 0.00ppm - 10000.00ppm in 0.00045 seconds.

theyosh, 2020-01-30:
Hi,

It looks like that the serial is still used by the Pi for logging. Have you disabled the serial interface?

Sudo Raspi-config —> Interfacing options —> serial —> no —> yes -> save -> reboot

Then, I need to know the line where the unicode error is happening. In Python3 they are bit stricter with the character sets. So it is a bit hard to fix without the actual hardware.

If you can test your own code with Python2 and Python3, and see if it works. And I hope you can tweak it a bit, so it will work for both. Then I can use that code to see what I need to fix in my code.

5point0, 2020-01-30:
Hi,

If i run the code manually with python3 I get this:

    Traceback (most recent call last):
      File "3co2.py", line 15, in <module>
        ser.write("\xFE\x44\x00\x08\x02\x9F\x25")
      File "/usr/local/lib/python3.7/dist-packages/serial/serialposix.py", line 532, in write
        d = to_bytes(data)
      File "/usr/local/lib/python3.7/dist-packages/serial/serialutil.py", line 63, in to_bytes
        raise TypeError('unicode strings are not supported, please encode to bytes: {!r}'.format(seq))
     TypeError: unicode strings are not supported, please encode to bytes: 'þD\x00\x08\x02\x9f%'

Im not sure if that will help somehow.

I emailed the suppliers of the sensor, maybe they have code for python3.

theyosh, 2020-01-30:
Ah, that is easy :) Just add a 'b' in front of the string:

ser.write(b"\xFE\x44\x00\x08\x02\x9F\x25")

I have updated my code, and I hope it will not work. Do a nother `git pull` and restart

5point0, 2020-01-30:
I have updated, but seems to be another issue.

In Screen:

    2020-01-30 17:27:57,801 - WARNING - terrariumSensor      - Measured value Noneppm from k30co2 sensor 'Co2 k30' is outside valid range 0.00ppm - 10000.00ppm in 1.50391 seconds.
    ord() expected string of length 1, but int found

Manual run:

    Traceback (most recent call last):
      File "3co2.py", line 18, in <module>
        high = ord(resp[3])
    TypeError: ord() expected string of length 1, but int found

5point0, 2020-01-30:
I have done some troublshooting and got the code working with python3 :)

    #rpi serial connections Python3 app to run a K-30 Sensor

    import serial
    import time
    
    ser = serial.Serial("/dev/ttyS0",baudrate =9600,timeout = .5)
    #print ( "  AN-137: Raspberry Pi3 to K-30 Via UART\n")
    ser.flushInput()
    time.sleep(1)

    for i in range(1):
        
    
        ser.flushInput()
        ser.write(b"\xFE\x44\x00\x08\x02\x9F\x25")
        time.sleep(.5)
        resp = ser.read(7)
        high = resp[3]
        low = resp[4]
        co2 = (high*256) + low
        print (co2)
        time.sleep(.1)

theyosh, 2020-01-30:
Hmm, I think this is not the right way. Because I think the value that is now measured is not correct.

So I think, you should change the following two lines and test again:

```python
high = resp[3]
low = resp[4]
```
to
```python
high =ord(str(resp[3]))
low = ord(str(resp[4]))
```

But I am not sure if that works. Because I think using the values directly as you do now, is not valid

5point0, 2020-01-30:
Okay, I have tried that but I get this error:

    Traceback (most recent call last):
      File "3co2.py", line 19, in <module>
        low = ord(str(resp[4]))
    TypeError: ord() expected a character, but string of length 3 found

From what I could tell I was getting the right value, not sure if this would pick up issues later on though.

Good news though, the supplier has responded to my mail and is looking into the code for python3.

theyosh, 2020-01-30:
Hm, ok. Update my code based on yours. Could you do a `git pull` and see if it works for you?

And then we see what the supplier says...

5point0, 2020-01-30:
So seems to be working now. 

    2020-01-30 21:55:02,554 - INFO    - terrariumSensor      - Updated k30co2 sensor 'test K30' from 0.00ppm to 416.00ppm in 1.50303 seconds
    2020-01-30 21:54:59,460 - INFO    - terrariumSensor      - Updated script sensor 'Co2 Custom' from 416.00ppm to 416.00ppm in 2.38574 seconds

Will let you know what they say..
Thanks for the help so far!

theyosh, 2020-03-06:
Still working? So can we close this issue?

5point0, 2020-03-11:
Yes still working and still waiting on the code. I will open again if they send it. Thanks for the help with this.

closed 2020-03-11 by 5point0