theyosh/TerrariumPI #178: Sun rise and set by dimmer and relais

AndrejWeber, 2018-08-11:
Hallo,
I try to simulate a sunrise end sunset by dimming a T5 Lamp.
what do I have to set, to make the following states (0% light till 100% light)

08:00 am: switch on Relais and raise PWM Dimmer 0% to 100% in 600s
07:50 pm: PWM Dimmer 100% to 0% in 600s
08:00 pm: switch off relais

switching on and dimming the lamp works great, but I do not figure out, how to dim to 0% back and then switch off.

just one more thing:
Is it possible to set a hardware button as signal input like the door switch?
I want to switch on the light for about 10 minutes in night with minimal light intensity by pressing a Button.

Thanks a lot

theyosh, 2018-08-13:
Hi,

if you managed to get the dimmer going up, the going down is pretty easy. You should use the field Dimmer off duration . If you put there also 600, it should dim when it is time.
This does only work if you specify the dimmer switch in combination with the lights in the environment part. 

[image]

This my running setup. This will also slowly dim the light when it should go off. But it will start dimming when it is time, which will then cause the light to run a few minutes more. Because I do not shorting the light window based on dimmers.

theyosh, 2018-08-13:
Sorry, my mistake. The logic for dimming has changed a bit. What I have done is the following:

Use the lights parts on the environment part to control the dimmer lights. Use the day part for selecting the power switches:
[image]

Then the environment should be configured to go on at 8:00 and off at 20:00 (timer  mode)

Then when the lights should go on at 8, it will 'power up' the dimmer every 15 seconds (environment check time). The dimmer will then go up for 'Dimmer environment step *' percent in 'Dimmer action duration *' in seconds. So based on my previous screenshot my dimmer will go up 3% in 14 seconds every 15 seconds. That means that my dimmer goes on in (100 / 3) * 15 seconds = 500 seconds. That is almost 10 minutes.

And when the lights should go off, the environment will 'power down' the dimmer the same way as powering up. I use 14 seconds duration to stay withing the 15 seconds environment duration check. Just to be safe then sorry...

So I used it through the environment part. And not with the powerswitch timer itself.

theyosh, 2018-08-13:
So you have created two power switches for this? One normal and one dimmer?

What if you set the timer option on the normal one that starts 1 or 2 minutes before the dimmer goes on through environment. And the timer goes off 1 or 2 minutes + configured dimmer delay after the off time of the environment. Leaf the relay switch outside the environment power settings. Only select the dimmer.

So the relay switch will be powered on before the dimmer kicks in and stays on after the dimmer is done. The only downside is that you have to use static times so it is a bit of hassle to change the timings.

And If the relay stays on the dimmer will work both ways? I am not sure of dimming down is working or not if there is power on the other relay. Then the 'problem' is that we have to delay the normal powerswitch that powers the relay. Which I think can be done with the above solution

AndrejWeber, 2018-08-13:
You are right, that is the first usable configuration.
I will give it a try right now.
Of course I cannot use weather switching.

The dimming is absolutly indipendent to Relais. So I can also use the Lamp without dimming or with a resistor or even with a switch. By shorten dimming input the light works with 1%, by open the input there are 100%.

AndrejWeber, 2018-08-13:
Ok, so a quick test do what it should do. thank you.
Now I have to set up realistic time and wait for next day.
Is it possible to add a new dimmer by you, that does not invert the output? Also this switch need its own PWM_DIMMER_MAXDIM = 1000. (if someone need somethink other)
This PWM dimmer can be used by me :D and others for own hardware and also PWM-fan.

before you close this issue...

is it possible to fire on the switch manually, that is controlled by timer?
And also can I read out a hardware button in realtime, to switch the ligts on? Yust for feeding the geckos in night-mode.

theyosh, 2018-08-14:
Euh to get things right. What kind of hardware do you use to get 1 - 10V for the OSRAM QTi dim light? Because my dimmer is made for 220V so that is different. If you can give me a link to yours, I can add a new dimmer power switch to the system. Is not that much work. And then I can use other maximum values.

You request for a button is a bit hard. I have the same question for opening the door that will also control some lights. But this will be a bit more complicated, because you can ask then for how long will the light be on? Or should something power down when the door is opened or button is pressed.
And also breaking in on the timer or environment system to power up lights that should be off... is also adding extra logic code. Which for now I think will a to complicated. I have to think about it.

The initial request is simple, but there are more options and more minds.... so I have to make it more generic and test different scenarios. So for now... nope.. sorry.

I have day geckos. So I can just feed them with the lights on :P

AndrejWeber, 2018-08-14:
There is no link to a hardware. I have a Breadboard with a self made amplifier logic. 
I use the native PWM signal with a capacitor, so I have a analog output from 0-3.3VDC. This will be amplified to 0-10V by operational amplifier OpAmp. 
It's a simple digital->analog converter. 
So you only need to make the same PWM-dimmer, like yours, by deleting the "100 -" 3-times in your switch.py
Then it is a normal PWM out with 100% duty cycle when switch is at 100%. 
Re-invert the given one is easy, but I was not able to edit code everywhere to add a second PWM Switch, that uses the is-slider. 
I do not have any skills in programming Python and Java, but I can read and edit the most of them. I am able to program in vba, there are some parallels. 
Yes, adding buttons is more complicated. 
To breaking in environment is the way I would do.
It seems to be simple to get an input button (allready there in doors) and make a new environment with doors instead of sensors. 
Then you "only" have to tell all other environments to wait or ignore (like looking for light is active), while the "new-environment" is active....
I see, it's not as easy and much work to edit most of your code. 
It would be easier to take a "door-input" and set it up with timer to activate a new gpio or even deactivate.
I could use this workaround, but is not usable for other hardware (i2c)
I have thousands of Ideas, how to realise it, but I'm not able to do :(

So the new PWM-dimmer is much more important for me. I can also bypass the switching of relais hardware side. 
Thank you for your great work.

theyosh, 2018-08-15:
I just created a new dimmer called 'dc-dimmer'. If you checkout the **development branch** (NOT master), restart the software **AND CLEAR BROWSER CACHE** you should see a new power switch type.

This one is based on your request, and should work the same as the PWM dimmer, but in reversed way and higher max dim value. Could you give it a try to test it? As I cannot test it here...

AndrejWeber, 2018-08-15:
Oh, I'm surprised 😅
I did not expect it so fast. 
I have to send my pi back to Amazon. Seems it was damaged by shipping in envelope. Package was jammed.
Sins Monday night it gets very hot and consumes to much power. PWM does nothing anymore.
Tomorrow or on Friday I will get a new one to test the new switch. 
Thank you very much for your great work!

AndrejWeber, 2018-08-16:
hey Yosh,
my new pi is here, so I'm ready to test.
First I'm not shure, how to instsall the development version.
But I realized, you are runnig the development Version. there I can see, that the dc-dimmer has not the settings like pwm dimmer.
Like I saw in code, you did not use the duration, step and so on for the DC-dimmer. so it would change the state immediately.
So, if you tell me, how to install, I can test the changes at my pi.
thanks a lot

theyosh, 2018-08-16:
Ah yes, missed that. That is now also fixed.

If you install the software like described in the readme trough git, you can then change to the development branch with the command in the folder `/home/pi/TerrariumPI`:
`git check development`
This should give some more files and update some others. But it should not interfere with your settings and history data. Then restart the software and it should run development version.

theyosh, 2018-08-16:
Due to #177 you can now also use the master branch to test the new dimmer

AndrejWeber, 2018-08-16:
IT WORKS GREAT!!!
I just added the PWM frequency to 15k, because the Fan ist whisteling by 5k.
Now I have to setup the environment and timer.
Thank you very much

And if you are bored some day, or week 😄 dont forget my idea to add a button as input.

AndrejWeber, 2018-08-20:
Hey Josh,
I have tested the new switch now for some days. 
My review:
It works like it should under environment, but not under timer-config. So, if I set the dimmer in environment and the relais as simple timer in switch config, then it works like expected. 
But in other direction, the dimmer rises and fall to set percentage immediately. 
The log shows " dimmer set from 8% to 80% in 600s, but it will done in a few milliseconds. 
You told, you changed the dimming logic. Is that timer-dimming not work anymore?
The second issue, is that the graphic for the dimmer do not show the state. It is a sero-line, in "home" it shows the correct wattage.
I "think" or "believe" I have clear all cache, since the new dimmer. I tested it at my iPhone, laptop (Firefox and I-explorer) and on the Raspi. No one shows the graph.
I can live with that, my geckos do not look for graphs ;) but it is not perfekt, like the other software strives for perfection. 
If you have time, take a look for this little problem.
At all, I am able to dim my Osram QTi and also a PWM computer fan. I am happy

Thank you very much.

theyosh, 2018-08-20:
I have to look into this...

theyosh, 2018-08-21:
Update the code to set DC-dimmer to 15K and PWM to 5K frequencies. Fixed the graph for dc-dimmer.

And yes, the logic for dimming is different between environment and powerswitch timer.... Should not be. So that I have to look into it.

AndrejWeber, 2018-08-26:
So, after testing a view days, it all running fine.
No new problems appear. 
Thank you once again!

AndrejWeber, 2018-10-26:
Hey Josh,
I have found a little Problem with the DC Dimmer.
It does not save the whole settings in config file. 
I think the following code from config.py does it for PWM-Dimmer. 

 # Switches config functions
  def save_power_switch(self,data):
    clearfields = ['state','current_power_wattage','current_water_flow']
    if data['hardwaretype'] != 'pwm-dimmer':
      clearfields += ['dimmer_duration','dimmer_off_duration','dimmer_off_percentage','dimmer_on_duration','dimmer_on_percentage']

The DC-Dimmer should also be saved. 

Have a nice Weekend. 
Thanks

AndrejWeber, 2018-10-26:
I did a little change in the code:
`if data['hardwaretype'] != 'pwm-dimmer':`
__`if data['hardwaretype'] != 'dc-dimmer':`
____`clearfields += ` [...]
Now the values are saved, when I restart the software.

Maybe there is a better way, you should know, what is the right way.

theyosh, 2018-10-27:
Thanks for this. I fixed it a slight different way. But it should now be fixed.

AndrejWeber, 2018-10-29:
Perferct!
Nice work, it is fixed!
thank you Joshua

theyosh, 2018-10-30:
So... lets close it a second time ;)

closed 2018-10-30 by theyosh