theyosh/TerrariumPI #180: None Raspberry py Support

chrigui, 2018-08-14:
Hello,
First thank you all for such a great project.
i would like to know if its possible to install TerrariumPI on a normal web server (Digitalocean VPS) and have Arduino/ESP8862 code communicate with TerrariumPI Server. 
The reason i am asking is we are designing an Arduino based Aquarium Controller and would like to have the web interface in the cloud without having to connect it directly to a Raspberry Pi.

it is possible if TerrariumPI can support Http post/get/MQTT or even mysensors protocols as external sensor/hardware.
Many thanks

theyosh, 2018-08-18:
Hi,

I have thought about this now for a couple of days, and I think it is not designed to do what you want to do with it. I made the software primarily on pull basis. So moving my software to the cloud will not work.

And adding code for supporting push sensors is not something that I need and will not make it in my software.

But there are a lot cloud providers that support sensors and actions. https://ifttt.com/

So am I afraid that my software will not work in the cloud. And for me as this is a hobby project, I do not need that. I have just opened my router to go the the webinterface on my home DSL connection.

theyosh, 2018-08-20:
Closing as we having direct contact per email

closed 2018-08-20 by theyosh