theyosh/TerrariumPI #182: DHT22 Humidity not reporting

romandiaz, 2018-08-18:
I have a DHT22 connected to my PI. It appears to work, but does not show any humidity within the TerrariumPi interface. It always shows 0.

Testing shows that it does report both temp and humidity:

>[image]

This is how my sensor page is setup with TerrariumPi, which you can see reports '0' for current humidity:

[image]

Using command `screen -r ` prints out the following warnings - not sure if these are relevant because it shows errors for temperature as well:
```
2018-08-17 17:06:35,481 - WARNING - terrariumSensor      - Measured value None% from humidity sensor 'DHT22' is outside valid range 50.00% - 100.00% in 12.63574 seconds.
2018-08-17 17:06:35,609 - WARNING - terrariumSensor      - Measured value NoneF from temperature sensor 'DHT22' is outside valid range 50.00F - 100.00F in 0.00112 seconds.
```

Does this make any sense? Is there any configuration I'm overlooking or suggestions for how to continue debugging? Also noting that even though I have system set to report temps in Fahrenheit, the temperature that is shown is still in Celsius. Is this a bug or have I not set things up correct? Thanks much, in advance!

theyosh, 2018-08-18:
You have entered BCM port number. I use physical port numbers https://github.com/theyosh/TerrariumPI#gpio-numbering

Change 18 to 12 and it should work. But I would also suggest not to use that GPIO port, as it is also used for PWM (dimming) actions. https://pinout.xyz/

Use physical GPIO port 16 (BCM 23) and you have still room for dimming software.

romandiaz, 2018-08-18:
Yes, that was my problem exactly! It is now working 100%.

Strange thing, when I used physical port numbers yesterday it reported '0'. The '32' showing up when I entered BCM made think that was correct - likely a problem on my side though. This is my first large PI build and there is so much to learn. Thank you again for your help, and for writing this awesome project!

closed 2018-08-18 by romandiaz