theyosh/TerrariumPI #183: Problem with environment temp night/day difference

TwoCH, 2018-08-18:
Hey,

we just setup our new terraium with your great software but I ran into a little issue.

I set up the temperature under environment to work with a sensor and a timed light.
When it is night I need the temperature to drop 7 grad. So far so good.

Now the problem, if I change anything in the environment settings when it is nighttime or the terrarium pi software restarts at this time, the day/night difference is substracted again. So our normal temprange is 28 - 31 grad. After working again on the environment and a restart of the software the sensor showed a range of -3 to -1, wich is not that nice :)

Maybe it would be better to not store the day/night difference as replace for the sensor alarms but instead add its own config values or calculate it on the fly and only store the difference as a value.

I hope you can fix this or explain me, what I am doing wrong to get this behavior.

Thank in advance

TwoCH, 2018-08-20:
Hello again,

unfortunately I ran into more problems with the temperature enviroment.

1. If I have an needed temperature range from 28 - 31 and the sensor gives a warning at 31 that it it to hot, the environment settings dont fire the shut off order until it is 32 degree. This is a bit annoying because the warning is there all the time.

2. Besides point 1 the temperature shut off and on seems to work on daytimes, but we need a night solution, too. As of now, there is the bug from the first posting and it just don't works at all in the nighttimes for us. On the dashboard it shows the right lower temperature range in the night, but the heatbulb still gets turned on, even if the range say 21 - 24 and it's 28 degree at the moment. We have no clue why this is happening.

This is our setup:
We set up a day/night switch with the light settings:
[image]

The temperature settings are the following:
[image]
On the picture the difference for day/night is 0, because we need to shut the spot out at the night, so it doesn't get turned on in the middle of the night, heating everything to over 30 degrees. Before we had -7 there. 

This is what the dashboard looked like yesterday night, when the -7 still was in place:
[image]

For my understanding, the spot should be turned of, it's 32 degree but the range is 21-24 degree, still the sensor was put on.

Do we something totally wrong? Did I missunderstand the function? Please help.

Thanks
Tom

theyosh, 2018-08-20:
Sorry I was a bit busy. Will respond withing 24 hours from now

theyosh, 2018-08-21:
Yes, changing sensors in night time will cause wrong values in the end. This is a know design flaw. But it should not happen with a reboot.

The idea is that when the temperature needs to be changed during the night, that is NOT stored in the settings file, but only activated in the sensors object in code. So a restart should not alter the values.
But changing your sensors during the night will cause wrong values. And I do change the sensor values so the graphs will reflect the changed sensor values. So you can see that the min and max values are changing in your graphs.

But how often are you changing the sensors values? That is mostly only done a few time during installation. For me, I have not changed the sensors for more then 6 months. System is running fine. Therefore, I did not make the code more robust for this. For now, it is not possible to change sensor data during night time when the environment has to change the sensors values during the night.

Second part will follow later on...

theyosh, 2018-08-21:
Ok, lets see if I have it correct. You want to control the temperature 24/7 with a heater (a lamp in this case) based on a sensor. And during the night, the temperature should drop a bit, but still the heater(light) should be on? That means when it is to cold, a light will go on?

As I can judge from the picture it looks good. But the power switch that is used for the heater, is that a dimmer by chance? Or did you also have a power switch selected at the high alarm options?

It looks all correct, so I am testing my self this setup, and see if I can find anything. What I do see is that the light settings are a bit strange. The on and off values indicates 12 hours light and then 12 hours no light in a period of more then 12 hours based on the timer start and stop values. Is this intended? I would expect that just the start and stop times are enough, and the on and off values can be zero, so that they will be ignored.
But as you selected, the temperature should ignore the light state, only to know when it has to lower the temperature values based on night time.

So for now, I have to think a bit more, and test a bit. Could you maybe upload the logfile? Maybe that will give a clue

theyosh, 2018-08-21:
For now, rebooting during night time will not influence the environment temperature difference. So that is working as expected.
Second a warning sign is always shown when the measured value is outside the min and max, either to hot or to cold. So that is normal. But here, it is at three decimal precise, so I can't reproduce your warning issues of going of to late.
Third, I noticed on your images that the environment dashboard is not the latest version. Are you running up to date code? Could you else update to the latest code or reinstall? Get the feeling you are not running the latest code.

theyosh, 2018-08-23:
After 24+ testing, it is working like expected. So I cannot reproduce your issue with the above data.

So I think that somehow you have mixed up code. Can you make sure you are running the latest version?

theyosh, 2018-09-09:
Any updates on this?

TwoCH, 2018-09-09:
Hey, sorry it took so long.
Over the weekend we did a reinstall and the problem diasppeared somehow. We cannot replicate it anymore

theyosh, 2018-09-09:
Ok. Then I close this now. Thanks for the update

closed 2018-09-09 by theyosh