theyosh/TerrariumPI #184: Add light state option to webcam archive configuration

h1kkan, 2018-09-08:
Hi @theyosh !
Could you please add "light state" param to the webcam archiving function? 
It would be nice to be able to configure camera archiving depending on light/day state. Thanks!

theyosh, 2018-09-09:
Just made. If you pull the latest master branch, restart and clear you browser cache, there are two new options in the webcam settings page under archive.

h1kkan, 2018-09-09:
Could it be also added to the timer options (currently it is available only for motion)? So it will be possible to take shots with interval only when light is on.

theyosh, 2018-09-09:
Ah crap... I thought that this would be a smart fix... but your right... I will change to add an extra option for light on and of option

h1kkan, 2018-09-10:
Tested it last night - all seems to be ok. Images are taken in configured interval and depend on env lights state. No errors occurred during tests. Thanks!

closed 2018-09-10 by h1kkan