theyosh/TerrariumPI #185: Watchdog feature

Natrixz, 2018-09-08:
Would it be possible to add Some sort of watchdog feature?
It has happend quite a few times that Some sort of error caused the software to hang or die. (Usually hang)
It would be very Nice to have Some sort of watchdog feature that checks and restarts the software if needed

theyosh, 2018-09-09:
Needs a bit more info. Is the Pi still running and accessible through SSH? Are there any errors? Is a restart of my software enough, or needs the Pi to be rebooted?

A way is to look at the logfile. When it is not changed for x minutes, you could try to restart. But it depends on the error you have.

Natrixz, 2018-09-09:
Yes, the Pi is just accessible through SSH.
All that needs to be done is; screen -r and then ctrl+c and it's fixed.

As far as I've checked it's ALWAYS preceded by an error and then dies off/hangs.

theyosh, 2018-09-09:
But... that means that I need to know the error, and make the code more robust. So could you post the error next time you have it?

AndrejWeber, 2018-09-09:
Don't forget to check your supply.
It can cause a lot of unexpected problems.

Natrixz, 2018-09-10:
Yep I will tell you next time it happens.
Sadly it's not listed in the zipped files :(

@AndrejWeber yes I know, it's already been replaced to rule that out (both times using an original supply)

bjornarfonn, 2018-09-10:
Have you checked the hardware monitor to see if the CPU usage is high? I experienced this issue once, where it would run at 100 - 150 % usage.. causing the interface to hang and become non-responsive, but still accessible through SSH. This was on a pi zero

Natrixz, 2018-09-10:
Yeh I did, actually the Pi is just very responsive.
Also can't find anything else strange.

It's just the Exception that crashes the software/hangs it.

I will post it when I see it again ;)
Then you can see haha

Natrixz, 2018-09-10:
This time no errors, just python hanging at 100% CPU.
[image]

theyosh, 2018-09-12:
Hmm, I do had a month ago something that looks the same as this. And that turned out that I also had a broken sensor, which my software just told and continued. But somehow, I had a hanger every 24 hours until I removed the sensor from the config and restarted. Still strange because the code should take care of it, but I can image that not all libraries will clear there memory on a sensor read failure. Maybe my own code is also not clearing memory of faulting sensors. That I need to figure out.

That the software continues is unfortunately not a guaranty that everything is fine... :(

So in this case, I would suggest to remove this strange behaving sensors and see if it does not hang anymore. I will check my code to see if there are other memory leaks.....

Natrixz, 2018-10-06:
2018-10-06 15:28:32,184 - ERROR   - terrariumSensor      - Error updating owfs temperature sensor 'Links-Koud' with error:
Traceback (most recent call last):
  File "/home/nick/TerrariumPI/terrariumSensor.py", line 380, in update
    current = sensor.get_temperature()
  File "/home/nick/TerrariumPI/terrariumSensor.py", line 169, in get_temperature
    self.__get_raw_data()
  File "/home/nick/TerrariumPI/terrariumSensor.py", line 159, in __get_raw_data
    self.__humidity = self.__sensor.humidity
  File "/usr/lib/python2.7/dist-packages/ow/__init__.py", line 347, in __getattr__
    return owfs_get(object.__getattribute__(self, '_attrs')[name])
  File "/usr/lib/python2.7/dist-packages/ow/__init__.py", line 159, in _get
    raise exUnknownSensor(path)
exUnknownSensor: '/26.C11421020000/humidity'
2018-10-06 15:28:32,191 - ERROR   - terrariumSensor      - '/26.C11421020000/humidity'
Traceback (most recent call last):
  File "/home/nick/TerrariumPI/terrariumSensor.py", line 380, in update
    current = sensor.get_temperature()
  File "/home/nick/TerrariumPI/terrariumSensor.py", line 169, in get_temperature
    self.__get_raw_data()
  File "/home/nick/TerrariumPI/terrariumSensor.py", line 159, in __get_raw_data
    self.__humidity = self.__sensor.humidity
  File "/usr/lib/python2.7/dist-packages/ow/__init__.py", line 347, in __getattr__
    return owfs_get(object.__getattribute__(self, '_attrs')[name])
  File "/usr/lib/python2.7/dist-packages/ow/__init__.py", line 159, in _get
    raise exUnknownSensor(path)
exUnknownSensor: '/26.C11421020000/humidity'
^CTraceback (most recent call last):
  File "src/gevent/greenlet.py", line 716, in gevent._greenlet.Greenlet.run
  File "/home/nick/TerrariumPI/terrariumEngine.py", line 421, in __engine_loop
    self.sensors[sensorid].update()
  File "/home/nick/TerrariumPI/terrariumSensor.py", line 388, in update
    current = sensor.get_distance()
  File "/home/nick/TerrariumPI/terrariumGPIOSensor.py", line 202, in get_distance
    return self.get_current()
  File "/home/nick/TerrariumPI/terrariumGPIOSensor.py", line 193, in get_current
    self.__get_raw_data()
  File "/home/nick/TerrariumPI/terrariumGPIOSensor.py", line 181, in __get_raw_data
    while GPIO.input(terrariumUtils.to_BCM_port_number(self.__datapin))==0:
  File "/home/nick/TerrariumPI/terrariumUtils.py", line 73, in to_BCM_port_number
    if index in pinout:

theyosh, 2018-10-16:
Hmm, this is something hard to judge. I made code for catching errors. And with OWFS there is no cleanup code. So I would expect that catching the error is enough. But there could be a memory issue with this. But then again, the source is OWFS (library). And that is not something what I can fix.

A solution could be, that I count errors in total (sensors, webcam, other) and after X amount, restart the hole system. But this would rater involve a lot of code.

Maybe a different way is to have some tool that is able to look into the logfile in realtime and counts the errors/warnings. And restart based on custom defined logic.

Natrixz, 2018-10-17:
Yes or a daemon checking to see if there's still log output beeing written and if there isn't, something's wrong, thus restart it?
(probably easier)

theyosh, 2018-10-18:
Idea:
`#!/bin/bash

MAXTIME=3600 # In seconds
LOGFILE="/home/pi/TerrariumPI/log/terrariumpi.log"
PID="$(ps fax | grep terrariumPI.py | grep -v grep | awk {'print $1'})"
AGE="$(($(date +%s) - $(date +%s -r ${LOGFILE})))"

if [ $AGE -gt $MAXTIME ]; then
  echo "Logfile is not updated for $MAXTIME seconds. Restarting..."
  kill $PID
fi
`

Natrixz, 2018-10-19:
Well im guessing that could work, let me try it out :)

Natrixz, 2018-10-23:
It appears to be working so far :)

theyosh, 2018-10-23:
Added a new script with almost the same content. I do expect the script to run as the same user running TerrariumPI

theyosh, 2018-10-27:
Closing now. There is a small cronjob script that could help with this issue. Needs manually installation.

closed 2018-10-27 by theyosh

theyosh, 2019-01-08:
Hi,

I think I found this issue last month. I also experienced these kind of problems with a hanging system with 1 CPU at 100%. It is due to the HC-SR04 sensor. Sometimes it will not get calibrated right, and will wait to get in that state which never happens.

The code fix will measure the calibration time and if it is not getting it within 2 seconds, it will abort and give a warning. And the next round it will try again. So no more hanging system.

Do a git pull and restart. Disable the watchdog cronjob and see if it keeps working.