theyosh/TerrariumPI #186: Wrong Humidity?

Natrixz, 2018-09-11:
Heya,

Been wondering this for a while now....im using a HIH-5031 and for a while now it says its been Reading above 100% Humidity.
(Both of them do)

Is it possible there’s Some error in the calculations? (Not sure how something can be above 100% RH) xD

theyosh, 2018-09-12:
Euh, if code is not changed.... hardware failure... :( It would be strange that calculations differ when code is the same.

A guess, as there are humidity sensors, that maybe some corrosion is happening due to to wet...? I have the same issue with the Chirp sensors. They are nice, but when the board is not shielded / covert somehow, the onboard resistors and chip started to corrode due to humidity of 60% - 70%. And sensors started to behave strange and interrupt the I2C bus.

Natrixz, 2018-09-12:
Hmmm that would be strange cause they both that since i have it xD
Can try another replacement but my other replacement does the same

Natrixz, 2018-09-12:
Maybe i should buy Some SHT20’s to avoid corrosion >,<

Natrixz, 2018-09-13:
Would you or anyone else have any recommendation's about humidity sensors that are waterproof and have atleast a cable of 1.50m (preferably longer) ?

theyosh, 2018-09-18:
Hmm, about cable length I am not sure, but I think that between my sensor and pi is certain 2 meter cable. But I use the following hardware:

http://www.sheepwalkelectronics.co.uk/product_info.php?cPath=22&products_id=30 ( on the pi)
http://www.sheepwalkelectronics.co.uk/product_info.php?cPath=23&products_id=56 (In my terrarium, for more sensors)
http://www.sheepwalkelectronics.co.uk/product_info.php?cPath=23&products_id=55 (I have older, smaller models)

But I added a small casing around it. I have a older board, which I could remove the RJ45 jacks, and only use telephone wire. The case is totally closed and added some holes to get the air flow and measurement. This way, the sensors stay pretty dry. Also, I have coated it the board so it would resist water. Except the sensor it self...

My sensor : https://terrarium.theyosh.nl/static/images/photos/20170704_163822.jpg
My sensor hub: https://terrarium.theyosh.nl/static/images/photos/20170704_163845.jpg
My Rpi : https://terrarium.theyosh.nl/static/images/photos/20170704_174223.jpg

Natrixz, 2018-09-20:
I pretty much use most of the same hardware :)
Pretty much everything except the sensors, I used these;
https://taaralabs.eu/1-wire-humidity-temperature-sensor/
One of them (after I removed the faulty one) still works fine, the other however died.....maybe it came in contact with water too much. (my mistking setup to keep humidity at 90-100% does a fairly good job of misting the whole enclosure lol)

Maybe I should measure and design a 3d-printable case around the sensors as well.....
Good point =)

However, strangly my 2 brand new sensors all report +100% humidity if I stick it in at the moment xD

theyosh, 2018-09-21:
I used these cases: https://www.conrad.nl/p/strapubox-2043-ow-modulebehuizing-54-x-37-x-21-522244

Drilled some small holes in it so that the air can flow through

Natrixz, 2018-09-24:
Ah, maybe I can fit my modules in those and do the same =)
Thanks =)

theyosh, 2018-10-16:
I think this can be closed? As I do not have other solutions for this

Natrixz, 2018-10-17:
Yes just close it :)

closed 2018-10-17 by Natrixz