theyosh/TerrariumPI #187: Wemo Switch support?

fenyx13, 2018-09-15:
Hi, is it possible to support the wireless Wemo power switch?

theyosh, 2018-09-18:
Hi, should be possible. I have seen some libraries to communicate with those. But I do not have the hardware at home, so it is a bit hard to test and make correct code.

Also, those power switches can also be powered of by phone or schedule, which means that not my software is controlling the switch, and needs to listing for changes. This is new, and could be a bit more work to make.

Needs to think if I am going to buy one for testing....

theyosh, 2018-10-13:
Just ordered two devices. I hope that they will arrive next week.

fenyx13, 2018-10-14:
Oh wow! Thank you so much for this! :)

theyosh, 2018-10-22:
It is not fully tested, so expect still some issues when the WeMo device is offline.

Also, TerrariumPI will also check every 30 seconds the state of the WeMo. So if you manually switch it offline, TerrariumPI will notice this within 30 secs.

theyosh, 2018-10-27:
Any issues?

fenyx13, 2018-10-27:
I just tested the Wemo switch by cooling a temperature sensor and the switch turned on my heater. 
No issues so far. Working as intended. Thanks!

theyosh, 2018-10-28:
Cool. Then I close it now. If there are issues later on, just re-open this issue

fenyx13, 2018-10-29:
I think I discovered an issue. When TerrariumPI loads up it toggles the switch to off.
I have a Wemo switch that I am not using for TerrariumPI but it is also toggled off.

theyosh, 2018-10-30:
Ah yes. That is correct :P That is/was currently default behavior. I have now changed that scanned devices will not change in state. But when the switch is loaded from the settings file it will be powered off. Because if the WeMo switch is in the settings file, you have saved the power switches and confirmed that it should be controlled by TerrariumPI.

Unfortunately there is no way to exclude WeMo switches during startup. That will add much more logic and interface screens.

theyosh, 2018-11-03:
Do you have an update if it is working better now?

theyosh, 2018-11-08:
Closing due to no response and should be fixed

closed 2018-11-08 by theyosh