theyosh/TerrariumPI #188: dht22 not working in terrariumpi but in shell comand

funware84, 2018-09-17:
i try to get my dht22 working.
in shell command it is sending temp and humidity. if i try to read it with terrariumpi it says 

 WARNING - terrariumSensor      - Measured value None% from humidity sensor 'terra humi' is outside valid range 0.00% - 100.00% in 12.63339 seconds.

i have an 4.7k resistor between 3.3v an data.

any solutions?

AndrejWeber, 2018-09-17:
Do you used the wrong pin-Number?
You have to use physical Number(1...40) in terrariumpi

swekley, 2018-09-17:
same problem here. i do use physical pin numbers even tried different pins but it keeps measuring value none. also bluetooth won't work but i'll make a new issue for that

AndrejWeber, 2018-09-17:
I had to supply my dht22 with 5VDC. The data sheet tells, you can only use about 10cm wires. The voltage drops to much, by using a long cable.

swekley, 2018-09-17:
> I had to supply my dht22 with 5VDC. The data sheet tells, you can only use about 10cm wires. The voltage drops to much, by using a long cable.

tried it on 5v now with no luck
the script takes about 7 seconds to give me the results, but in terrariumpi it still doesnt work

pi@raspberrypi:~/Adafruit_Python_DHT/examples $ sudo python AdafruitDHT.py 22 27
Temp=27.4*  Humidity=56.9%

2018-09-17 18:13:34,172 - WARNING - terrariumSensor      - Measured value NoneC from temperature sensor 'temp' is outside valid range 0.00C - 100.00C in 12.66952 seconds.

[image]

AndrejWeber, 2018-09-17:
Now I accidentally deleted my post 😂
I took a look in the data sheet. 100cm is possible with 3.3V

theyosh, 2018-09-17:
@swekley You are using sudo with testing. Does it work without sudo? If not, then there is a rights problem. Retry to run the installer. `sudo ./install.sh` . And yes the installer needs sudo

theyosh, 2018-09-17:
@funware84 did you test it also with sudo? Try without sudo. And in my software I use physical numbering, so that is a different number then when you are testing.

swekley, 2018-09-17:
> @swekley You are using sudo with testing. Does it work without sudo? If not, then there is a rights problem. Retry to run the installer. `sudo ./install.sh` . And yes the installer needs sudo

Yeah that worked, kinda.. Apparently i didn't install it with sudo.
temperature was working after a while, now i added humidity and its back at the same warning.

`2018-09-17 21:10:53,137 - WARNING - terrariumSensor      - Measured value None% from humidity sensor 'humidity' is outside valid range 0.00% - 100.00% in 12.66631 seconds.
2018-09-17 21:10:53,295 - WARNING - terrariumSensor      - Measured value NoneC from temperature sensor 'temp' is outside valid range 0.00C - 100.00C in 0.00791 seconds.`

[image]

I'll try a fresh installation tomorrow.

funware84, 2018-09-18:
thanks, i tried a new install, even if i had used the comand provieved by your install guide
now it is working, with 3.3v.

theyosh, 2018-09-18:
I had in the past that the installer is not that stable. So running it multiple times could help. But happy that it is working now.

theyosh, 2018-09-18:
@swekley  sounds like some mix-up with GPIO pins or shortage?

KineticShadow, 2018-10-10:
I've been having the same issues here with an AM2302 sensor. I'm on my 20-ish or so reinstall, various ways. For the most part I've done complete formats of the sd card and started from scratch. Most of the reinstalls were using the debian full, but I've switched to debian lite. What is really odd is that my command for the test:
sudo python3 AdafruitDHT.py 2302 4
Was working to execute the test in SSH. Somewhere along the line though, it started producing this error (with and without sudo):
Traceback (most recent call last):
  File "AdafruitDHT.py", line 24, in <module>
    import Adafruit_DHT
ImportError: No module named 'Adafruit_DHT'
But when I enter this:
sudo ./AdafruitDHT.py 2302 4
I get a readout, but it also takes seconds longer than it did before. When I use it w/o sudo, i recieve Failed to get reading, try again!

The power is feeding off of the pi, pins 1,7,9, Which I may not have a good grasp (nor do I find a defining image) of what physical pin means. My assumption is that it's the count from pin 1 back and forth (i.e. the ones I'm using). The dashboard reads without 'error' (while still not receiving any readings and sits at zero) only when I use 4 in the address. When I use 7, it reads both warning and error.

I could use all the help I can get. I can only say that the change in commands occured after I attempted to perform the install after using: sudo su. and even though I've done full formats of the sd card, the change remains.

theyosh, 2018-10-13:
@KineticShadow Only Raspbian Strecth Lite is supported. The full version is not tested, and other people reported to me that it does NOT work with the full Raspbian version. So use the lite version: https://github.com/theyosh/TerrariumPI#installation

Then the error 
`import Adafruit_DHT
ImportError: No module named 'Adafruit_DHT'`
This indicaties that the installer is not run correctly. So could you run the installer script once more as root. Repeat step 5 from: https://github.com/theyosh/TerrariumPI#installation

KineticShadow, 2018-10-13:
Thank @theyosh, I don't know if I was clear about it but yea, a good chunk of those installs were raspian full, but I had made the switch to raspian lite after reading this thread a few times (thick head when focused on other details.) Just to be clear; Run the TerrariumPI installer as root (sudo su, then run the file)? I will give it a try w/o formatting for the moment. Many thanks for the response.

theyosh, 2018-10-16:
@KineticShadow All the above is not helping and sounds like you have already problems with running the raspberry pi. So I want to make sure that you are able to just to install Raspbian Strecth Lite on a Pi with SSH. Nothing more!

Then only and ONLY run my installation steps! It will take care of the needed software. You are installing Python3 which I do not support. So please, start over with a clean PI and only execute the steps below as normal user pi.

1.     sudo apt -y install git
2.     git clone --recursive https://github.com/theyosh/TerrariumPI.git
3.     cd TerrariumPI
4.     sudo ./install.sh
5.     sudo reboot

KineticShadow, 2018-10-17:
performed as recommended.
Raspbian Stretch Lite image: 2018-10-09-raspbian-stretch-lite
Installed with Etcher on a clean SD card
Enabled SSH
ran the 5 commands
Added the two sensors to the interface
first attempt set them on 4 (for gpio 4). errored, no measurements.
second attempt set them on 7 (physical pin) errored, no measurements.
Ran sudo python AdafruitDHT.py 2302 4. successful read.
Ran python AdafruitDHT.py 2302 4. Failedto get reading. Try again!

theyosh, 2018-10-18:
Hmm, this sounds like a rights issue. I expect you are using the normal user 'pi' for installation. Correct?

Could you then past the output of `cat /etc/group | grep pi`:

KineticShadow, 2018-10-19:
pi:x:1000:
spi:x:999:pi
I am also trying to do the reinstall. 4 tries so far, same result.

fenyx13, 2018-10-20:
I also had this problem with AM2302 sensor.

theyosh, 2018-10-21:
Hmm somehow a reply off my did not make it here... So here is it again.

Could you start once more with a clean install? Which I mean just format the SD card and install my software according the above 5 steps. Then do the following and paste the output here:

`cat /etc/group | grep pi` (yes again)

`groupadd -f dialout`

`groupadd -f sispmctl`

`groupadd -f gpio`

`usermod -a -G dialout,sispmctl,gpio pi`

It depends on the outcome of the above commands how to proceed next. Do not try other things in the mean time. That makes debugging harder....

KineticShadow, 2018-10-21:
Clean install. Note: near the end, the screen splashed full red with output of the process scattered all over in lighter text.
Installer still finished.

Output of cat command:

pi:x:1000:
spi:x:999:pi

passed all groupadd commands.
usermod failed, assumed because of rights,
sudo usermod -a -G dialout,sispmctl,gpio pi went through.

theyosh, 2018-10-21:
This is strange. Did a full and clean re-install and found no issues. It is all working like expected.
But it looks like you are not installing it under user pi? Did you create a new user on the pi?

Cause with a clean install I would expect the following output of:

`cat /etc/group | grep pi`

`adm:x:4:pi`
`dialout:x:20:pi`
`cdrom:x:24:pi`
`sudo:x:27:pi`
`audio:x:29:pi`
`video:x:44:pi`
`plugdev:x:46:pi`
`games:x:60:pi`
`users:x:100:pi`
`input:x:101:pi`
`netdev:x:108:pi`
`pi:x:1000:`
`spi:x:999:pi`
`i2c:x:998:pi`
`gpio:x:997:pi`
`sispmctl:x:1001:pi`

And it you just re-run the installer? Will it again give a red scrambled screen at the end? I can't figure it out right now, as it works for me. So we need to figure out the differences...

KineticShadow, 2018-10-21:
Alright, didn't trust the last install so washed, rinsed, and repeated. 
Output of cat command the same.
passed all groupadd commands as sudo
passed usermod as sudo
additions to dashboard, once at 7 (physical pin) once at 4 (gpio) both error out.
examples test w/ sudo successful. w/o sudo failed. try again.

If it hasn't been said before, much appreciated for taking time to work this out. My knowledge of linux is just enough to get me into trouble and google the rest.

KineticShadow, 2018-10-22:
Sorry, I was grabbing the colon at the end of the first time you gave the command. Output is similar, I believe.

adm:x:4:pi
dialout:x:20:pi
cdrom:x:24:pi
sudo:x:27:pi
audio:x:29:pi
video:x:44:pi
plugdev:x:46:pi
games:x:60:pi
users:x:100:pi
input:x:101:pi
netdev:x:108:pi
pi:x:1000:
spi:x:999:pi
i2c:x:998:pi
gpio:x:997:pi
sispmctl:x:1001:pi

I have a Pi 2 B+, my wires are less than 20 cm to the sensor, using 1,7, and 9 physical pins. Everything else has been verbatim.

theyosh, 2018-10-22:
Hmm, this is not handy. 

> Sorry, I was grabbing the colon at the end of the first time you gave the command. Output is similar, I believe.

As you can see the output IS different, and it DOES matter! All my tests are not helping now, as the installer is doing its work. The user pi is in the groups gpio, sispmct and spi. As far as I know that should be enough to get it to work.

But now it is getting harder to fix this. As far I know Pi2 should work. But at the moment I am not able to test that. I use a Pi3 for testing. So I need some more time to figure this out.

theyosh, 2018-10-27:
What happen when you change from pin 7 to pin 11. And test once more. Else I have no other ideas. I cannot reproduce and there is nothing known about this.
If a different gpio pin does not make a difference, I am out of options. And the only option that is left is buying a Rpi3 and start with that.

fenyx13, 2018-10-27:
I just wanted to mention something in case it may be helpful.

I was using an AM2302 sensor I bought from Amazon on RPI3 and it worked but with many false readings. It died so I took the sensor apart and inside it actually says it is an AM2320.
It is not a DHT series so I am surprised it worked at all.
AM2320 data sheet: https://akizukidenshi.com/download/ds/aosong/AM2320.pdf

If you want to read about the different types of hygrometers, I found this link very informative: 
http://www.kandrsmith.org/RJS/Misc/Hygrometers/calib_many.html

KineticShadow, 2018-10-29:
Thanks for both. I will get on testing on other pins. I popped the shell, and isn't the 2302 (it's labelled) supposed to be built with a 4.7k resistor? More accurately questioned, I opened it up, a smaller white housing similar to the bigger one is soldered to a skimpy chip board, but I'm not seeing any sign of resistor. It would be outside of the sensor, correct? It just seems to be a stand in to mount the pinset to a clip-in wire set.

AndrejWeber, 2018-10-29:
Raspi 3b+ works for me. So, not only 3b will work.
I also had some Problems with this Senor. You should also test 
`python AdafruitDHT.py 22 4`
DHT22 and 2302 is nearly the same. I also can change it in TerrariumPi, without any difference in reading.

theyosh, 2018-10-30:
I am running and developing on a Rpi 3B. So that will also work.

For my software there it does not matter. Rpi 3B+ is a bit faster and has newer hardware.

theyosh, 2018-11-08:
Closing for now. As it looks like an hardware issue

closed 2018-11-08 by theyosh

stuartabrown, 2019-01-01:
Hi.  I am also having this exact issue on an RPi3B.

I have been through the steps above `cat /etc/group | grep pi` shows:
`adm:x:4:pi
dialout:x:20:pi
cdrom:x:24:pi
sudo:x:27:pi
audio:x:29:pi
video:x:44:pi
plugdev:x:46:pi
games:x:60:pi
users:x:100:pi
input:x:101:pi
netdev:x:108:pi
pi:x:1000:
spi:x:999:pi
i2c:x:998:pi
gpio:x:997:pi
sispmctl:x:1001:pi`

I have run all the `groupadd` and `usermod` commands (latter had to be run under sudo) but doesn't seem to make a difference.  The log shows me e.g. `2019-01-01 20:32:21,811 - WARNING - terrariumSensor      - Measured value NoneC from temperature sensor 'DHT22 Temp' is outside valid range 5.00C - 55.00C in 12.63418 seconds.`  However if I run `python AdafruitDHT.py 22 4` temp and humidity print out no problem.