theyosh/TerrariumPI #191: Audio startup issue

fenyx13, 2018-10-10:
Hi theyosh,
Is there a way to disable the features I am not using for TerrariumPI?
My camera works but not the audio. I have no need for the audio feature and it is causing me error on startup of TerrariumPI.
Another example would be disable scanning for MiFlora Bluetooth devices. I do not have one.
Thank you for your time.

fenyx13, 2018-10-14:
This is the error I receive when I try to load TerrariumPI with a PS Eye Camera connected by USB. I think the error is because of the microphone type on the camera.

2018-10-13 21:35:17,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.0.31:8090 ...
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 22, in <module>
    terrariumEngine = terrariumEngine()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 136, in __init__
    self.get_audio_playing)
  File "/home/pi/TerrariumPI/terrariumAudio.py", line 33, in __init__
    hwcards = terrariumAudioPlayer.get_sound_cards()
  File "/home/pi/TerrariumPI/terrariumAudio.py", line 138, in get_sound_cards
    if 'PCM' in alsaaudio.mixers(**{'cardindex': i}):
alsaaudio.ALSAAudioError: Invalid argument [hw:1]
2018-10-13 21:35:37,000 - INFO    - terrariumWrapper     - Restart counter: 1/5
2018-10-13 21:35:37,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 20 seconds. Press Ctrl+C now to terminate TerrariumPI.

fenyx13, 2018-10-14:
My apologies for repeat posts. I am sorry.

I finally got the PS3 eye camera working properly.
I have changed so many setting that I am not sure what one fixed the issue.
I will do a fresh install in a few days and report back which change fixed this for me.

My guess, is that, none of the changes worked but then:
I started TerrariumPI with no camera
then after starting TerrariumPI
I plugged in the camera to USB

I will report back soon with the solution.

theyosh, 2018-10-16:
Hmmm, this is new to me. I have to think about it a bit. And I think I do need some more information from you later on. Maybe some testing...  But it should be able to work / able to detect properly.

theyosh, 2018-10-16:
Update: latest USB related webcam errors should be fixed. See #192 

And for the audio part. Could you try the following for me. It involves some python codes and commands.
Als normal Pi user start python:

`python`

And you should see something like

`Python 2.7.13 (default, Sep 26 2018, 18:42:22) 
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>>`

Then enter the following code:
`import alsaaudio`
`print alsaaudio.card_indexes()`
And then you should see something line: `[0,1]`
For every number above do the following:
`print alsaaudio.mixers(0)`
Where 0 is the used number. The output would look like:
`[u'PCM', u'Mic', u'Auto Gain Control']`

If this problem is not there anymore, then it was a software/hardware glitch...

Then copy all the output and past it here.

fenyx13, 2018-10-17:
```
pi@raspberrypi:~ $ python
Python 2.7.13 (default, Sep 26 2018, 18:42:22) 
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> 
>>> import alsaaudio
>>> print alsaaudio.card_indexes()
[0, 1]
>>> print alsaaudio.mixers(0)
[u'PCM']
>>> print alsaaudio.mixers(1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
alsaaudio.ALSAAudioError: Invalid argument [hw:1]

pi@raspberrypi:~ $ alsamixer
cannot load mixer controls: Invalid argument
```

ALSA does not like the 4 channel microphone used by the ps eye camera. 
When I do not have the camera attached I can load and run TerrariumPI then plug in the camera and it works but on restart it tries to load camera and then the audio error. 
I realize this is an issue with my hardware.

theyosh, 2018-10-17:
I made another update, which should take care of this. The audio of your eye camera will/should now be ignored during startup. You can update by doing a `git pull` in the TerrariumPI folder

fenyx13, 2018-10-17:
Issue solved! Thank you very much!!

theyosh, 2018-10-18:
Ok then I close this. There is still issue #192 that could also affect you.

closed 2018-10-18 by theyosh