theyosh/TerrariumPI #192: USB Webcam doesnt Work

Marvv90, 2018-10-10:
The USB Webcam „dev/video0“ work‘s only for seconds on the startup from the Raspberry PI 3 !

After the Startup the Image from the Webcam doesnt Refresh or the Webcam work ! 

Log-File:

2018-10-10 20:11:00,623 - INFO    - terrariumWebcam      - Done updating webcam 'Video0' at location /dev/video0 in 9.11650 seconds
2018-10-10 20:10:59,621 - WARNING - terrariumWebcam      - Attempt 3 of 3 for getting raw for usb type 'Video0' did not succeed at location /dev/video0. Will retry in 1 second.
2018-10-10 20:10:56,583 - WARNING - terrariumWebcam      - Attempt 2 of 3 for getting raw for usb type 'Video0' did not succeed at location /dev/video0. Will retry in 1 second.

Attempt 1 of 3 is the same !

I have tried different Webcams ! All the same!

After every Reboot / Shutdown the same failure! 

Sorry for my broken english!

Best Regards 

Marvv

theyosh, 2018-10-16:
Hi @Marvv90 

I also have those errors, but did not pay that much attention to it. I was thinking it was a webcam issue and expected to work correctly. But as you have tried multiple webcams, it sounds like a software issue.

So I will investigate this and see what I can find

theyosh, 2018-10-16:
Should be working now. Downside is that the webcam is continuously running (red light on) and therefore a bit more power consuming.

Opening and closing/releasing the webcam did not work properly.

Marvv90, 2018-10-17:
And when i try to Change the Settings from the Camera then crash the Webcam with this Code:

libv4l2: error setting pixformat: Device or resource busy
HIGHGUI ERROR: libv4l unable to ioctl S_FMT
libv4l2: error setting pixformat: Device or resource busy
libv4l1: error setting pixformat: Device or resource busy
HIGHGUI ERROR: libv4l unable to ioctl VIDIOCSPICT

And Only a Restart from the System helps.

Only a Info: This Webcams works properly with "Motion"

theyosh, 2018-10-17:
Damm, you are right. I also see the delay here. Did not test that.... And changing settings could also be an issue now. So I have to rethink this over.... :( Will unfortunately take some more time

theyosh, 2018-10-23:
Yes! found the bug. It should now working like expected. Due to threading to run more processes in parallel the usb webcam got stuck. Now the threading is at a higher level and that works much better.

You can change settings without issues and the camera will take proper shots.

Marvv90, 2018-10-23:
i test it tomorrow early, and report to you !

Marvv90, 2018-10-24:
Yeah it work properly !!!! Good Job ! 👍

closed 2018-10-24 by Marvv90