theyosh/TerrariumPI #193: Notifications sent to LCD attached to arduino

rrgrow, 2018-10-11:
Hi theyosh

In Notifications section where I2C address is, 
would it be possible have a option to send  data to a LCD display attached to a arduino via USB? 
I will write the Sketch file for ardiuno. 
Link below shows a way below:
https://www.instructables.com/id/Raspberry-Pi-Arduino-LCD-Screen/

theyosh, 2018-10-16:
Hmm, I think this can be done, but then we need to agree that the same code is used as on the page you added. Because we both need to know how to specify for example the line nr to write to.

`String Format: ##String`

And I will probably not support dimming with PWM. Because I cannot test it, so that will be hard to make by guessing the code...

theyosh, 2018-10-16:
Just added the code :)

Use as input /dev/ttyXYC,9600 (or other serial speed)

I used the same logic as on https://www.instructables.com/id/Raspberry-Pi-Arduino-LCD-Screen/

This is all untested

rrgrow, 2018-10-21:
I see in ur the code, if start with /dev/.. I then complete the address but the page doesn't save so I tested with other address formats I2C address etc still not saving I then and also put random numbers in other notification fields on the page and it saved but only without the I2C address field empty.. my serial resides /dev/ttyUSB0
Thanks for looking at this..

theyosh, 2018-10-22:
I made an update. Could you test it? I was able to save the notification settings page and it looks that it is loading. But I get errors due to (hopefully) missing hardware.

rrgrow, 2018-10-23:
Thanks theyosh I will Test this when I’m back home from working away.

theyosh, 2018-10-28:
Any updates on this?

rrgrow, 2018-10-31:
> update:

the results i'm getting are the complete warning string with date and time etc..

`[2018-10-31 22:27:54,171 - WARNING - terrariumSensor      - Measured value 11472.0ppm from co2 sensor 'Co2' is outside valid range 0.00ppm - 3000.00ppm in 0.03119 seconds.
`

I have tried editing all fields in notification from %raw_data% to ie "on", "off" but all I see is the complete info displayed on 20,4 LCD on 4 lines but at different stages (I understand different names and warning info lengths). I have no way of customising.
I haven't used the feature on the pi direct as I have no easy access hence I use the arduino mega for addons, so its hard to tell if i'm doing something wrong here but from what I can see, the data coming down the serial is full complete details like listed above.

theyosh, 2018-11-01:
Hmmm, not sure what you mean. But system errors and warnings can not be customized. They can only be turned on and off for the display part.

But this is how it should look like : https://theyosh.nl/speeltuin/TerrariumPI_LCD_Demo.mp4

rrgrow, 2018-11-02:
https://1drv.ms/v/s!AhNwmEADFEappQJj2RAuIaF3Si4T

rrgrow, 2018-11-03:
ok I see whats happening.. for some reason the first 2 characters are being missed and the pos isnt being read and then displayed on LCD  "       01TerrariumPI".  I'm working to sort this. I have had it working with the standard sketch like on link above but only on lines 1,2,3. line 0 does not show up till data spill from another line.

theyosh, 2018-11-03:
Ah, I see that I start with line nr 1 and not 0. So I made an update. Do a `git pull` and restart. Now it should be better.

theyosh, 2018-11-10:
Any updates about this?

theyosh, 2018-11-16:
Closing due to no response. If issue is not fixed, please re-open

rrgrow, 2018-12-20:
https://1drv.ms/v/s!AhNwmEADFEapvjshmZpldmNTbgXn

rrgrow, 2018-12-23:
The problem was the 'Delay' command in the loop of the sketch. I had to rearrange the code so the loop had no delay commands at all. And where sensors need resting time I used the minis command instead.. heres the link to the arduino sketch 
https://1drv.ms/u/s!AvHCK7bGGTJjh2HcqmF9Hv8dYIQf

theyosh, 2018-12-23:
Hmm, I could not see the issue on the movie. But this feature has also be made without having the supported hardware. I do not develop on an arduino. But is it working now like you want?

Then I could add the arduino sketch to the contrib folder for other users.

rrgrow, 2018-12-27:
It is working for me as I want it and your more then welcome to use my sketch if other want to use the 2 way option ie sensors attached to the arduino and display on LCD but stess the delay (); command is not to be used in the void loop just millis(); command instead you'll see in the sketch I posted above.

rrgrow, 2018-12-27:
Turns out to be:
posY = posY * 8.666;
The settings in Terrarium stay the same ie 20, 4 the actual size on OLED at 0.5 is 21, 4

I have split the sketches into 4 Example files.
Here's the Zip:
https://1drv.ms/f/s!AvHCK7bGGTJjh2YyWhGwIXRRKb9i

rrgrow, 2018-12-27:
Turns out to be:
posY = posY * 8.666;
The settings in Terrarium stay the same ie 20, 4 
The characters on OLED work out to 21, 4 and I Do get a '0' on the 21st character sticking around..I think this is the setting are set to 20,4 in Terrarium maybe a easy fix from ur end?

I have split the sketches into 4 Example files.
Here's the Zip:
https://1drv.ms/f/s!AvHCK7bGGTJjh2YyWhGwIXRRKb9i

rrgrow, 2018-12-28:
[image]

theyosh, 2019-01-13:
Hi,

I had to change the code for this serial LCD due to other logic in drawing the text. It would be nice if you can test this.

To test it, check out the `oled` branch and rerun the installer. Then restart TerrariumPI software. It should auto convert the configuration for your serial device

theyosh, 2019-01-14:
At the moment, it is already in the master branch. So just normal updating should give you the new code to test

rrgrow, 2019-01-19:
While testing Display.
Hardware (Section) lost choices in drop down and cant seem to get them back.

theyosh, 2019-01-20:
Ah, that is not correct. New code should fix it. Do a `git pull` and restart TerrariumPI. That should give you a list of hardware devices

rrgrow, 2019-01-24:
This works for me remotely but if you make a mistake on the hardware choice, on next boot the server errors till you change the details in Notification.cfg . I also noticed this happening when adding sensors (Mainly 'Remote Hardware' JSON) that come back with 'warning' reading ie 'outside range' which is fine if you know how to edit settings.cfg to yet thing back up and running. I have also Noticed when changing Distance to Volume in sensor settings its not possible till you delete and add again with the change to volume then it works.  
The Display works as before but still has a 0 on the last character on line 0. What command do you send to clear line? just out of intervest.
Cheers

theyosh, 2019-01-27:
Hmm not sure about that zero. I have made some changes to tackle wrong hardware settings for displays. So at least it will start now. And then you are able to change it through the interface. Only thing that will not work yet, is reloading the display settings after saving. So a restart is still needed..

But about that clearing of the screen. As far I know, I do not do that. I 'clear' by always make a text string for WIDTH chars using a space as filler. So every time a new line is written, all the previous text should be overwritten by spaces.

But I use a double digit number for lines, so could it be that you just read one char to much? Could you test it with the latest `git pull`?

I will also address the JSON errors, but that will be done later on

theyosh, 2019-01-27:
Wrong remote JSON urls in sensors will not crash the software. And when you watch the sensor, it will not show any values and after 10 minutes an error icon. That should trigger a response of looking into the logging (either trough the web or ssh) and there you can see that there is an error. At that point you are still able to change the http address in the sensor settings to correct the issue. So that is acting as should and therefore I will not change that.

theyosh, 2019-01-27:
And changing a sensor type is not allowed. As that will give issues with generating the right sensors ids when sensors can be added as multiple types. Like MiFlora or Chirp sensors. So that is also working like expected. I do admit it is not the most elegant, but again, how often does a sensor type change? So that will also stay as is.

But thanks for giving this feedback. I hope that my explanations are clear.

theyosh, 2019-02-09:
Is still on going? As far I can see there is nothing to fix for me. And I cannot fix arduino code

closed 2019-07-05 by theyosh

rrgrow, 2019-09-20:
I have this working with the sketch below for OLED:

/* Sketch Start:
   Terrarium Adafruit_SSD1306 Display from Pi serial(USB)
   Display Setting in Terrarium Notification Page works fine on LCD 20,4).
*/

#include <Adafruit_SSD1306.h>
#define OLED_RESET 4 // not used / nicht genutzt bei diesem Display
Adafruit_SSD1306 display(OLED_RESET);

// Terrarium Display Adafruit_SSD1306
String input = "";
String stringRec = "";
String pos = "";
int posX = 0;
int posY = 0;

void setup()   {

  // Terrarium Serial Display Adafruit_SSD1306
  // initialize with the I2C addr 0x3C / mit I2C-Adresse 0x3c initialisieren
  display.begin(SSD1306_SWITCHCAPVCC, 0x3C);
  Serial.begin(9600);
  display.clearDisplay();
  display.setTextSize(0.5);
  display.setTextColor(INVERSE);
}

void loop()
{
  // Terrarium Serial Display Adafruit_SSD1306
  if (Serial.available() > 0) {
    input = Serial.readString();
    pos = input.substring(0, 2);
    posX = pos.substring(0, 1).toInt();
    posY = pos.substring(1, 2).toInt();
    stringRec = input.substring(2);
    posY = posY * 8.6666; //Changed to fit the OLED display with LCD 20,4 setting
    display.setCursor(posX, posY);
    display.setTextColor(WHITE, BLACK);
    display.print(stringRec);
    display.display();
  }
}
//END OF SKETCH

And working with LCD 20,4 this sketch below:

/* Sketch Start:
   Terrarium LiquidCrystal Display from Pi serial(USB)
   Display Setting in Terrarium Notiforcations Page works fine on LCD 20,4).
*/

// LiquidCrystal
#include <LiquidCrystal_I2C.h>
LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);

// Terrarium Display LiquidCrystal
String input = "";
String stringRec = "";
String pos = "";
int posX = 0;
int posY = 0;

void setup()   {
  // Serial
  Serial.begin (9600);
  // Terrarium Serial Display LiquidCrystal
  // LiquidCrystal
  lcd.begin(20, 4);
  lcd.clear();
}

void loop()
{
  // Terrarium Serial Display on LiquidCrystal
  if (Serial.available() > 0) {
    input = Serial.readString();
    pos = input.substring(0, 2);
    posX = pos.substring(0, 1).toInt();
    posY = pos.substring(1, 2).toInt();
    stringRec = input.substring(2);
    lcd.setCursor(posX, posY);
    lcd.print(stringRec);
  }
}
//END OF SKETCH