theyosh/TerrariumPI #194: Full Authentication Bug with remote function not able to read json file

rrgrow, 2018-10-21:
Hi theyosh
After changing "Changes only" to "Full Authentication" remote collect for data from the json file stops working.
After changing back to "Changes Only" it works a again.
Thought you should know, 
I suspect another bug in System environment not saving, no loading bar across top. I'm thinking another system setting could be stopping this from happening but I haven't gone through the setting one by one.
Hope this helps.

theyosh, 2018-10-22:
Hi,

if your remote data is in the /static/extern/ folder, this is correct. That path is also password protected then.

The sensor can not know this, as it is being treated as just an external source somewhere on the network.

The only solution I can think off, is that the location /static/extern/ will not be protected at all.

rrgrow, 2018-10-23:
As I’m not at home atm was able to VNC to the pi, update and install after which the full Authentication and remote bows works fine. Many thanks

closed 2018-10-27 by theyosh