theyosh/TerrariumPI #197: Install TerrariumPI Error

Marvv90, 2018-10-24:
Hi theyosh , me again ^^,

Today I have completely reinstalled my SD Card,  and by the first install Routine from TerrariumPI the screen coloured red with many Error's. Only a second install from TerrariumPI after the first solved the many Error's.

Hard to explain. I hope you know what i mean !

During bottle_websocket install the failure occurred.

Best Regards ! And many thanks for your Work !

theyosh, 2018-10-27:
Yeah... this is a known issue. Rerun the installer and it should be good.

Problem is that I never encounter these issues when I test it on a clean Pi. Often an issue with network connections with downloading the python libraries.

It has a low priority for know.

theyosh, 2018-10-28:
I added a known issue page for this: https://github.com/theyosh/TerrariumPI/wiki/Known-issues#installer

closed 2018-10-28 by theyosh