theyosh/TerrariumPI #198: Litre option in setting along side cm and inch

rrgrow, 2018-10-28:
I already have cal my distance in arduino to output in Litres and was wondering if you could put a Litre option in settings inside cm and inch field.
Thanks

theyosh, 2018-11-01:
Hmm, how could the pi read out this information? Is the Arduino providing some kind of HTTP server as a remote sensor?

Else, it is usb/serial? That is a but harder to add.

rrgrow, 2018-11-02:
yeah the data comes from the arduino #124 to a *.json file. and using the remote function. I thought just adding L to the drop down cm, inches in settings to display in litres.

theyosh, 2018-11-03:
Made a new sensor type. Now you can add volume sensors in Liters. Also, these volume sensors can be used with the water tank environment settings.

rrgrow, 2018-11-11:
Tested and Works a charm! I also like the nav update with the sensors only showing sensors used.. much easier to nav. Good Work!

theyosh, 2018-11-11:
The height info is still needed, but not used. It is a bit harder to make those fields variable in requirement when selecting sensors. Hence, just fill in some values.....

The volume in environment is not there. Is figured that it would be used as the water tank source. So what I did is adding volume sensors in the same way as height. But on the environment there is only the water tank that can use it. There is also no height environment part for height sensors...

For the CO2 environment part, do you get an error somewhere? Could help debug it.

theyosh, 2018-11-11:
Found it... should not also be fixed. Clear cache as there is a .js file update which are cached in the browser.

theyosh, 2018-11-16:
Closing due to main issue is fixed. If the other issues are still there, please open a new issue

closed 2018-11-16 by theyosh

rrgrow, 2019-09-20:
This works
Many Thanks!