theyosh/TerrariumPI #199: Energenie Pi-mote socket board

EoinH, 2018-10-30:
Hi,

Would it be possible to add support for the Energenie pi-mote & 4-way RF extension lead?
https://energenie4u.co.uk/catalogue/product/ENER010-PI

The GPIO Zero library  added built-in support for Energenie sockets so hopefully it's easy to incorporate into your code!

Thanks,
H.

theyosh, 2018-10-30:
I think this is possible. Only I cannot test it. So if you have the hardware, I can add code and you have to test it. Is this possible?

Also, the code is never possible to readout the state or listen to the RF remote for signals.

EoinH, 2018-11-01:
I have the hardware already so I would be happy to test for you!  And checking the state or that is not necessary, just switching the sockets.

Much thanks,
H.

theyosh, 2018-11-01:
Ok done :)

Just chose for Energenie Pi-Mote and enter the number os the socket. That should do the trick

EoinH, 2018-11-06:
Great news - the switches work perfectly!  
I've been able to set up all 4 switches and put two of them on timers.
I had some suggestions for improvements to the software also - is that something you are interested in? and if so should I start a new thread?

Thanks again for the updates!
H.

theyosh, 2018-11-06:
Cool! For new features please open an issue per feature. That makes the communication a lot easier.

closed 2018-11-06 by theyosh