theyosh/TerrariumPI #200: Can I use it ?!

Marvv90, 2018-10-30:
Hi theyosh,

Can I use the Plugin pigpiod for Switches for Dimmer ? Command pigs p PIN 0-255 ?! I will dimm a led strip like this https://dordnung.de/raspberrypi-ledstrip/ ! With the Command pigs p 22 255 or 0 works fine and I see you use pigpiod !

I Hope you know what I mean

theyosh, 2018-11-01:
Ah, that sounds nice. I have seen the page, but there is also discussed about changing colors. That will not be supported.

Also, I am not sure if you can call this dimming. Because it is more like color changing. as 50% of pin 22 is somewhat yellow, where 100% is green. So I am not sure if this will give the desired effect.

But have you tested this at home? We can try to add this to the software.

Marvv90, 2018-11-20:
Hi theyosh, 

Sorry for my late Reply , Yes i have it tested at home & we can try it ! 

Best Regards

theyosh, 2018-11-20:
Yeah, that is what I thought and makes it a bit harder to support. Cause I need to control three GPIO ports. That can be done. But then I have no way to specify colors. And when I cycle from 0 - 255 it will not get brighter, but I think you get a some kind rainbow effect of all kind of colors.

So, I will need some more information how you dim it. Do you first set the color, and than dim just one GPIO port? So am I right that that only 1 GPIO pin will only do one color. And that you need three colors / GPIO pins?

theyosh, 2018-11-21:
Hmm I think I got an idea for this.... So... I will try to make some code that you then need to test for me... It will not be able to control the colors. It will go from black to full white. And that with a dimming effect

AndrejWeber, 2018-11-21:
Hey Yosh,
The RGB LEDs should be easy to dim and set the Color. 
You made the DC Dimmer for me can also dim the leds.
You need 3 PWM signals with a resolution of 256 Steps
White is always when all 3 LEDs are working with same power

So when all three dimmer are set to 1% or 1/255 you have white in low dimming

All three dimmer at 100% or 255 is white at full power. 

Orange for example is Red:255, Green:165, and Blue:0
For testing with existing DC Dimmer:
Set red gpio to 25.5%
Green gpio to 16.5%
And blue to 0%

Now it is about 1/4 of full power orange
Max orange is at red 100%, green 74%, blue 0%

There are RGB calculator to look for the Color code.

AndrejWeber, 2018-11-21:
You can also use a 4th Dimmer for dimming the whole brightness 
By dividing the 255 states from 0% to 100%
Do not care about 255, it is only a common way to specify a RGB Code
You can use 25,5 as 100% or anything else

AndrejWeber, 2018-12-08:
I do not use it at all for Terrarium. I only know, how LEDs are dimmed by PWM. Most of the rgb-LED driver use PWM signal to dim and set the colour. It is fast switching on/off instead of change the current for each LED.
4 GPIOs are not needed. Only 3 (one physical for each colour) and one software-Dimmer to dim all together by multiplying each state. 
There are also RGBW or RGBwW LED-Stripes. They have a white LED as well. There you need 4 physical and a software-dimmer. I can not test it here, because I have not enough MOSFETs at home to test my stripes (under the bed) with 12V Input.

theyosh, 2018-12-09:
Then, I think I have an option right now. As @AndrejWeber says that the DC dimmer should work for the leds, you can now enter multiple gpio pins as address in the DC dimmer. So you can now enter `23,24,26` for example, to specify the three gpio pins.

The dimmer will now just go from 0 - 100% on all three the pins. This will make it go from 'of' to full bright. Not sure how the coloring will do during that period and at different values.

theyosh, 2018-12-28:
@Marvv90 Hi, are you able to test this? It is in the current released code

ese23, 2019-01-26:
Hello, i have a question. I read AndrejWeber issue with PWM dc dimmer. And read a lot of freq of these PWM modulators :-)
This module its prety cheap, but the PWM Receiver Frequency is 1KHZ-3KHZ, is it compatible with PWM generatet by RPi?

[https://www.aliexpress.com/item/PWM-to-Voltage-Converter-Module-0-100-to-0-10V-for-PLC-MCU-Digital-to-Analog/32834921874.html](url)

I want use it for diming Osram QUICKTRONIC DIM T8.

Its my wet dream, to dimm fluorescent light, with weather info from real destination (wet-aquarium) :-D

PS new rapsPI model 3 is running, no errors, only little issue with setup weather. Waiting for relay and sensors.

AndrejWeber, 2019-01-26:
Hey @ese23,
Dimming quictronic is possible with PWM, and changing the PWM frequency is also very simple in the code. 
I am not shure, if the logic will work with 3.3V of the Raspi. There is written something of 4.5 to 10V logic. 
But it is very cheap, so it is not a bad idea to test it. 
I made my own converter. 
I have access to any resistor and capacitor I need at workplace, so it was quite simple to do it at my own.
If it will not work with this hardware, I can also make a drawing, what I have used.

theyosh, 2019-07-05:
Closing as there is not much that I can do at the moment.... still a bit difficult to understand PWM dimming ;)

closed 2019-07-05 by theyosh