theyosh/TerrariumPI #201: Add support for SHT31-D sensor

lonelybrewer, 2018-10-31:
Relevant material:
[Sensor info](https://learn.adafruit.com/adafruit-sht31-d-temperature-and-humidity-sensor-breakout/overview)
[Datasheet](https://cdn-shop.adafruit.com/product-files/2857/Sensirion_Humidity_SHT3x_Datasheet_digital-767294.pdf)

I have the sensor connected and can successfully read data from it.

theyosh, 2018-11-01:
Just added. But I cannot test it, so if you can test it that would be nice.

Make sure you do a git pull to get the latest code. Rerun the installer to make sure you will have the new needed python libraries.

Then restart: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-restart-terrariumpi

Clear your browser cache to make sure you will get the new sensor settings page. There you can add a new SHT3X sensor. Temperature and humidity is supported. Enter the I2C address as seen by i2cdetect command

lonelybrewer, 2018-11-02:
First of all, thank you so much!

I have updated and am now trying to test it out but I am getting the following error:

2018-11-01 18:07:50,931 - ERROR   - terrariumSensor      - global name 'SHT31' is not defined
NameError: global name 'SHT31' is not defined
self.__bus = SHT31(self.__address)
File "/home/pi/TerrariumPI/terrariumI2CSensor.py", line 640, in __enter__
with hardwaresensor as sensor:
File "/home/pi/TerrariumPI/terrariumSensor.py", line 383, in update
Traceback (most recent call last):

I don't know anything about programming, but isn't SHT31 defined as 'class SHT31(object):' on line 53 of the Adafruit_SHT31 library you have imported in terrariumI2CSensor.py?  I can't figure it out.

lonelybrewer, 2018-11-02:
So I got it working by simply pasting the entire contents of Adafruit_SHT31.py into terrariumI2CSensor.py.

I'm assuming this is not ideal.

theyosh, 2018-11-02:
Hmm, this is now fixed. Do another git pull and restart. Make sure that the file terrariumI2CSensor.py is updated. If you manually changed it, you can do a `git checkout terrariumI2CSensor.py` first and then `git pull`

lonelybrewer, 2018-11-03:
Works great!  Thanks again.

closed 2018-11-03 by lonelybrewer