theyosh/TerrariumPI #202: Multiple Environments

Natrixz, 2018-11-02:
Hey Yosh,

This is more of a feature request;
Are there any plans to make this support more than one environment?
Saves me from having to buy 3 pi's, 3 powersockets, etc :)

I understand it's not that easy to create that, but I figured I'll ask it anyway!

theyosh, 2018-11-06:
Hmm, in the past more people have asked for this. And I did some code rewrite to make somewhat a start to support this. But it takes a lot of work to make everything variable.

So I keep it somewhat on the background. But I am still rewriting parts of code to make it possible to control multiple environments. But there is no ETA for this. And there is also a change that it will never made if fully.

And I have had already issues with the duration of reading out all sensors. So, when multiple environments are supported, the issue that there are to much sensors could arise. And that is something I would rather avoid.

Natrixz, 2018-11-06:
Yes I do understand those points.
And I could also understand if it never really makes it.

It does add an extra layer of complexity with both adding new code, bugs, sensors etc.
I don't mind if you close this "issue" or keep it open if you like.
For the time beeing I will just add another Pi =)

cyber-shifter, 2020-09-09:
I would like to second this request. Just to show further interest in getting it.

theyosh, 2021-04-10:
[image]

Yesh!!! Party time... Finally after a few years.... it is here!! If you have time, could you have time to test this: #539