theyosh/TerrariumPI #203: Add door state option to the webcam archive configuration

h1kkan, 2018-11-06:
Hi @theyosh !
Could you please add "door state" param to the webcam archiving function? Thanks!

theyosh, 2018-11-06:
What should it do? If the door is open, then do not make archive images?

There are a lot of requests for all kind of triggers when the door opens. That rather hard to develop and will not be made. So taking a webcam image for the archive when the doors opens will **not** be added

h1kkan, 2018-11-06:
> If the door is open, then do not make archive images?

In my case yes, that is what it should do. It should just depend on door state, and shouldn't be used as trigger at all.

theyosh, 2018-11-12:
It is testable in the python3 branch for now. That will become the new master when everything is tested on python3.
 
If you checkout python3 branch, make sure you run the installer. A lot of code had to change to make the switch to python3 easier. The installer will take 10+ minutes. And could fail the first time. Just rerun it.

h1kkan, 2018-11-22:
I'll check and test it all on this weekend

h1kkan, 2018-12-03:
Hi @theyosh, i've tested cam archiving with door sensors for a week and no errors occurred.

But i've faced with issue during update. When installer stuck for the first time, `ps` showed that apt process is still active, so i've checked `/var/log/apt/term.log` log and was able to identify why it hang - APT was waiting on interactive action from user, but this _question_ wasn't displayed on installer screen:
```
Setting up pi-greeter (0.9) ...

Configuration file '/etc/lightdm/pi-greeter.conf'
 ==> Modified (by you or by a script) since installation.
 ==> Package distributor has shipped an updated version.
   What would you like to do about it ?  Your options are:
    Y or I  : install the package maintainer's version
    N or O  : keep your currently-installed version
      D     : show the differences between the versions
      Z     : start a shell to examine the situation
 The default action is to keep your current version.
*** pi-greeter.conf (Y/I/N/O/D/Z) [default=N] ?
```

So i've killed apt and installer processes, manually updated software with `apt-get -y full-upgrade`, then ran installer again and got app installed in a 5 min.

theyosh, 2018-12-03:
Hmm, this is strange, but cant do anything about it. Somehow system issue. So then I would like to close this issue for now due to initial request is done.

Thanks for the heads-up about the installer.

theyosh, 2018-12-03:
Oh.. forgot to mention... please read https://github.com/theyosh/TerrariumPI/issues/210 :)

closed 2018-12-03 by theyosh