theyosh/TerrariumPI #204: Push button (or any other type of button) to activate a switch

Natrixz, 2018-11-09:
Would it be possible to add some sort of push-type style button or some other button we can connect to the Raspberry Pi?
Either with a timer and/or a push to enable and push to disable?

A couple of my snakes need to be fed at night, so to do this I enable a red-night light in the enclosures and then feed them.
Would be nice if it's possible to do this with a switch I can attach somewhere near the enclosures.

theyosh, 2018-11-12:
hmm, I get the idea. But I am not sure how to implement. There are more ways.

But for now, I see already two solution.

1. Use a WeMo power switch. It can be controlled by TerrariumPI and it has a physical switch on the device. That means the WeMo should be accessible.

2 Use an old mobile with the web interface.

But I will see if I can think something out. Features:
- Adding 1 or more push buttons
- How long will the push button do it's action. (Timer)
- Stop the timer when pressed for a second time
- Does long holding trigger a different action??

So.. it will take some time. And need to figure out a way that work for multiple ways.

theyosh, 2018-11-16:
This is also an option: https://energenie4u.co.uk/catalogue/product/ENER010-PI Only not sure if there are with european connectors

Natrixz, 2018-11-16:
I have been looking at that as well, I can only find the single plug in European plug:(

theyosh, 2018-11-16:
how about..... : https://www.conrad.nl/p/kabel-digitus-computer-stroom-apparaat-1x-engelse-stekker-1x-apparaatstekker-female-c13-10a-18-m-zwart-1556217 :)

I think that is workable...

Natrixz, 2018-11-16:
Smart/good find!
I actually didn't search that way yet, nice =)

That would work yes

theyosh, 2018-11-16:
like some challenge.. but I am just wondering that the RF module will take the hole pi. And you cannot add sensors.. So still it is all thinking out of the box...

I will not be accountable for wrong purchases :)

Natrixz, 2018-11-16:
Yes im sort of wondering that too....as well because I use the 1-wire module for all my sensors readings =)
Hmmm, let me think about it a bit more, maybe im able to think up something else :)

I did stumble upon https://flic.io/ 
But 35$ for a BT button....meh, not sure if that's a good deal :P

ese23, 2019-01-26:
I buy this kind of relay board.
[https://www.amazon.com/MagiDeal-Channel-Infrared-wireless-Control/dp/B06XS7FZGM](url)
And if i have luck it will work. In description say, that you can independently switch relays, either with IR or GPIO from RAPI. Soo its little overoverride :-) but many users, that buy it, say it didnt work :-/
Soo next weak i will test it.

jornobe, 2019-03-10:
Have added a hardware switch on pin 18 to trigger outlet 1 on the Energenie LAN (already controlled by TerrariumPI) whenever pushed.

The Energenie LAN outlet has a hardware schedule set to switch it off every 30 seconds. Was done already before to make sure that the outlet would not be kept on for whatever reason (e.g. communication problems between PI and plugs). **If you use another type of plug that does not support this type of settings, then a sleep timer and turn off command is needed in the  rain_on.sh script...**

4 steps:

- Switch connected to pin 18 and a ground pin

- Created this script (rain_on.sh) to login to the Energenie, switch the outlet on and logoff the management webpage again: (you can replace this with whatever code works for your plug)

```
#/bin/bash
wget --post-data 'pw=pwd' 'http://192.168.178.45/login.html' -O foo.html --delete-after
wget --post-data 'cte1=1&cte2=&cte3=&cte4=' 'http://192.168.178.45' -O foo.html --delete-after
wget 'http://192.168.178.45/login.html' -O foo.html –-delete-after
```

- Created this Python script (rain_check.py) to monitor the pin and launch the script when button gets pressed:

```
import RPi.GPIO as GPIO
import time
import os
GPIO.setmode(GPIO.BCM)
GPIO.setup(18, GPIO.IN, pull_up_down=GPIO.PUD_UP)
while True:
    input_state = GPIO.input(18)
    if input_state == False:
        os.system("/home/pi/rain_on.sh")
        time.sleep(5.0)
```

- Launched the Python script in a screen called "rain" so it runs in the background and I can attach to it if:

`screen -dmS rain python rain_check.py`

djarrard, 2019-04-26:
You can also do this by using requests to the API to manipulate switches. I like this better because all of my devices run on schedules via Terrarium Pi. This means if I want to turn on a switch that’s scheduled to be off in Terrarium Pi, the system just reverts that switch to its scheduled state within a few seconds (unless manual mode is enabled). I created a script that turns manual mode on and then turns on the switch via the api (see #282). I’ve got a separate button that turns manual mode back off for all switches. I  Works very well so far.

theyosh, 2019-07-07:
Ok, after some long thinking, I came up with an example script. Which can be used in other scripts.

This makes it possible to make your own hardware button, and interact with the software. The script is pretty easy to read. It can just toggle the state, or set it to a requested state (on/off). But that will leave it in manual mode.

When running with 'on 40' it will stay on for 40 seconds, and then go off. And will also return to normal mode and not in manual mode. So you can over write the system for X amount of seconds.

So that will probably as far I can support this. 

https://github.com/theyosh/TerrariumPI/blob/master/contrib/toggle_switch.py

theyosh, 2020-01-28:
Ok, closing this, as the example toggle switch is the way to go. Supporting all kind of buttons is not part of the core system.

closed 2020-01-28 by theyosh