theyosh/TerrariumPI #205: Request: Remove erratic readings

fenyx13, 2018-11-14:
Hi TheYosh,

Would it be possible to filter out erratic sensor readings?

I found an example of this for a DHT sensor here:
https://forum.dexterindustries.com/t/solved-dht-sensor-occasionally-returning-spurious-values/2939

Thanks!

theyosh, 2018-11-15:
Hmm, I have tought about this already. But this is a bit harder for me to make. As I only measure once every 30 seconds, I do not have that much time per sensor. So I have made the code to check a DHT sensor for max 10 seconds. If that fails, it is no just an error.

I am thinking of changing the error to a message and use previous values.

But strange readings are hard to fix for me. Because I do not have the time to make multiple measurements, due to the slow DHT sensors. That would cause other timing issues.

So, is you case that you have drops due to errors during readout? Those can maybe fixed. But if it is jumping up and down, I am afraid I cannot fix that right now.

fenyx13, 2018-11-16:
Thank you for taking the time to respond and explain things.
I understand the dht is a slow sensor and fairly unreliable. Soon  I will have a miflora sensor and I plan to use the dht in another terrarium, I only ask because it is the most easily accessible sensor for beginners. ( international post and customs is slow )

I’m not sure if it is errors or something else.
Here ist an example in the second image.
 https://imgur.com/a/QGLhaDG

theyosh, 2018-11-16:
Hmm, does drops should be fixable. I will have a look into this.
I do know that those DHT are easy to start with, but they are also the most unstable sensors there are. I had already a lot of issue with non working DHT sensors. So it is a bit of a trade off. Using cheap and easy sensors vs. stable and more expensive sensors.

theyosh, 2018-11-16:
After a bit more studying, the values are 'real values' in the sense there are no code errors. So that leafs that I/we have to decide how much a value may change compared to the previous value. That is currently the only way to do it in a simple way. Reading more data takes to much time. Or from the DB will also take time.

So I am currently testing a rather simple formula.

- get the min and max allowed values as 100% 'distance' (not the alarms, but limit values)
- calculate the difference between now and previous measurements
- calculate the difference / max - min in percentage

if the difference percentage is to high, I ignore the value and keep the old one. Still I have to figure out if this works for all values. Also some light sensors due spike a lot due to clouds or bright light etc..

fenyx13, 2018-11-18:
I will update now and report back soon.

jornobe, 2018-11-19:
I have seen the same drops as shown in the initial screenshot but the software would still continue to work using later readings from the same sensor. 

This morning, after the upgrade last night, I see the humidity sensor having an error state. Will the software still try to read from it later on or will it be ignored until I restart the Pi completely?

fenyx13, 2018-11-19:
Well... It has only been a few hours but so far all is good, my dht sensor is now reporting without erratic readings. My other sensors seem to read fine as well. 

Unlike my German homework due in the morning. :)

theyosh, 2018-11-19:
@jornobe what kind of error? Erratic values and errors should be skipped. Erratic values are now also cached for 30 seconds (will be changed) so those errors are now shown twice with. One when the error occurred, second the cached value. With an error, in the next run it should just being read again.

In other words, fails readings are ignored and tried 30 seconds later again.

There is only one problem still, when the first reading after startup is a very low or high value (that is wrong, but can't know it) other values will all be ignored. And if this is continuing for more then 10 minutes, the sensor will get in an error state. This should be fixed in code. This could be the case with your sensor

jornobe, 2018-11-19:
The sensor shows “warning” and “error” behind its name in the humidity sensor overview page. 

It did after a reboot come back to life but a few hours later it went back to the above mentioned state.

The biggest problem I had today with this is that, although for the humidity environment all values are shown as 0% (current as well as range), the switch is being triggered over and over again without this being shown in the graph of that switch. Turned out to be a very rainy day in the jungle today...

jornobe, 2018-11-19:
The part of the graph is not correct, sorry, it shows a thicker bar when it happened, my mistake. But it does trigger while not having a reading from the sensor.

theyosh, 2018-11-19:
What is er in the console for messages: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

Or go to system -> log and use the download option in the wrench option on the right part of the venster

warning badge is not that bad.. error badge is bad! That means you could have the issue I just described above.

jornobe, 2018-11-19:
Had to go further back than what I first saw. Top few entries are okay, then the low limit is hit triggering the switch to spray, the humidity value increases and the erratic massages come. While these are being logged the spraying system is being switched on and off for quite a while.

[jorno_be_log_19nov2018.txt](https://github.com/theyosh/TerrariumPI/files/2596821/jorno_be_log_19nov2018.txt)

theyosh, 2018-11-19:
Hmm, I see that the humidity is rising quit a lot after a spray action. Which is normal... So how to fix.

What are you min and max limit values? Because the first erratic message is from 66.8 to 78.0 which should pass. I have made the code that the values can fluctuate with 20% of the last value based on the limits values. So when the humidity sensor limit are 0 and 100, the max diff in value is then 20 which is more then the 78.0-66.8 = 11.2
But when you have altered the limit values to for example: 30 - 90
Then we get in this case: 90 - 40 = 50
Then (11.2 / 50) * 100 = 22.4 which is higher then the 20% limit

So could you check your sensor min and max limit values? And increase it to a minimum of 60/65 difference in between?

There are much better formulas to fix this, but that means that adding data should then be analyzed before setting the sensor. That cost to much CPU power and time.... So that is way I am trying this way to fix it

theyosh, 2018-11-19:
Also did update the code, that after 5 times getting an erratic value, it will use that value and should update the sensor data

jornobe, 2018-11-19:
Thanks for explaining and for the update. 

The set values: 
Alarm Min 67
Alarm Max 95
Limit Min 45
Limit Max 100

Will change the limit values to 0 - 100. The min difference between the 2 can maybe be added to the pop-up info.

theyosh, 2018-11-19:
Actually I was thinking of adding a max diff value on every sensor. Which will directly show what the max/mun jump that is allowed based on the entered values. That way you can play around with the values to get it right for the used setup. Now it is hard coded for first testing...

jornobe, 2018-11-19:
Just updated the values and the software. Will monitor the logs tomorrow but as the limits are within what you expect it should work out. 

(My humidity boosts sometimes like this as I also have a humidifier that runs on a timer. When it rains after the mist the effect is this change in sensor values.)

jornobe, 2018-11-24:
DHT22 sensor has been working fine for the last few days since this update. This morning it was in error state and after a reboot it just worked again. 

Can this be caused by having out of alarm set values for a couple of hours? Would think that as long as there is no software restart this should not happen. The limits for the sensor are 0 - 100.

If there is a more reliable humidity sensor I would exchange but have not been able to find something else.

Natrixz, 2018-11-24:
I can confirm the one's from taaralabs work pretty well
(I have about 5 of them, so far only 1 died wich was my own fault for not conformal coating it correctly and running it in a 90%-100% humidity environment)

jornobe, 2018-11-24:
Thanks for the responses and the info provided. Will look into this new humidity sensor!

theyosh, 2018-11-29:
Closing due to working as requested

closed 2018-11-29 by theyosh