theyosh/TerrariumPI #206: Useless notofication system :(

kaju666, 2018-11-16:
Hi 

So i try to explain why.
I have a temp sensor with set AlarmMin: 31℃ and AlarmMax 31.2℃.
In environment i have set heat to use this temp sensor.
So now i have all the time around ~31℃ on Heated terrarium bed.

But in notification i can only enable action when Sensor alarm is high or low which is useless becouse this state is all the time when using temp control.

In conclusion there should be a possibility to set Warning temerature in notofication. so for example when my heating bed will broke and temperature drop down below 30℃ or there will be some bug in code and it rise more that 32℃.

theyosh, 2018-11-16:
Hmmm, let see if I get it right. 

You are looking more like 'operation values' and 'warning/error values' where the operation values control the power switch to maintain certain temperature, and the warning and error values should trigger warnings to alert you?

There is maybe a partly solution right now. You keep you current alarm values, but change the limit values closer. So limit min is 28 and limit max is 32. When those values are hit for 10 minutes, an extra error icon will show at the sensor as indicating that the measured values are incorrect.
In your case, that would then be interpreted that there is something broken with your hardware.

Else, I do not have a solution right now

kaju666, 2018-11-19:
> You are looking more like 'operation values' and 'warning/error values' where the operation values control the power switch to maintain certain temperature, and the warning and error values should trigger warnings to alert you?

Yes, exactly this should be added becouse is important to be calm of you animal.

and to your temporary solution this will not send me any notification i think. 
Or this is set in "System warning notification"?

theyosh, 2018-11-19:
When you lower the min and max limits of the sensor, a warning will be issued when the sensor goes below or higher the limit.

When you enable the mail settings and select System warning notifications, you will get an email when this happens. Also, after 10 minutes, the environment will then also send warnings and errors due to wrong sensor readings. 

So you can get notifications when the sensor goes wrong. But it is maybe not fully what you need. But you could give it a try.

But, I think that failing hardware is a case which my software should not to deal with, as it can be number of issues. So, it will sent out warning and error messages. And the software will continue to operate. My software cannot replace hardware :).

theyosh, 2018-11-20:
Your example about a broken sensor is already covert: #142 . When you have 1 sensor in the heater environment settings, and that sensor is failing after 10 minutes it will shutdown the power. So that should take over of overheating issues.

So, when a sensor is also reading values outside the limit values for more then 10 minutes, it will shutdown the power. Because measurements outside the limit values are not valid and will trigger the same logic as a failing sensor.

As a sensor is failing due to either limit values or broken, it will sent out a warning. Those warnings can now be mailed to you. Or use pushover for direct notifications on your phone.

So I think your issues are covered by changing the limit values of the temperature sensor by putting them 2 a 3 degrees above and below the alarm values. Take in account, that at the moment the limit values are also used for the erratic values calculation as explained in #205

theyosh, 2018-12-28:
Closing now as I think there are enough measurements to tackle this issue. Adding more options will make it even more complex to setup.

closed 2018-12-28 by theyosh