theyosh/TerrariumPI #209: TerrariumPI not starting after update

Natrixz, 2018-11-18:
As the tile already say's;
TerrariumPI is not starting anymore after the update.
is just sticks with "starting terrariumpi server at location"

Natrixz, 2018-11-18:
I stand corrected, it does start.
However it does take ages before it starts.
About 3 to 5 minutes before anything happens at all.

theyosh, 2018-11-18:
Ah yesh! That is the database cleanup.... Mine is 20+ minutes :)

Every update it will clear the database in order to make sure there is no wasted space. Maybe I should make some kind of notification that that is happening on the screen. It is mentioned in the logfile though

theyosh, 2018-11-19:
Closing this now. As there is now a warning when the cleanup is taking place. Also add it to the know issues. https://github.com/theyosh/TerrariumPI/wiki/Known-issues#slow--hanging-restart-after-upgrade

closed 2018-11-19 by theyosh