theyosh/TerrariumPI #210: Pictures of running TerrariumPI system

theyosh, 2018-11-19:
[image]

I would like to know how people are using this software. As I have developed a lot of user features ,which is cool, I could imagine that there should be at least a dozen TerrariumPIs running. So I was wondering you can send me foto's of the environment that is powered by this software.

You can upload images to this issue. Just to see how you all using it! Maybe we can even share handy tips...

jornobe, 2018-11-19:
This is my 2m40x1m50x65 Chinese Waterdragon enclosure. Just recently added the TerrariumPi to get better control on the environment. Have been using a mix of wifi sockets, timers and thermostats but it was all too static. 

What I currently have attached to TerrariumPi:
- Rain system triggered by humidity controlled environment
- 3x 50W SolarRaptor HID bulbs controlled by lights environment
- Temperature sensor in the pool
- Temperature sensors on the basking spots

Upcoming days:
- Connect the 2 Ceramic Heat Emitters to a relay and set up temperature environment
- Install second Pi for humidifier control based on second humidity sensor
- Connect pool heater to second Pi and add temperature environment

[image]

kaju666, 2018-11-20:
**What I currently have attached to TerrariumPi:**
- 2x DS18B20 in COLD and HEATED FLOOR AREA
- LED LIGHT controled by relay
- Half Heated floor using cable and controled by relay
- DHT22 for humidity and AIR temperature show
- RPI CAMERA v1.3

**Upcoming features:**
- ADD longer (2m) RPI Camera tape.
- Print and mount a holder for camera

**Terrarium View:**
[image]
**Control box**
[image]

**My new family member :P**
[46521033_915757315284824_4724618700257230848_n](https://user-images.githubusercontent.com/12675680/48792076-a2f56780-ecf3-11e8-8267-5559eeb5a2a7.jpg)

Natrixz, 2018-11-20:
Some older pictures of one of my enclosures, this one is for a Brazilian Rainbow Boa.
Plants etc have changed quite a bit but other than that;

Pi running;
3x TaaraLabs Humidity and Temp sensor through OneWire
2x DS18B20 for extra temperature and water temperature for my Mistking system
1x Energenie EG-PMS2
2x Jungle Dawn LED Lights
1x Arcadia T5 UVB
1x Arcadia Deep Heat Projector
1x Mistking System
1x HC-SR04

[image]
[image]
[image]

My second enclosure runs pretty much the same system, 3rd system is currently being build up.
Both second and third system are for Emerald Tree Boa's.
Might figure out a way to heat the water with some heat cable's so I can spray with warmer water.

@jornobe curious to your solution to do that =)

jornobe, 2018-11-20:
@Natrixz instead of heat cables you could add a 25W aquarium heater (with thermostat) inside the water supply (I use a 40L jerrycan) and attach it to a sensor to cut off power when the water level goes below a certain point. Could connect the pump of the rain system to the same sensor to disable the system completely if needed.

Natrixz, 2018-11-20:
Awesome idea! That’s gonna be my next project times 2 haha :)
Thanks @jornobe :)

AndrejWeber, 2019-02-02:
So, after winter rest, my **Gecko Wold** is already allive!
OSB Terrarium 150x60x60 cm³
elctronics cabinet above about 20cm hight.

Pets: 5 Leopardgeckos, 1x male, 4x female

[image]
RasPi 3B+ with Joy-It EXPLORE 500 extension
1.3" OLED-Display I2C
OSRAM qti 1x54W dim
Breadboard for connections and dimming
power strip with relais (modified from USB to GPIO-switching)
3x DHT22 (warm side, coldside, electronics)

[image]
[image]
Rear wall and caves made of styrodur and tile glue
Big wet-box (Tupperware covered by styrodur cave)
Day-light T5 54W Lamp with UV 
75W halogen spot
PWM-controlled Fan (right upper side) for cooling and drying.

theyosh, 2019-02-02:
@AndrejWeber "Gecko World" Brilliant and very good looking in the back wall!!! I have also support for a 'traffic' light indicator of the status of the system: https://www.amazon.com/Pi-Traffic-Light-Raspberry-pack/dp/B00RIIGD30 / https://github.com/theyosh/TerrariumPI/wiki/Hardware#led That way you could have some operating status in next to the LCD screen

AndrejWeber, 2019-02-02:
@theyosh thank you, I have thought a lot about ways, the geckos can climb at the wall, not to big distance between platforms... at the end, the tile glue has so much grip, the geckos can climb where they want. one of them like to sleep on the warm halogen spot, after shutting off.
the traffic light I also saw in the documentation. that will come together with support for push-buttons - you only have to implement ;-)
I have a complete button and LED component from PC-Tower. Power and Reset button, power and HDD-LED. I will not use a green indikator, for all right, only red and yellow for something wrong.

toads100, 2019-02-14:
Still a work in progress

Running AM2302 for temp and Humidity For top reading
1wire Temp for middle on enclosure  reading 
HC-SR04 for Water level
Working on misting system now.
https://www.youtube.com/watch?v=md7RS8VD0aQ
[image]
[image]
[image]

stuartabrown, 2019-02-14:
@toads100 very cool!  Where did you get your RPi case from?

toads100, 2019-02-14:
I 3D printed it, here is the link to that print https://www.thingiverse.com/thing:3008921
I changed it a little to fit my power supply. I wanted a power supply that had 12 volt and 5 volt out to run my cooling fan and Pi. I found at after the fact I could'nt run my rain system on the same power supply.

Raerion, 2019-02-18:
Hey,
I'm a **Chinese stripe-necked turtle** (_Mauremys sinensis_) and I'm living in a turtle tank automated with **TerrariumPi**! 
[image]

I'm sharing the water level with some fishes (_Poecilia reticulata_ and _Poecilia sphenops_) that may end up as food and snails (family _Neritidae_). ;)

**Facts:**
- 120x50x40 cm tank
- Raspberry Pi 3 Model B Plus
- Energenie EG-PM2 (USB)
- External filter with heating (200 W; curently off due to winter time, will be automated with temperature environment via EG-PM2)
- LED-Panel (25 W) with PWM via GPIO (lighting environment for PWM and power switch via EG-PM2) 
- UV-Lamp (Lucky Reptile Bright Sun UV Desert 70 W; power switch via EG-PM2) 
- MiFlore Bluetooth sensor (land)
- 2x DS18B10 (OWFS) for air and water temperature
- Raspberry Pi Camera (livestream mode)

**Current to-do's:**
- VEML6075 to monitor the life time of the UV-Lamp
- secondary camera (maybe with night vision)
- tidy up the wiring and technical equipment... -.-

**Pictures of the current setup:**
[image]

[image]

**Some more pictures:**
[image]

[image]

djarrard, 2019-04-25:
First of all, thank you @theyosh for putting this beautiful piece of software together. I've had so much fun working with it.

After seeing all of your beautiful setups, I feel kinda silly posting mine. It's not quite finished yet, as I don't have my ball python yet. I grew up with one and always struggled to keep the environment within the correct parameters, so I decided that this time I wanted to over-engineer it. This will be the forever-enclosure for my soon-to-be specialty morph ball python. It still needs a bit of work. I need to decorate the interior, and I need to cleanup the massive wiry mess that's in my server box. I should add that almost everything here was custom built by me. The entire enclosure, the sliding glass doors, even the humidifier is a hack. As you can tell, I'm not master of fine furniture or beautiful circuits, but I think it'll get the job done with some extra detail work. I should also mention that this is my very first Raspberry Pi project. It's amazing how something that starts as a means to an end can quickly become an obsession!

Facts:

- 4ft by 2ft custom enclosure built with melamine
- 12 watt LED strip Light for main lighting
- Hacked humidifier for humidity control
- Night light with a low wattage red LED (coming soon)
- Pro Products Radiating Heat Pad (coming soon)
- Herpstat Proportional Thermostat (coming soon)
- Sliding plexiglass doors on aluminum rails
- Raspberry Pi 3B +
- JBtek 4 channel relay
- 2 DHT22 sensors for measure humidity inside and outside of enclosure
- 4 DS18b20 sensors for measuring temperature inside and outside of enclosure
- 4 buttons for enclosure controls
- 1 led indicator for status

[image]

The first picture shows the layout of my enclosure. It's pretty bland at the moment, but the pieces are there. I'd like to add some door sensor for the sliding glass doors. I ordered a custom-size radiating heat pad from Pro Products...will install it as soon as it arrives., Lighting, humidity, and sensors are all regulated using Terrarium Pi. I'm planning to regulate the heat using a Herpstat proportional thermostat, just to be safe. If my humidifier malfunctions, the worst thing that could possibly happen is that the enclosure turns into a foggy abyss for a while. If the heater malfunctions, I could bake the little guy...and that would just suck.

[image]

The second image shows a button layout that I put together. These buttons came with the raspberry pi kit i ordered. Nothing special, but very clicky! These little guys are hooked up to always-listening python scripts that interface with the Terrarium Pi API to turn on manual-mode and toggle the various switches. I could have just wired the scripts directly to the GPIO ports, but I decided the API was a better approach for two reasons

1) I don't want to cause interference by having two entities try to control the same GPIO pin.
2) I want whatever I'm doing to be federated under Terrarium Pi for easier management.

The humidifier switch is a bit special because it cycles the humidifier for 2 minutes and then turns it back off. I have an indicator light at the top there that turns solid red if manual mode is enabled on any of the switches (which happens if I hit any of the first three buttons). If I hit the 4th button, it sends a message to the API and manual mode is turned off on all switches. That way I can mess with the environment when in the room and hit the last button on the way out to put everything back on schedule. The indicator light does one special thing too. If at any time the Terrarium Pi Web Server can't respond within 2 seconds to a request, it starts blinking. That way I know at a glance if something is up.

[image]

The third shows my "server box". Yeah, it's a bit of a nightmare at the moment. I bought this tray on sale for $7, spray-painted it white, and started mounting the equipment in there. It's very messy, I know. At some point I need to upgrade the wiring to dupont wires and make it look a bit more organized. I'm going to build a white hinge lid for this thing that covers the server box to make it look cleaner.

[image]

The fourth shows my humidifier hack. A lot of reptile foggers and misters have very....mixed reviews. They fail, start making noise, or start leaking. Household humidifiers, however, tend to last forever. I bought this guy for $30, and with about $15 in parts from the hardware store, I was able to connect a hose that diffuses the fog in two directions in the enclosure.  Hoses can get moldy, so I also designed it so that the hoses are easily removed for deep cleaning without having to disassemble any of the connectors.

Maybe I'll post back again in 6 months once everything is finalized. If you guys have any suggestions, I'm all ears. There's still time to change things.

funware84, 2019-07-09:
[image]
[image]
[image]
[image]
[image]
My pond i am monitoring and control.
Difference Sensor show diff between water overflow and bottom. If it raises to much it will start a 2nd pump.
In future it will control solar heating as well.
I can toogle filtration pump which zurkulate 20.000l/h 
With mi flora i check conductivity.
It seems as it can measure the height of waterlevel in this case. I wil evaluate this next days

romandiaz, 2019-08-08:
Many MANY thanks to @theyosh for creating this software and documentation and supporting it so well. Just amazing.

[image]
I built a 2-in-1 enclosure. The cage measures 48"x16"x21" with a divider and removable passage in the middle to connect the sides. Each side has a temperature sensor and radiant heater, and there's a series of LED light fixtures which provides light to both sides from one switch. The Pi and all the electronics are built into a cubby in the left side enclosure (accessible from the back). This setup leaves one open switch on the relay - thinking of adding intake fans or something like that in the future. 

[image]
Here's it is in place. All plants are plastic - easier to clean and keep green. Plus, the snakes don't care. I wired a red LED as an indicator for when the heating element is on; for monitoring at a glance. So far it has been up and running for over a week, keeping things nice and stable. Here the left is off and the right is on - but during normal operation they get switched on/off at the same time. I wish that I could operate the "Environment" in both cages independently (for example: keep one at 80F degrees and the other one at 85F degrees) - is this possible with a single Pi?

[image]
Just another shot - this one showing the removable passage in the cage divider.

[image]
Happy snake. 

Temperature failsafe hack - 
All of the power for the cage comes through a single PC power cord. I plug this cord as if it were a "heat source", into an off-the-shelf thermostat (from the old setup) set to 100F and put the probe in one of the cages. If temps go over 100 degrees - indicating something has gone wrong, since my max temp is 88F - this thermostat assumes it has reached the desired temp and turns the "heat source" power off, shutting off the power to the cage and stopping the danger. A little hacky but it works.

bpengu1n, 2019-10-10:
> First of all, thank you @theyosh for putting this beautiful piece of software together. I've had so much fun working with it.
>
> After seeing all of your beautiful setups, I feel kinda silly posting [...]

How well is your system running? It looks amazing!!

fitzgepn, 2019-11-02:
I am working on a system to control the vivarium for my family's bearded dragon. I designed this 3d printed case to house a Raspberry Pi Zero W, 5v power supply, solid state relay board, and 5 inch touch screen.
[image]
I have also been working on a custom Python UI using Kivy, as the web UI doesn't work so well on the 5 inch screen and the Pi Zero. When it reaches usable state, I'll submit it for your consideration.
[image]
The temperature sensors are two DHT22 boards, but I'm planning to upgrade to BMP280 as the one wire protocol is a bit flaky over the 2m wire to the cool side of the enclosure. I also have a VEML6075 UV sensor to monitor the output of the lights.
[image]
I would like to suggest adding PID thermal control for the heaters, and may eventually get around to implementing it if you don't do it first. Ideally, I would like to be able to have setpoints for day and night, and have TerrariumPI turn the heaters on and off on a duty cycle to keep the temperature. In any case, you have made an excellent project as it is.

gale-force, 2020-01-20:
I'm playing around with this for some simple monitoring (and alerting) of my Leopard Gecko viv, two DHT22's so far, adding a third and also a RPiCam for streaming, it's a great project, thanks for the work.

Haidy777, 2020-01-25:
In the last few months, I've been working on a Terrarium with my Father. It is a great experience to learn how to work with wood from a professional (my dad).
Today we transported the Terrarium to my Apartment. The base is mainly done and it is now my turn to get it set up with TerrariumPI and interior.

Currently, there is only one lamp and three temperature sensors mounted but I already ordered some more sensors. The windows look blue because the protection is still on the acrylic glass for now.

Hopefully the leopard gecko I plan to house inside will enjoy it.

Dimensions: 120 cm wide, 60cm deep and 70cm heigh.

[image]
[image]
[image]
[image]
[image]
Current cable mess on top. We built it so I could also open the top side if I would need access.

theyosh, 2020-01-26:
Nice, I have a brother with that skill! That is always handy :) Great looking!!

xagesz, 2020-01-28:
[image]

I decided I wanted to learn pcb design and an adapter board for a rpi would be a good start. I set it up to run triple dht22’s. Ultrasonic sensor, veml6075, and a relay board.  I will use a female 20pin to set this directly on the board and run 3ft extensions off of these. Also a 6pin hub for my 4ch relay board.  I ordered 3. If someone wants one let me know. This was just a practice exercise. I have a few more I want to build. 

[image]

Ultameciia, 2020-03-04:
Kinda late to the party but now that I've just figured out a central monitoring piece I was able to finish the overall monitoring and control layout. I'll add a pic of the most recent enclosure as well. Currently I've got 8 Pi's in rotation (and still need 4 more). 
1 is a Pi Zero that aggregates all the sensors on one page, this page is displayed on a old TV in the reptile room.
1 is my "environment" Pi it controls the mistking and light routines. I have a 24 port power strip plugged into one plug on a Meross power strip and the mistking plugged into the power strip. This controls all light for my Tokay Tanks. Mistking is kicked on or off based on the average humidity of the tanks(those sensors are added via remote) it also has an ultrasonic distance sensor for the mistking resevoir. I'm also planning on adding a spare DHT11 or 22 just to monitor ambient temps/humidity in the reptile room.

The other 6 pi's have either a dht22 or an SHT20(i'm slowly replacing dht22's). They're primarily just to monitor and I'd eventually like to add rpi camera's to them. 

Now if I could just get my calendar working I'd be set! I'd like the Pi Zero to be where the calendar is, but it seems to ignore any URL I input. The only things left I'd like to do are door sensors and setup the remaining pi's. I absolutely love this project. 

Here's a pic of a screw block terminal case I use and love.
[image]

And here's the most recently finished enclosure, it's for one of the tokays I've produced this season. He's a holdback. I love the iheartgeckos conversion kits.
[image]

kahuwi14, 2020-04-07:
Dear yosh,
Dear community, 

after a long time of taking advantage of this great project and trying to be a part of the TerrariumPi - community as well, I finally managed to upload some pictures of my enclosure.

I set it up with a DS1820, a DHT22, a Heatpad, of course some light and a misting system.
The enclosure is running now, for about 3 years and I'm absolutely happy!
[image]
[image]
[image]
[image]
[image]
[image]

Beside this, last year in fall, I started another little project: little grow house based on a hydroponic ebb and flood system. In that enclosure I'm growing some hungarian hot wax with great success. 
[image]
[image]
[image]

All in all, yosh, thank you very much for the great work!

Best wishes to all and stay healthy!
Kahuwi

Hemibon, 2020-05-10:
And here is my terrarium for which I am using TerrariumPi. 
I will introduce some Ranitomeya imitator banded in a few months. First, the plants, springtails and isopods need to properly establish :)

[image]
[image]
[image]
[image]
[image]
[image]

havealoha, 2020-06-24:
TLDR This isnt a terrarium.

This is my first attempt at growing microgreens and of course I want to make it geeky. I tried [FruxePi ](https://github.com/fruxefarms/FruxePi)but there are some schedule reliability issues to work out there before it can be relied upon and the developer is MIA. Others suggested TerrariumPI and here I am. Install was way easier and it even worked on my 4b full install 2020-05-27-raspios-buster-full-armhf despite the tutorial stating that it wouldnt work.

Ideally I'd like to control each shelves lights, fans, pumps, and heating pad independently as their own environment (@theyosh 
 may support that some day) but there are only so many GPIO's unless I use something like [this](https://sequentmicrosystems.com/index.php?route=product/product&search=relay%20raspberry&product_id=50). And you can only fit so many outlets and relays in a small box. I made use of all 8 relays by separating some of the outlet circuits.

Thanks for producing such great software!
 
Whats attached?
Huayao 8 Channel DC 5V Relay Module
DHT22
TE215 (Not supported yet)
Barrina LED T5
AC Infinity MULTIFAN S7-P
smartpond 80-GPH Submersible Fountain Pump

[image]
[image]
[image]
[image]
[image]

theyosh, 2020-06-24:
Cool, that is something different! Nice to see more different setups!

1 mayor tip: Replace you DHT22 with a BME280 sensor. ;) https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed+DHT22

havealoha, 2020-06-24:
> Cool, that is something different! Nice to see more different setups!
> 
> 1 mayor tip: Replace you DHT22 with a BME280 sensor. ;) https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed+DHT22

Wow, thats a lot of DHT22\AM2303 issues!

ewiandr73, 2020-07-05:
Here are the three enclosures I have completed I have another four to go....

[image]

[image]

[image]

nicholseric, 2020-08-02:
> > Cool, that is something different! Nice to see more different setups!
> > 1 mayor tip: Replace you DHT22 with a BME280 sensor. ;) https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed+DHT22
> 
> Wow, thats a lot of DHT22\AM23## issues! This is an interesting read http://www.kandrsmith.org/RJS/Misc/Hygrometers/calib_many.html

My first DHT22 failure. After a quick reboot, back to normal.
[image]

blatogera, 2020-08-05:
Hello,

Finally everything up and running here, 
- Rpi 4, 2Gbyte ( after trown away Pi1 B, ) with a 4ch relay board
- 2x spray heads whith an amazingly good priced chinese 220V pump
- 2x BME280 sensors, ( after some endless fight with DHT22 sensors ) works like a charm
- 2x UV lights 5.0, 10.0, 
- 1x warmer, 1x daylight bulb
- custom "enclosure :D " for the electronics in a wooden box 10x13x35cm with 4 wall sockets
- old Galaxy Tab 2 for local control
- 1x happy panther chameleon

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Dave0408, 2020-12-18:
This is my 80x50x40cm terrarium and of course my box for all of my electronics.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

gam3t3chelectronicshobbyhouse, 2020-12-29:
Just a bit of testing with the software, great stuff but missing somethings I was hoping to incorporate into my project but all and all A+  front panel and a bunch left to do and get this guy fully enclosed.   will post some better complete pictures after. 

[image]
[image]

jeryko87, 2021-01-14:
So... Finished the build today. 
Still have some environment config to do but i'll get there...
[image]

samwsimpson, 2021-02-13:
how many enclosures could you run off a single TerriumPi setup?

killwater, 2021-05-29:
Hello, 
Here is my setup for one male Laudakia Stellio Picea:
[image]

[image]

[image]

[image]

[image]

Currently running with one DHT22 Temperature sensor in the basking area and 4 relay switch driving a HID Bright Sun lamp for light/UVB and 50W halogen lamp for heating. 
Planned:
Cage extension to make the footprint bigger
PWM controlled fan
PWM controlled halogen dimmer made of desk lamp dimmer controlled by a photoresistor
UV A and B sensor on the basking spot VEML6075
IR temperature sensor on the basking spot MLX90614
DHT11 sensor for ambient 
DHT11 sensor for humidity around the basking spot
DHT22 sensor moved to the cold side of the cage
Night deep violet 1W diode 420nm

theyosh, 2021-05-29:
Hi @killwater,

thanks for the pictures. Looks nice.
One advise. Drop DHT sensors and go for BME280 sensors. There are a lot of issues with DHT sensor in the long run: https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed+dht

killwater, 2021-05-30:
I will replace them once I get some issues as I already bought DHTs. Seem to be working fine for now. The most important temperature readout is going to be taken from MLX90614 both for basking spot and air in hot end. Being a desert setup I do not really need humidity sensor for control - only for reference. 

I do have an issue with IR readout of the basking spot though - when the lizard will go there sensor will measure avarage temperature of the spot and the lizard dropping down the value and triggering the basking lamp switch even if not needed. Does anyone have any experience with that? 
The only solution I can see is to set it up to ambient and once lizard comes on the spot and temperature drops extra basking bulb is trigerred. This would save some electricity as well as it would only work when the basking is in progress.

theyosh, 2021-05-30:
> I do have an issue with IR readout of the basking spot though - when the lizard will go there sensor will measure avarage temperature of the spot and the lizard dropping down the value and [...]

Please open a new issue... That is easier. I think there are maybe ways to make it work...

morrell7, 2021-06-25:
[image]
[image]
[image]
[image]

danimalia2017, 2021-08-24:
Hi @theyosh  

Thank you for such great work!

I have an ocellata orumastyx. I am running: 

4x 1Wire sensors (Basking spot, Hot Hiding Spot, Ambient & Cool Hiding Spot)

1x DHT22 for Humidity 

1x VEML6075 for UVA, UVB and UVI 

1xWebcam with wide angle lens 

2x GPIO to control the T5 tube and UV/basking spot lamp. 

[image]

[image]

[image]

[image]

grahnema, 2022-04-01:
@theyosh 

Thanks for all the great work on this project!

I just started using the program to control my custom dart frog enclosure.  I have 6 Dendrobates Leucomelas, I 3D printed a control box that houses all my electronics. 
Mosfett trigger for air circulation inside the enclosure
Mosfett trigger to pull in fresh air and reduce humidity
IRF520 for main LED plant lights  
IRF520 for moonlights 
Mosfett trigger for DC powered PTC heater
1 wire temp probe
BME280 humidity sensor
Sonoff (tasmota) for Mistking 

I realize some parts are overkill but I used what I had on hand. The Raspberry pi is also powered over GPIO connected to a LM2596 to bring the incoming DC 24v down to 5v. Everything is also controlled by one incoming 24v 10amp power supply so there are a few other buck converters to regulate voltage to various parts. 
[image]
[image]
[image]
[image]
[image]

[image]
The box is a little messy ATM and a few parts missing at the time I took the picture but this current setup is also a prototype.

ewiandr73, 2022-05-29:
I thought it was about time I added a few pictures of my enclosure (1 of 15) for one my Green Tree Pythons.

I built it between 2 rooms with front and back viewing, just need the snake to grow a little before releasing permanently in it's new home. I now have 6 of the 15 enclosures running on a single TerrariumPi without any issues.

[image]
Front

[image]
Back

[image]
Moonlight

[image]
Equipment

[image]
[image]
Future Resident

samwsimpson, 2022-05-29:
Can you list all of the components you used?

ewiandr73, 2022-05-29:
Do you mean electronic or all components (timber, acrylic, etc...)?

samwsimpson, 2022-05-30:
Electronics

theyosh, 2022-06-03:
> I now have 6 of the 15 enclosures running on a single TerrariumPi without any issues.

Damm!!! That is freaking nice! I expected 2 or maybe 3 enclosures. Not 6 or more... :D :D But this is very cool.

samwsimpson, 2022-06-03:
That is nice, I had started this project but ended up buying herpstats to
run my infrared heat lamps since I needed so many. Glad to see more than 1
setup being ran, but I would like to know how you did it (parts used) :)

Sam Simpson
RAD Octopus LLC
https://radoctopus.com

418 Peoples Street, Suite 110
Corpus Christi, TX 78401

ewiandr73, 2022-06-05:
> That is nice, I had started this project but ended up buying herpstats to run my infrared heat lamps since I needed so many. Glad to see more than 1 setup being ran, but I would like to know how you [...]

Hi,

I have one Pi4 running the TerrariumPi SW and also with as few DS18B20's & BME280 connected to the Pi GPIO.

I have 3 Sonoff 4CHR Relays flashed with Tasmota with DS18B20's connected to them, 2 more ready to add more enclosures.
and a couple of 4Gang power boards also flashed with Tasmota for anything that needs transformers to run.

All you have to do then is make sure you have the right temperature sensor and relay connected to the right enclosure.

**Equipment list**
1 x Pi4
5 x Sonoff 4Chr Replay (3 connected)
2 x 4 Gang power board
8 x DS18B20 (5 more waiting to be connected)
1 x BME280
1 x BME680 (for new enclosure)

Once all connect that should give me 28 relays and 15 sensors (might need a few more sensors).

So far I don't get any issue really, sometimes the sensor disappear from the Sonoff device, but rebooting the device normally fixes that.  The Sonoff devices allows you to have the enclosures in different rooms with out the need to run control cables.

I would like to try and workout a way to add temperature displays to each enclosure and read the value from the appropriate sensor, just as a visual check when I pass b the enclosures.

Andy

samwsimpson, 2022-06-06:
Thank you and please keep us posted if you get the temp displays working.

ewiandr73, 2022-07-17:
> Thank you and please keep us posted if you get the temp displays working.
> […](#)

Just finished adding another 2 enclosures one of which now displays the temperature on a little display.
The Display is connected to one of the SONOFF tasmota devices and display the temperature along with the date and time separate to the terrariumPi.

**HW:**
1 x Sonoff 4CH Relay (Flashed with customer Tasmota)
1 x BME680 sensor
1 x DS18DB20 Sensor
1 x SSD1306 OLED display

You should be able to get 2 screens working per sonoff device and I'm still playing with the setting in Tasmota to get it to switch between the sensors.

[image]

Here is my latest enclosure made out of a display fridge that is triple glazed so doesn't loose much heat and you can view from all 4 sides, just needs a few more plants :)

[image]

Here is a simple set of 4 shoe display boxes setup with a heat cable across the back and added as a single enclosure for my new hatchies.

[image]

Potentially you could send temperatures to the displays from the TerrariumPI if you could configure KNX to work on the TerrariumPI.

https://tasmota.github.io/docs/KNX/

Regards

Andy

sni89, 2022-07-19:
Setup for Uroplatus Species 

1x VEML6075
2x BME280
6x 1W Temperature 
2x MI Flora plant sensor 
1x OLED Display
1x PCA9685
1x 8 Relaisboard

4 x Light dimming via MW LDD-500L and LDD-700L

3x High pressure pump for misting

1X 5v 7A and 1x 36v 3A 

The System controls 3 enclosures for now which are in the making 😝
Will update when it’s all installed in position 

[image]
[image]

EvilHomerGD, 2022-10-15:
@theyosh Thank you for the wonderful work on TerrariumPi. I am using it to control, at least partially, two enclosures and monitor my breeding rack (still a work-in-progress).

Enclosure 1 is an Exo-Terra 40G bioactive setup housing my female super pastel piebald ball python. That setup consists of 2 SHT30s for temp and humidity, a Chirp to monitor soil moisture levels and a Kasa power strip to control the lighting. I'm going to be adding some additional relays to manage the humidity levels and a dimmer so that I can control the heat lamp a little better.

Enclosure 2 is a 20G aquarium that houses my 2 baby map turtles. That whole setup was a spur of the moment purchase and will be getting more automation added as time and money allow. Currently, I'm using another Kasa power strip to control the heat and UVB lights. Eventually, I plan on adding some temp sensors and using those in combination with the Kasa and a tank heater to manage the water temps.

The final enclosure is my breeder rack. It currently houses 3 juvenile ball pythons and will be home to more when I finally start breeding in a year or so. Currently, I am only using it to monitor the temps and humidity with SHT30s. When I expand the rack, I will add additional relays/dimmers to control the belly heat and monitor/manage temps and humidity.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

EvilHomerGD, 2022-12-09:
@theyosh , as promised, the latest pics of my enclosures, including my Pi Pico W setup for remote temps and door sensors. I'm still awaiting a few things from Adafruit but those last bits should be in tomorrow.

This is my poison dart frog enclosure that houses 3 juvenile Dendrobates tinctorius Patricias:

[image]
[image]
[image]
[image]
[image]
[image]

This is the enclosure housing the adult bearded dragon I just rescued and the Pi Pico. Her name is Charlie (I'm only 95% sure she's female):

[image]
[image]
[image]
[image]

And this is my newest ball python enclosure that is housing my male banana pastel named Bruce Bananner:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

I'm waiting on the SHT30s in the weather resistant housing for the dart frog and BlackBox enclosure for my ball python. I will get those wired up and mounted in a fashion similar to the bearded dragon's enclosure.

mink8mink, 2023-04-08:
Amazing setups! 👏👏👏👏

oldmanbluntz, 2023-10-09:
Oh, and after running the whole setup for the same time using DHT11 sensors, I have not run into issues with them persay. They occasionally, according to the log, have bad readings, but it doesn't affect the overall function of them. After everything I have read, I have expected them to do crazy things, but they've been good. 

As you can see in the graphs, they aren't smooth, but they still give enough clean data to be usable. 
[image]

oldmanbluntz, 2023-10-29:
The newest addition. Gus Gus the Bearded Dragon. 20x24x40 inch enclosure. 3 DS18B20, 1 DHT11, 1 BH1750, and 1 Mlx90614 to read the temperature of his bask rock. 1 exhaust fan (soon to be 2), 1 circulation fan that pulls heat from the Bask light and CHE fixture. Arcadia 24 watt 22" UVB light and fixture.

This enclosure is a remote enclosure. It uses an ESP32 running the Tasmota firmware to send out the sensor data over the LAN that Terrarium Pi reads.
[image]
[image]
[image]

tassieKev, 2023-11-28:
My setup for a couple of nepenthes plants (tropical carnivorous plants). Still a work in progress and definitely needs some prettying up ;) but the plants are loving it. Running at around 90% humidity and temperature in mid 20s celsius.

Tank is a repurposed BiopodAqua. Came from a kickstarter back in 2015 I think. Worked well for a while but has been in a shed for many years. Raspi Pi 4 using the docker version.

Sensors
SHT20 I2C Temperature & Humidity Sensor (Waterproof Probe)
DS18B20  Waterproof Temperature Sensor

Lighting is from a Fluval 3.0 Plant Spectrum Bluetooth LED 32W 61-85cm using it’s inbuilt timer.
Heating and the mister run off Meross plugs. Sadly they haven’t released an Australian version of their power board but the Outdoor Smart Plug (2 socket) and Smart Plug (single socket) work fine with TerrariumPI.

Thanks to @theyosh and all the other contributors.

Cheers Kev

[image]
[image]

HappyFrogVivarium, 2024-01-13:
I've been using TerrariumPi for a while on a smaller enclosure but I now have it running my 450G/1700L Dart Frog Vivarium.  
[image]
I'm running two temp sensors and humidity sensor, circulation fan, exhaust fan to change the air inside because its completely enclosed. I also run my day and moonlights as well as misting . Also created Youtube video of the interior.  [https://youtu.be/w_ah6Rseanc](url)

manolof, 2024-02-22:
My setup for my newly adopted 9-year-old Crested Gecko.
Self-made terrarium out of plywood, water sealed, carved styrofoam, and colored tile adhesive.

The electronic system:

- 1x Raspberry Pi 4B 1GB, with a GPIO Screw Terminal Hat
- 1x SRD-05VDC-SL-C Relay Module (double)
- 1x Custom made PCB to connect all things together
- 1x BME280 sensor
- 1x HC-SR04 sensor inside the water tank (in progress..)
- 1x water pump (MIUS, from Aliexpress)
- 1x RPi IR camera (67 degrees)

and a lot of smaller equipment such as cable connectors etc.

The relays control the ceramic heater and the main lights, as well as the water pump for the misting system.

I am planning next week to upgrade the following:
- Add LED Display
- Replace the relay with a Solid State Relay for better reliability
- Find a better way to power the whole 5v system, instead of relying on a 5v charger

[image]
[image]
[image]
[image]
[image]
[image]

Ludwig:
[image]

HighwingZ, 2024-03-12:
Here is a report on my vampire crab vivarium :)
[https://imgur.com/gallery/YODMTVP](url)

[4 more replies]