theyosh/TerrariumPI #211: Email notifications sent to invalid recipient address

jornobe, 2018-11-20:
Sending email notifications to a recipient address in the format "firstname@domain.org" results in messages being addressed to "firstname+terrariumpi@domain.org"

"firstname.lastname@domain.org" works fine as recipient address using the same relay server settings.

theyosh, 2018-11-20:
Not sure what the issue is? Because the + is used for 'tagging' the mail so you can easily created filters in your own mail client. And it should be valid.

all mail at firstname+whatever@domain.org should just be delivered at firstname@domain.org. If that is not the case, the receiving mail server does not support this. Which is a bit strange, as almost any mail server should accept it.

If you have mail server error logging, maybe it gets clearer...

jornobe, 2018-11-20:
Understand the idea. The tagging doesn’t seem to work on my @gmail.com target address

jornobe, 2018-11-21:
Was an SMTP sever issue... changed to gmail relay and it works as expected!

theyosh, 2018-11-21:
Cool! Then I am closing this issue now.

closed 2018-11-21 by theyosh