theyosh/TerrariumPI #212: OWFS file system is not active

Natrixz, 2018-11-21:
After the most recent update I suddenly get this warning;
2018-11-21 08:08:34,760 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device!
2018-11-21 08:08:34,764 - WARNING - terrariumSensor      - Measured value None% from humidity sensor 'Rechts-Warm' is outside valid range 50.00% - 180.00% in 0.01454 seconds.
2018-11-21 08:08:34,897 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device!
2018-11-21 08:08:34,899 - WARNING - terrariumSensor      - Measured value NoneC from temperature sensor 'Rechts-Warm' is outside valid range 15.00C - 40.00C in 0.00789 seconds.
2018-11-21 08:08:35,146 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device!
2018-11-21 08:08:35,148 - WARNING - terrariumSensor      - Measured value NoneC from temperature sensor 'Midden' is outside valid range 15.00C - 40.00C in 0.00791

Curious as to what happend =)

theyosh, 2018-11-21:
And what give: `sudo netstat -tulpn | grep owserver` ?

theyosh, 2018-11-21:
And what is in the `/etc/owfs.conf`? There should be a line like: `server: port = localhost:4304`

theyosh, 2018-11-21:
The line: `Active: active (exited) since Wed 2018-11-21 08:20:09 CET; 5min ago` looks wrong. 

I have `Active: active (running) since Sun 2018-11-18 19:05:11 CET; 2 days ago` I have state running and not exited.

so is `ps fax | grep owserver` give the output:  
`847 ?        Ssl    6:26 /usr/bin/owserver -c /etc/owfs.conf --pid-file /var/run/owfs/owserver.pid`

Natrixz, 2018-11-21:
sudo netstat -tulpn | grep owserver Strangely doesnt report anything...

However catting owfs.conf shows the correct line at the bottom

Natrixz, 2018-11-21:
Hmmm perhaps my owfs controller died right before/during/after upgrade?
Everything else seems fine

Natrixz, 2018-11-21:
> The line: `Active: active (exited) since Wed 2018-11-21 08:20:09 CET; 5min ago` looks wrong.
>
> I have `Active: active (running) since Sun 2018-11-18 19:05:11 CET; 2 days ago` I have state running and [...]

ps fax | grep owserver
 2316 pts/0    S+     0:00              \_ grep --color=auto owserver
So im thinking the controller has an issue (perhaps)

theyosh, 2018-11-21:
Hmm this is strange. Then you should try to reinstall owhttpd and start that. If that does start, it should provide a web interface on port 2121 of your pi. It that is also not starting, we need to figure out how to debug this and see if it is indeed failing hardware. But that would be somehow strange after an update...

And could it be that my installer has changed your device settings in the /etc/owfs.conf settings?

And I noticed that systemd sucks with owserver. So make sure you run a couple of times `service owserver stop` to make sure it is stopped and dead... then do a restart or run manually the command" `/usr/bin/owserver -c /etc/owfs.conf --pid-file /var/run/owfs/owserver.pid`

Natrixz, 2018-11-21:
When I really make sure it's killed and then manually run the command, I get the same results (strangely enough)
I compared it to a different Pi that I haven't updated and the file in /etc/owfs.conf looks exactly the same.

When I get home in the afternoon I will swap out the hardware to see if that's the thing that failed.
I have both a new one and one in use I know works so thats an easy check.

theyosh, 2018-11-21:
DO NOT TOUCH IT ANYMORE :) :) :)

But this is very strange.... I have had those situations also.... My experience is that they are rare... But again, I am not a hardware specialist... but I think it has to do with the electronics somehow..

Natrixz, 2018-11-21:
Well I cloned the SD-card before doing anything, but I can't seem to replicate it on my other Pi, weird in itself...
Must have been residual power in some of the components or something....Can't imagine any other issue if it's not present on the cloned SD-card >.<'
It's still electronics in a very humid environment (sensors that is) so who knows

Natrixz, 2018-11-21:
Closing due to can't reproduce.

closed 2018-11-21 by Natrixz

5point0, 2019-01-30:
Hi Yosh

I get this error after update:
WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device! If this is not correct, try 'i2cdetect -y 1' to see if device is connected.         

 and this when i2cdetect -y 1:
i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --         

Any idea whats wrong? I have reinstalled 2 or 3 times.

From what I can gather this has to do with one wire devices and I have two DS18B20 devices.

theyosh, 2019-01-30:
If you have only DS18B20 devices, you should wire them as 1-wire and use 1-wire option.

OWFS is only available for I2C Rpi boards/hats.

If you do have an I2C Rpi hat, then this indicates that it has a power issue, maybe short circuit? I had this in the past. Just poweroff the Pi and re-atach the I2C hat. And then power on again.

5point0, 2019-01-30:
Thanks, I have wired them as the 1-wire method and its working.

5point0, 2019-02-04:
Seems after the update the DS18B20 sensors stop working after some time and i get an error on them. Is there something specific i need to do after the update on these?

theyosh, 2019-02-04:
Please paste the error here. And mostly (95%) it is not my update, but a hardware issue. So I guess shortage or died hardware.

And could you paste the outcome of the following command:

`cat /boot/config.txt | grep w1`

5point0, 2019-02-11:
Here is the log with the errors:

2019-02-11 22:54:00,320 - ERROR   - terrariumEnvironment - Environment temperature sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2019-02-11 22:53:53,600 - INFO    - terrariumEngine      - Update done in 2.65754 seconds. Waiting for 27.34246 seconds for next update
2019-02-11 22:53:52,673 - WARNING - terrariumSensor      - Measured value NoneC from w1 sensor 'Temp probe' is outside valid range -10.00C - 85.00C in 0.00012 seconds.
2019-02-11 22:53:52,548 - INFO    - terrariumSensor      - Updated dht22 sensor 'Temp Top' from 25.90C to 25.90C in 0.00016 seconds
2019-02-11 22:53:52,427 - INFO    - terrariumSensor      - Updated dht22 sensor 'Humidity bottom' from 54.60% to 54.60% in 0.00017 seconds
2019-02-11 22:53:52,305 - WARNING - terrariumSensor      - Measured value NoneC from w1 sensor 'Temp prob 2' is outside valid range -10.00C - 85.00C in 0.00012 seconds.
2019-02-11 22:53:52,179 - INFO    - terrariumSensor      - Updated dht22 sensor 'Humidity Top' from 68.70% to 68.80% in 0.52668 seconds
2019-02-11 22:53:51,532 - INFO    - terrariumSensor      - Updated dht22 sensor 'Temp Bottom' from 26.90C to 26.80C in 0.52598 seconds
2019-02-11 22:53:45,323 - INFO    - terrariumEnvironment - Update done in 0.01066 seconds. Waiting for 14.98934 seconds for next update

And output for cat /boot/config.txt | grep w1
dtoverlay=w1-gpio

If open a screen session and restart the software the sensors come back. Any idea what this could be?

theyosh, 2019-02-12:
If you do a `dmesg` on your terminal. Do you get [ 7783.434613] Under-voltage detected! (0x00050005)
message? That could indicate a power problem.

And when the problems are rising, what does the following give you in the terminal:

`ls /sys/bus/w1/devices/ -l`

There it should show the 1wire sensors like:

```
lrwxrwxrwx 1 root root 0 Feb 12 18:17 28-0115b231f3ff -> ../../../devices/w1_bus_master1/28-0115b231f3ff
lrwxrwxrwx 1 root root 0 Feb 12 18:17 28-0315b2966fff -> ../../../devices/w1_bus_master1/28-0315b2966fff
lrwxrwxrwx 1 root root 0 Feb 12 18:13 w1_bus_master1 -> ../../../devices/w1_bus_master1
```

First two lines are the sensors. The last is the communication bus. If there are no sensors when the problem arises, it has to do with the OS or power I guess. Because I read the sensors from that location as it is provided by the OS

5point0, 2019-02-12:
I cant seem to find any "Under-voltage detected!" errors with the dmesg.

If i punch in ls /sys/bus/w1/devices/ -l i get:

lrwxrwxrwx 1 root root 0 Feb 12 21:41 28-02099177b694 -> ../../../devices/w1_bus_master1/28-02099177b694
lrwxrwxrwx 1 root root 0 Feb 12 21:41 28-020a9177fa83 -> ../../../devices/w1_bus_master1/28-020a9177fa83
lrwxrwxrwx 1 root root 0 Jan 31 22:17 w1_bus_master1 -> ../../../devices/w1_bus_master1"

So I see both sensors here and they are currently not showing in the software. I only have two of these, two DH22 sensors and a relay with a decent power supply.

I also get this if i punch in "cat w1_slave" under one of the sensors directories:
c4 01 55 05 7f 7e 81 66 9e : crc=9e YES
c4 01 55 05 7f 7e 81 66 9e t=28250

This value changes if i try again in a few minutes.
c3 01 55 05 7f 7e 81 66 4e : crc=4e YES
c3 01 55 05 7f 7e 81 66 4e t=28187

theyosh, 2019-02-12:
And this is all done as user pi or as user root? And you are saying that when you looked at those directories, my software had errors reading them??? that is not good :(

5point0, 2019-02-12:
As pi user.
Yes, when I looked at the directories the software showed an error on those sensors and the directory worked but about an hour later I tried again under the directory and its gone and also gone from devices list. 

I see this under dmesg now but still no "Under-voltage detected!":
```[ 7362.499526] w1_slave_driver 28-020a9177fa83: Read failed CRC check
[ 7363.689538] w1_slave_driver 28-02099177b694: Read failed CRC check
[ 7392.339803] w1_slave_driver 28-02099177b694: Read failed CRC check
[ 7899.469386] w1_master_driver w1_bus_master1: Attaching one wire slave 28.02099177b694 crc 03
[ 7899.529530] w1_master_driver w1_bus_master1: Attaching one wire slave 28.020a9177fa83 crc 40
[27597.293319] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[269788.157005] w1_slave_driver 28-020a9177fa83: Read failed CRC check
[269815.447189] w1_slave_driver 28-02099177b694: Read failed CRC check
[269850.687415] w1_slave_driver 28-020a9177fa83: Read failed CRC check
[269852.047423] w1_slave_driver 28-02099177b694: Read failed CRC check
[270291.075059] w1_master_driver w1_bus_master1: Attaching one wire slave 28.02099177b694 crc 03
[270291.125060] w1_master_driver w1_bus_master1: Attaching one wire slave 28.020a9177fa83 crc 40
[809414.294608] w1_slave_driver 28-020a9177fa83: Read failed CRC check
[809414.954613] w1_slave_driver 28-02099177b694: Read failed CRC check
[809438.194784] w1_slave_driver 28-020a9177fa83: Read failed CRC check
[809438.864789] w1_slave_driver 28-02099177b694: Read failed CRC check
[809476.835072] w1_slave_driver 28-020a9177fa83: Read failed CRC check
[809477.895079] w1_slave_driver 28-02099177b694: Read failed CRC check
[810230.945528] w1_master_driver w1_bus_master1: Attaching one wire slave 28.02099177b694 crc 03
[810241.225600] w1_master_driver w1_bus_master1: Attaching one wire slave 28.020a9177fa83 crc 40
```

So assume there is something wrong on my side since these sensors come and go??

theyosh, 2019-02-12:
[809476.835072] w1_slave_driver 28-020a9177fa83: Read failed CRC check
[809477.895079] w1_slave_driver 28-02099177b694: Read failed CRC check
[810230.945528] w1_master_driver w1_bus_master1: Attaching one wire slave 28.02099177b694 crc 03
[810241.225600] w1_master_driver w1_bus_master1: Attaching one wire slave 28.020a9177fa83 crc 40

Those message looks like bad wiring... so I think you have to check your wires....  because the OS is losing the connection and then it comes back. That is the cause of this problem I am afraid