theyosh/TerrariumPI #214: Control Linear Actuator?

Swiftnesses, 2018-11-21:
Hello,

Firstly, thank you for creating this awesome software!

I'm going to use it control my Chicken Coop (and even sprinklers perhaps!).

The coop is operated via 12V linear actuator (via a motor control board), can this currently be controlled, or can support be added?

I'm using this board: https://github.com/sbcshop/motor-shield

theyosh, 2018-11-21:
Thanks!

The motorshield is never going to work or supported. I would suggest to use: https://shop.sb-components.co.uk/collections/featuredproducts/products/pirelay-relay-board-shield-for-raspberry-pi

Which is much easier to use. It can just handle your 12V also. And that board will work out of the box with my software. The motorshield not.

So, not sure if it is all supported, but I would recommend to use relay boards for powering devices rather then motor board/shields

theyosh, 2018-11-21:
> Hi,
> 
> I use the shield because the sprinklers require a min 80ms pulse (positive and negative). Can this be achieved with any of the boards you're aware of?

It is unclear what this means, but looking at the code at https://github.com/sbcshop/MotorShield it needs a lot more then just open or close a relay. Therefore, I cannot develop this. It is to much code for just powering a sprinkler system. As I am just spraying my system with just powering on and off the spray system.

The only way this can be supported if you can provide working python code, so that I can see if I can incorporate in the software. But I do not have the hardware, and developing software without the ability to test, is hard to do.

Swiftnesses, 2018-11-21:
Hi @theyosh 

Sorry for not being clear. Basically I use the motor controller because to open the latched solenoid on the sprinkler valves I need to open them with +12V, to close them, I apply -12V (simple reversing polarity like you would on a motor to spin it the other direction).

It's as simple as that really, not sure if there's an alternative way of doing this that would be compatible with your software. I noticed the above library has some Python classes, examples etc, but I'm not a coder :(

Thanks in advance.

theyosh, 2018-11-21:
Yeah, but again, I cannot test it. So that would be hard to make. I can make a test version for it. But that can take some time.

Are you able to do some testing? In other words, if I provide some code, can you hookup the hardware on your Pi and test it?

Swiftnesses, 2018-11-21:
For sure, I can do all the testing you need!

It would save me a bunch of work and combine 2 systems into one, so I'm happy to help where I can.

I know the board can control motors, and has the option to read some sensors / control the onboard LEDS, all of which are useful, but it's the 'motor control' (really just switches?) that I need more than anything...

To confirm, to turn on a sprinkler currently I just issue a 100ms pulse (can be longer obviously) of either +12V or -12V.

Thanks so much for looking into this for me.

theyosh, 2018-11-21:
Ok, just to be clear. I have not worked with those devices. When you put on the power/action/pulse it goes open/start to sprinkle. But what happens when the power is dropped. That will not close the sprinkle system. In other words, I just give it some juice and it will start working. Then it needs a negative juice to stop. I am trying to figure out, if the power is needed all the time, or that only sending a pulse is enough. Because for motors, you have to send the power continuously when you want to move. I guess for your sprinkles that is not needed?

Because I do have my own logic for timers for spray systems.

Swiftnesses, 2018-11-21:
The solenoids are 'latched' (the technical term), they don't require power once they've changed state - obviously you can keep them powered, but that generally burns out the solenoid coil, so a pulse is the only thing required. The board can control motors, which would obviously require constant power as you said, not sure if it could be flexible, i.e. a toggle to select constant, or timed.

theyosh, 2018-11-21:
I think I have enough info to get started... But I do have another 'requirement' ;) That you post pictures when it is running at #210 :)

Swiftnesses, 2018-11-21:
You got it!

theyosh, 2018-11-21:
Hmm, maybe we mist something. But the motor board does not change the voltage from +12V to -12V. It does cross the wires (polarity) so that the power is flowing the other way.

I have looked in the code and basically they open a GPIO pin and that flows power. But it will never go negative. So I assume that it is based on polarity and not negative voltage

theyosh, 2018-11-21:
Yes... but the better I understand it, the easier it will be to make the software... so now worries

Swiftnesses, 2018-11-22:
Agree, can't wait for this, thank you again.

theyosh, 2018-11-22:
If you do:
- `git pull`
- `git checkout solenoid`
- restart software
- clear browser cache

There should be a new switch type called solenoid. The address value is the following:
`[energie_pin],[forward_pin],[reverse_pin]`
The values can be found at: https://github.com/sbcshop/MotorShield/blob/master/PiMotor.py#L25

It is all untested..... so lets see what happens

Swiftnesses, 2018-11-22:
Superb!

I just left home for a business trip, I'll test as soon as I'm back... Looking forward to it :)

Swiftnesses, 2018-11-22:
@theyosh Is there a way to control the pulse length, it's not good to constantly power the solenoids (latched), i.e. can we select the pulse length via an option, perhaps it would be better longterm to have an option to either send either constant power, or send a pulse of user designated length?

Swiftnesses, 2018-11-24:
Hey @theyosh,

Coming back to this I thought I'd try this with a new PiRelay I just bought.

Seems to work great, with exception I need to toggle 2 switches in the UI to change the actuator direction (which reverses polarity to the actuator). 

Is there a way around this, one of the actuators is very heavy duty and thus I think relays would be better than the motor driver, worst case I can always use 2 buttons...

theyosh, 2018-11-24:
You mean to have to double press? Could it be that you need to enter values other way around, or change the cables.

I do not understand 'two switches'. Have you made two switches in the settings?

Swiftnesses, 2018-11-24:
I have 2 seperate switches in the UI (needed to reverse the polarity with relays)...

theyosh, 2018-12-03:
any updates? I would like to add this feature... But I am a bit stuck with your remark about the relay and second switch

Swiftnesses, 2018-12-04:
I'll take a look when I rebuild the controller, I moved to relays temporarily as you suggested, for sure though I'd like to swap back to the motor controller as it's simpler.

theyosh, 2019-07-05:
any updates on this?

theyosh, 2020-01-28:
Closing due to no response in 1 year

closed 2020-01-28 by theyosh