theyosh/TerrariumPI #219: Cannot add DHT11

Swiftnesses, 2018-12-01:
Device works fine using the Adafruit examples, adding via the Terrarium UI gives an error:

```
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 862, in _handle
    return route.call(**args)
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 1740, in wrapper
    rv = callback(*a, **ka)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 47, in webserver_headers
    return fn(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 85, in wrapper
    return func(*a, **ka)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 272, in __update_api_call
    result['ok'] = self.__terrariumEngine.set_config(path,postdata,request.files)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 1109, in set_config
    update_ok = self.set_sensors_config(data)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 670, in set_sensors_config
    self.__load_sensors(data)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 184, in __load_sensors
    self.__unit_type)
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 317, in __init__
    sensorid = self.get_address().upper() + self.get_type()
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 567, in get_address
    return self.sensor_address
AttributeError: 'terrariumSensor' object has no attribute 'sensor_address'
```

theyosh, 2018-12-02:
New pull will fix this issue

theyosh, 2018-12-03:
Cool... could you also look at https://github.com/theyosh/TerrariumPI/issues/210 ?