theyosh/TerrariumPI #222: Support webhooks

Swiftnesses, 2018-12-04:
Hi,

I deployed my Duck COOP / Garden Irrigation controller today (pictures to follow). 

I'd like to integrate it further with my smart home system if possible... Currently I poll the API for the status of things, like door switches and temperatures. Ideally we could implement simple webhooks for actions / state changes, is this possible?

Swiftnesses, 2018-12-06:
Looking into this, wondering what the best way to update my smart home is, rather than poll the switches etc via a HTTP call (would rather have realtime updates, if possible!).

theyosh, 2018-12-06:
Hmm, what I can make is that you can specify 1 url where I then post messages to. This will be configured in the notifications area. It will not post sensor changes. That will be a post action every 30 seconds for every sensor.

So, updates of the sensor state should always be polled. But power switch changes, door changes, sensor alarms will be posted. Look at the notifications page for what notifications there are currently available.

Swiftnesses, 2018-12-06:
My home automation allows sockets like this:

/[event]/[value]

I can then make flows of the information included in the URL, i.e. event and value.

i.e. event = coopdoor value = on, then...

theyosh, 2018-12-06:
That is not working for me. That will introduce a lot of code in order to make that work. And the next person some other urls or data

So, I can only make a system that post data to 1 url. You could make some PHP code that translates it for your setup. That is how far I can go.

Swiftnesses, 2018-12-06:
I can make it work, do your magic my friend :-)

I'm ready to test as this means I can get statuses of the coop quickly, which is a winner!

theyosh, 2018-12-06:
Post data will look this:

'name':'Deur links'
'hardwaretype':'gpio'
'state':'closed'
'address':'38'
'now':'Thu Dec 6 20:14:48 2018'
'id':'a45610bf624765c23d0fee785a9d8f32'

But this depends on what triggers it. Above was a door that closed.

Swiftnesses, 2018-12-06:
It's JSON or something?

theyosh, 2018-12-06:
Almost... but I can make it JSON...

Swiftnesses, 2018-12-06:
You're the man, I can work with that :)

theyosh, 2018-12-06:
Ok, add new code for this... You now the drill :P . `git pull`, restart, clear browser cache, go to notifications  settings.

There you can enter a full url where the data should be posted to. On the receiving server I tested with the following PHP code:

```
<?php
ob_start();

$raw_json_data = file_get_contents('php://input');
$php_object = json_decode($raw_json_data);

echo($raw_json_data . "\n");

var_dump($php_object);
print_r($php_object);

$data = ob_get_contents();
ob_end_clean();

mail('[YOUR_EMAIL]','test',$data);
?>
```

Will produce the following output:

```
{"water_flow": 0.1, "timer_on_duration": 0, "current_water_flow": 0.1, "timer_off_duration": 0, "hardwaretype": "wemo", "timer_start": "00:00", "timer_enabled": false, "current_power_wattage": 33.0, "state": 100, "address": "192.168.5.55", "id": "52bf5188168c87f6a2389c35a7a189b7", "now": "Thu Dec  6 22:45:30 2018", "timer_stop": "00:00", "power_wattage": 33.0, "name": "wemotest"}
object(stdClass)#1 (15) {
  ["water_flow"]=>
  float(0.1)
  ["timer_on_duration"]=>
  int(0)
  ["current_water_flow"]=>
  float(0.1)
  ["timer_off_duration"]=>
  int(0)
  ["hardwaretype"]=>
  string(4) "wemo"
  ["timer_start"]=>
  string(5) "00:00"
  ["timer_enabled"]=>
  bool(false)
  ["current_power_wattage"]=>
  float(33)
  ["state"]=>
  int(100)
  ["address"]=>
  string(12) "192.168.5.55"
  ["id"]=>
  string(32) "52bf5188168c87f6a2389c35a7a189b7"
  ["now"]=>
  string(24) "Thu Dec  6 22:45:30 2018"
  ["timer_stop"]=>
  string(5) "00:00"
  ["power_wattage"]=>
  float(33)
  ["name"]=>
  string(8) "wemotest"
}
stdClass Object
(
    [water_flow] => 0.1
    [timer_on_duration] => 0
    [current_water_flow] => 0.1
    [timer_off_duration] => 0
    [hardwaretype] => wemo
    [timer_start] => 00:00
    [timer_enabled] => 
    [current_power_wattage] => 33
    [state] => 100
    [address] => 192.168.5.55
    [id] => 52bf5188168c87f6a2389c35a7a189b7
    [now] => Thu Dec  6 22:45:30 2018
    [timer_stop] => 00:00
    [power_wattage] => 33
    [name] => wemotest
)
```

I hope this is something you can work with..

Swiftnesses, 2018-12-06:
Will check it out... Looks like I need to switch my brain on for this.

Swiftnesses, 2018-12-07:
Hi @theyosh,

That does really help I think :-)

I'm not that experienced with PHP, can you point me in the direction of how I can see the output with the PHP script you posted? I should be able to take it from there...

Swiftnesses, 2018-12-07:
@theyosh it would be easier if the %name% was the same as the API, i.e. "switch", temperature
humidity, distance, moisture, conductivity, ph, light, fertility etc?

Swiftnesses, 2018-12-07:
On testing this, it seems my home automation system can either be triggered on a GET or POST, do you know if the data is currently sent via POST? I can't get it to trigger :(

Swiftnesses, 2018-12-07:
Okay, an update.

I can get it working using the fixed notification URL, but not the /%name%/%state% option right now.

I'm testing using the PHP script you posted to set things up, sadly I cannot get a notification when the door state changes (open/close), even though I have it selected, switches work fine, bug?

Swiftnesses, 2018-12-07:
Got it working!

Think my tinkering killed something though, a reboot fixed it... (the door stopped updating!).

ck(self.get_data())
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 742, in toggle_door_status
    self.notification.message('door_toggle_closed',data)
  File "/home/pi/TerrariumPI/terrariumNotification.py", line 650, in message
    self.__ratelimit_messages[title][now] += 1
KeyError: '1544196960'
2018-12-07T15:37:03Z <Greenlet "Greenlet-0" at 0x6d083e00: <bound method terrariumDoor.__checker of <terrariumDoor.terrariumDoor object at 0x6c5c2890>>> failed with KeyError

theyosh, 2018-12-07:
> @theyosh it would be easier if the %name% was the same as the API, i.e. "switch", temperature
> humidity, distance, moisture, conductivity, ph, light, fertility etc?

I don't think so. You have only two variables. So when you add switch to it, you do not know which switch. Or you do know the switchname, but then you can't add a switch value to the url.

But, I am a bit confused regarding your last messages. Do you have errors in the code when the door is toggled? I think that there was a small bug, which is fixed now. Do a git pull to get the latest code.

For the PHP code it is just an example.

The message "Max total messages per minute 5 reached" should influence this webhook code. So you can ignore it for now. Also the JSON message is debug and can be ignored.

The requests are always in POST mode. Because I will sent data with the request for further processing.

Swiftnesses, 2018-12-07:
I have it working, using the 'constant' url method, I tried the variable route but wasn't sure which options were possible and my home automation system doesn't expose the call - so that sucked. I just filtered the JSON and made sure my switch / doors etc had unique names.

Swiftnesses, 2018-12-17:
@theyosh As you're refactoring, have you considering adding MQTT to notifications? It would be a great companion to Websockets and is super lightweight and popular in the smart world, I know there's already libs out there that would make this an 'easy' integration...

fenyx13, 2018-12-19:
Could I use this to link the TerrariumPI sensor data to home assistant? (Hass.io running on another Pi)

Swiftnesses, 2018-12-19:
@fenyx13 Yes you could! HA has a websocket module I believe...

theyosh, 2018-12-19:
Not for temperature sensors. The update every 30 seconds, and that would produce a lot of data. But you can read them out with: https://www.home-assistant.io/components/sensor.rest/ and https://github.com/theyosh/TerrariumPI/wiki/API

theyosh, 2018-12-19:
@Swiftnesses for MQTT please open new issue.... this getting long...

theyosh, 2018-12-28:
For now I close this issue. If you want MQTT support, please open a new issue. But it will be at a low priority for now.

closed 2018-12-28 by theyosh