theyosh/TerrariumPI #223: Webcam stream (as option)

Swiftnesses, 2018-12-04:
Hey @theyosh,

Loving this controller, mega work my friend - thank you.

I'm currently using MJPG Streamer (available here on Github) to stream my Duck COOP webcam, rather than use static images. On my Pi 3 (A+ with only 512MB RAM), I can stream and use the controller while only using around 15% CPU (which is amazing tbh!). I doubt this is even an effective method of streaming - likely modern browsers support much better alternatives (H264/265?).

I'd love to have the stream inside Terrarium Pi rather than running via another application. Could you please add streaming as an option, given the low CPU utilisation, it seems a real possibility which would hugely improve the options available to people wanting live updates (I need them, as there's a small chance a Duck could get caught in the door, sure this applies to reptiles too?).

Many thanks,

Mark

Swiftnesses, 2018-12-04:
Seems like UV4L offers amazing streaming with low CPU usage and Pi cameras 👍

theyosh, 2018-12-04:
The low cpu is due to low quality streaming. Both is [motion JPEG](https://en.wikipedia.org/wiki/Motion_JPEG) And that is only supported in Firefox and Chrome. I think Safari can't handle it. So on iPhone it does not work well.. Also, the bitrate will be very high, and works well on a LAN. But going over the internet, you need a rather steady and fast upload speed.

So there is H264 and H265. H265 is not mature enough, so that is out of the picture. H264 is currently industry standard. But converting a stream to H264 cost a lot of CPU power. The Pi is able to do it... but will eat all CPU cores. So that is not an option.

I found a solution to use the Pi GPU for H264 encoding. You can mannully compile FFMPEG with --enable-mmal --enable-omx --enable-omx-rpi and then you can use the hardware encoder of the Pi. Which works pretty well. I have a Wifi webcam in a (birdhouse)[https://vogelhuisje.theyosh.nl/]. It is producing a multibitrate HLS stream. This takes about 1 CPU due to ffmpeg keyframe alignment and creating HLS chunks. But compiling ffmpeg takes about 1.5 hours. So not an option for normal users.

So it is a bit more complicated. And I do know, as my job is video transcoding and streaming for a company. So I have more then 10+ years experience in video encoding.

So, it is not that easy. Also both uv4l and mjpeg streamer have there own webserver and different ports that also needs to be forwarded.

And currently, I tile the webcam images for zooming like Google Maps. That is not possible with a live stream. And therefore the webcam controls should be changed...

What I can support is this: https://medium.com/@jaredtoddpetersen/home-security-live-streaming-with-raspberry-pi-f9293efca7ba

This will produce the same type of stream as my birdhouse. And that can be added as a webcam so that you have a livestream in my interface. But this will also need some more linux experience in compiling the software you selfs.

theyosh, 2018-12-04:
Ah, all browsers support motion jpeg now a days

theyosh, 2018-12-04:
Alternative: https://raspberrypi.stackexchange.com/questions/7446/how-can-i-stream-h-264-video-from-the-raspberry-pi-camera-module-via-a-web-serve

Swiftnesses, 2018-12-04:
Thanks for the explanations, really helps me understand. Keeping it light, with as little extra bloat as possible would be the best solution, I'm streaming now at 1296x972 @ 30FPS in H264 on my Pi 3 and it's not using much CPU at all...

I'm not bad with Linux, but it would be awesome to have an option built in to activate streaming... the last link you posted sounds interested, fairly lightweight and compatible...

theyosh, 2018-12-04:
How are you currently watching? Using vlc? Or using the WebRTC option?

Swiftnesses, 2018-12-04:
Right now I'm using http://raspberrypi:8080/stream/video.h264

theyosh, 2018-12-04:
that does not work in a browser right? That is playing in VLC?

Swiftnesses, 2018-12-04:
Works in a browser ;-)

I'm using UV4L...

theyosh, 2018-12-04:
Ok... cool... but I think I will try another way. I am currently compiling ffmpeg manually with the right codecs. And will test if that will produce HLS. That can be added to my software then...

But this will take some time to develop. And it will involve some manual labor ;)

theyosh, 2018-12-04:
And with an usb audio device and a mic you should be able to hear them also!

Swiftnesses, 2018-12-04:
Hearing them would be very cool :-)

I'd like to add another camera in addition to the Pi V1 camera I use, any suggestions?

theyosh, 2018-12-04:
Following: https://github.com/jaredpetersen/raspi-live#ffmpeg (so, manually compile ffmpeg)

created a ffmpeg with HLs support and H264 hardware. Now testing full Hd from the rpi cam to hls. cpu load 1% of 1 core :) This looks promising.. 

running:

`raspivid -b 1000000 -o - -t 0 -w 1920 -h 1080 -fps 30 -g 30 | ./ffmpeg -nostdin -re -i - -vcodec copy -f hls -hls_time 1  -hls_list_size 3 -hls_delete_threshold 5 -hls_flags split_by_time+delete_segments+second_level_segment_index -strftime 1 -hls_segment_filename '/home/pi/TerrariumPI/static/hls/file%s-%%d.ts' /home/pi/TerrariumPI/static/hls/stream.m3u8`

watch url: http://pi:8080/static/hls/stream.m3u8 (vlc or safari/iphone) Will add a hls player to the webcam part.

Swiftnesses, 2018-12-04:
Nice! 👍

Swiftnesses, 2018-12-05:
How did you get on, I'd like to try your stream method and check out utilisation....

theyosh, 2018-12-05:
Ok, turns out... it is pretty easy to have a live stream now... Here we go!
- install ffmpeg: `sudo apt install ffmpeg`
- update my software: `git pull`
- restart my software: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-restart-terrariumpi
- clear browser cache
- add new webcam with location: `rpicam_live`

And that should give you a live webcam in the webcam sections. It is currently missing all zoom controls. Also archiving and rotation is not yet working.

And it is still alpha. But I got it working with 1.0% CPU usage :)

Swiftnesses, 2018-12-05:
Testing now... :)

Swiftnesses, 2018-12-05:
Okay, got the live stream... 2 issues I can see here:

1) My ducks are sideways, which looks uncomfortable (they need rotating!) :-)
2) I normally use 1296x972 on my V1 camera (Full FOV and fast FPS) - https://picamera.readthedocs.io/en/release-1.12/fov.html

For some reason changing the resolution has no effect, currently it's cropped their heads off...

Thanks for getting this working though, superb!

theyosh, 2018-12-05:
Yeah, that should be possible. But somehow my camera just stopped working ... :( And it is getting late here.. so tomorrow is another day.

theyosh, 2018-12-05:
Ok, found the issue... it is getting late so stupid errors...
Anyway, resolutions and rotations should work now

theyosh, 2018-12-05:
Only the FPS is currently locked at 30 fps....

Swiftnesses, 2018-12-05:
At this rate, my ducks will be getting more sleep than us :-)

I'll give it a test now and report back, get some sleep! :-)

Swiftnesses, 2018-12-05:
To confirm, do I just do a pull, and then restart?

Swiftnesses, 2018-12-05:
Hmmm, not working here, it now only plays for 5 seconds when I go to webcam (seems to be the same clip), no rotation or resolution change either?

theyosh, 2018-12-05:
Just wait... do a new pull now. Just made an error..
and yesh. Do a git pull and restart.... that should do it

Swiftnesses, 2018-12-05:
Did a pull, rotation / resolution still not playing here dude - sorry :-(

theyosh, 2018-12-05:
hmmm... crap.. I have it working at the moment. Sorry for now... tomorrow I will try again

Swiftnesses, 2018-12-05:
Hang on, one final cache clear and reboot of the app, and it's working! Good work.

theyosh, 2018-12-05:
Ok, new update. This will write the files to memory so it will not hurt the SD card. 

But you have to remove a folder by hand before restarting. So look at the location TerrariumPI/webcam/ There should be a folder like a94df035fc91a158c1f9d7a1c769ca66

That folder needs to be removed. It is possible to pull the new code first, and restart the software. During the 10 sec countdown, you have time to remove the webcam folder. Then it will be recreated with different settings.

Else you could get some errors.

Swiftnesses, 2018-12-05:
[image]

Seems to work just fine ;-)

theyosh, 2018-12-05:
Donald just woke up :) :D

Swiftnesses, 2018-12-05:
It's crazy seeing them, one just tried to fly inside the coop, sometimes it's better not to see them ;-)

Swiftnesses, 2018-12-05:
Now I just need to work out how to get the websockets working behind my proxy (nginx), it's serving okay, but switch statuses etc don't work when going via proxy pass.

theyosh, 2018-12-05:
https://github.com/theyosh/TerrariumPI/blob/master/contrib/nginx.conf#L3-L6 and
https://github.com/theyosh/TerrariumPI/blob/master/contrib/nginx.conf#L22+L23

Swiftnesses, 2018-12-05:
Sorted it ^^^

Swiftnesses, 2018-12-06:
Hey @theyosh this works great with sa Pi Cam - but I seem to have a delay of 5-6 seconds, any idea why?

theyosh, 2018-12-06:
@Swiftnesses the playlist consists of chunks of video of 2 seconds each. The list has 3 chunks, so 6 seconds of video. It could be that the player is having some trouble loading it, and will pause a bit. And that will increase the delay a bit. There is for about 12 seconds of video on disc. That is that the max delay. But 10 seconds is not a bad value.

@fenyx13 this technique is not suitable for usb or remote webcams. For that to work, a manually compiled ffmpeg is needed with special Pi H264 hardware support. That takes about 1 hour to compile and is rather technical. And then it 'could' work. There are more factors that can cause issues with USB or remote camera's. Also, that will also disable the use of the raspicam. You can either use the H264 hardware chip of the pi to watch H264 of create. But not both. So I will not make it for USB cams, as that is more complicated and goes beyond my support.

On the other hand, I am adding code to support HLS streams as webcams. Then there is a possibility to try for your self to install all the needed software. But I cannot give support on that.

Manual compiling according here is needed: https://github.com/jaredpetersen/raspi-live#ffmpeg

Swiftnesses, 2018-12-06:
Hmmmm, the issue for me is that 10 seconds means I could have squished a duck before I realise :-)

theyosh, 2018-12-06:
Sorry, but  if you need UV4L, you have to implement your self. Keep in mind, that I need to make software that works for everybody. And not to complicated. UV4l has its own webserver, which introduce new ports that needs to be opened on the router if you want to use it over the internet. So that is not an option.

HLS Technique: https://en.wikipedia.org/wiki/HTTP_Live_Streaming

I will never make the claim that my software is able to use as a kind of security system. It is a home automation system. For mission critical stuff is my software not suitable

Swiftnesses, 2018-12-06:
@theyosh I can live with the 10 second delay if it can't be improved, the fact I can actually stream seamlessly is great, appreciate that!

theyosh, 2018-12-06:
Ok. then we leave it as is... it should be max 6 seconds which I still think is pretty nice

theyosh, 2018-12-08:
So, then this is fixed now? For non Raspicam devices this technique is not usable.

Swiftnesses, 2018-12-08:
Unless you can reduce the lag, we can call it a success!

I have 2 cameras that offer static jpg, I'm hoping to add them to my duck run in various places, remote is supported right @theyosh ?

Swiftnesses, 2018-12-08:
Also, I noticed there's a nice time overlay on iOS (but not on desktop), will you be adding archiving and overlays back?

Swiftnesses, 2018-12-08:
[image]

🥇

Swiftnesses, 2018-12-08:
Saving a RAW displays this, so does the webcam settings page? 

Not sure archive is working, I click it and nothing happens?

[image]

theyosh, 2018-12-09:
> Unless you can reduce the lag, we can call it a success!
> 
> I have 2 cameras that offer static jpg, I'm hoping to add them to my duck run in various places, remote is supported right @theyosh ?

Static jpg is supported. Just enter the full url as address

Swiftnesses, 2018-12-09:
> Error in console: 2018-12-08 16:34:58,101 - ERROR - terrariumWebcam - Raw image 'Coop Interior' at location rpicam_live is not available!
> This happens in the first 30 seconds after startup. After 1 minute full running, it should show the right image. This is due to the use of the internal webserver, which will be started as latest. Therefore, the first attempt to get a image of the webcam fails and show an offline message. This will disappear in about 30-60 seconds after startup

Happens to me all the time. I never get an image here, nor can I seem to access the archive, albeit perhaps there isn’t one if it isn’t working right...

Swiftnesses, 2018-12-13:
@theyosh I have 4 camera in my garden which output:

H264 Streams:
Path to feed : rtsp://192.168.1.220:8554/unicast
HLS : http://192.168.1.220:8554/unicast.m3u8
MPEG-DASH : http://192.168.1.220:8554/unicast.mpd

Is it possible to feed these in as a webcam?

kaju666, 2018-12-15:
> Ok, found the issue... it is getting late so stupid errors...
> Anyway, resolutions and rotations should work now

Updated today to 3.9.1 and in rpicam_live rotation not work for me.

theyosh, 2018-12-16:
Hi, I am currently a bit bussy with some code refactoring. So not sure how fast I can fix stuff right now.

If the camera was already running, then a restart is needed to get the rotation working. Else I do not have an idea why.

For remote cams, I do have a solution for: HLS : http://192.168.1.220:8554/unicast.m3u8
So that will be added in the next release

Swiftnesses, 2018-12-16:
@theyosh You're a bloody legend my friend, thank you!
@kaju666 Works great here, if I remember I had to restart as @theyosh said.

Things are working perfect, apart from my DHT22 which seems to have stopped reporting readings. I assume it's died.

Swiftnesses, 2018-12-17:
@theyosh Any idea when the next release is planned (even for testing...)...

kaju666, 2018-12-17:
unfortunatly not for me. Only rpicam works and can be turned 90 degrees. When i use rpicam_live it does not want to rotate even after restarts or readding camera.

theyosh, 2018-12-17:
@kaju666 That is correct and a bug. In the development version I have it working. Will get in the next release.

@Swiftnesses if you want to test: checkout the development branch. Keep in mind, that this is an upgrade, and needs the installer to be re-run. And not sure if everything will work at once.
Also, there will be some sensor data rewritten. And therefore, you can go back to the master if something is failing... So keep a local copy of the TerrariumPI folder before trying

Swiftnesses, 2018-12-17:
@theyosh I'm on it... wish me luck ;-)

Swiftnesses, 2018-12-17:
@theyosh Up and running will configure now and test (btw you have some default wemo plugs in switches), any idea when you'll try and take a first shot at HLS?

theyosh, 2018-12-17:
@Swiftnesses HLS is already working... maybe clear browser cache?

theyosh, 2018-12-17:
Else just add the full HLS url as source for the webcam. Other settings like resolution will be ignored

Swiftnesses, 2018-12-17:
No luck with HLS, even tried this test stream:

http://www.streambox.fr/playlists/test_001/stream.m3u8

The error is HLS:4

[14 more replies]

closed 2019-01-08 by theyosh