theyosh/TerrariumPI #227: Ph showing C after pH like this 6.360 C

rrgrow, 2018-12-14:
In dashboard in the environment section I just noticed the ph is showing results with a C for Celsius.

theyosh, 2018-12-16:
You are correct. Will be fixed in next release. Thanks for spotting!

rrgrow, 2018-12-23:
Thanks...
Also I also use more then one sensor in my system (2 air temp, 2 water temp, 2 hum, 2 ph and 2 distance) so when viewing in dashboard I only see what I have chosen in Environment with 3 decimal points which is great. 
Would it be easy enough so make sensor section 3 decimal point(s) too instead of rounded to the nearest decimal. Then I can read all sensor(s) accurately. 
thanks for your time.

rrgrow, 2018-12-23:
Works Perfect! Thanks

jornobe, 2019-02-07:
@rrgrow ; what sensor are you using for measuring Ph values?