theyosh/TerrariumPI #228: Sytanx error in TerrariumPi sudoers file?

Benjin, 2018-12-19:
After installing TerrariumPi, I'm unable to `sudo` anything.  For example, when I try running `sudo raspi-config` in order to open the RaspberryPi config dialog, I get the following error message:

```>>> /etc/sudoers.d/terrariumpi: syntax error near line 1 <<<
sudo: parse error in /etc/sudoers.d/terrariumpi near line 1
sudo: no valid sudoers sources found, quitting
sudo: unable to initialize policy plugin```

when I look at the `terrariumpi` file, it has this as its contents:

` ALL=(ALL) NOPASSWD: /usr/sbin/service pigpiod restart`

Is TerrariumPi installing an invalid sudoers file?  What should it be instead?

theyosh, 2018-12-19:
Hm, it is missing the userid in the start of the string. It should be:

`pi ALL=(ALL) NOPASSWD: /usr/sbin/service pigpiod restart`

You are missing the user pi. Are you installing it under a different user? It expect that you are logged in as a normal user, and run the installer with sudo ./install.sh . If you directly login as user root, I am not sure what will happen.

AndrejWeber, 2018-12-19:
I had the same problem, a file needed for sudoers. is corrupt. You have to fix it by hand. 
I changed it on PC. I will look later which file it was and tell you what you have to change. 

@theyosh it happens, when I try to install by using vnc. Then the Installer asks me if I want to install with the user „“...
There is no user, so the sudoers.d will change to blank user.
I have to install by ssh, then the installer asks me to install with user „pi“ and it works fine.

theyosh, 2018-12-19:
Ah, that explains.... So, when the name is empty, I can force the user pi. Then there would be more issues, due to setting the rights on files....