theyosh/TerrariumPI #229: I want this, can it be added? (new multi camera board!).

Swiftnesses, 2018-12-19:
I want to order this:

https://www.robotshop.com/uk/arducam-multi-camera-adapter-module-raspberry-pi.html

Would allow me a chance to have multiple camera angles (sure many might consider this for multiple live views?).

theyosh, 2018-12-19:
Those board will only support low quality video when all used together.

Also, you need to specify which camera you want to use before taking picture. So it is possible, but I would use it as a still image webcam rather live.
All 4 live at fullHD is not possible with this hardware.

But I would like to say that I will slow down the development a bit... it is still a hobby project. So do not expect this be added soon. Also, this is again hardware I do not own, and have to make the code without testing. So it will get a low priority

theyosh, 2022-04-08:
Hi,

this is a long standing issue, and I still have no time to implement this. As without the hardware it would be hard.

But, there is an option to still use this hardware. When you create your own script to take pictures of the different cameras you can use the Remote or Local file webcam option to load the images in TP4. But that means that you need to do some more work, and make some code. I am willing to help if you have found some working code to create your own scripts.

So, I am sorry to say, but this will not be implemented in TP4.

I would like to close this issue. If you still want to use this hardware, you can always reopen this when to need help.

theyosh, 2022-05-01:
Closing due to no further acion

closed 2022-05-01 by theyosh