theyosh/TerrariumPI #230: Intro Tutorial

stuartabrown, 2018-12-21:
Hi,

First of all, awesome project!  I installed on a RPi3 last night and all went smoothly.

I'm an experienced reptile keeper and reasonable in a few different programming languages but totally new to electronics.

As an intro to using TerrariumPi I would like to set up a heat and humidity sensor and use that to control a couple of different desk lights (ultimately of course these will be lights / humidifier in the vivarium, but I'll practice in a safe environment first!), however I'm unsure what hardware I need.

Going through the docs here I think I need:
- DHT22 - https://www.modmypi.com/raspberry-pi/sensors-1061/temperaturepressurehumidity-1066/temperature-and-humidity-sensor-am2302-dht22
- A relay - https://www.modmypi.com/raspberry-pi/relays-and-home-automation-1032/relay-boards-1033/8-channel-12v-relay-module

Is that it or am I missing a few fundamental things?  @theyosh if you're interested I'd be happy to try to write up a `total newbie's guide to TerrariumPI` documenting the process as I learn and add to the wiki once I get underway.

Benjin, 2018-12-24:
I'm in the middle of upgrading a kickstarted terrarium/aquarium tank where the company went MIA for warranty to TerrariumPI + RPi3b.  Before heading home for the holidays, I was able to get a temperature sensor (1Wire/I^(2)C), an irrigation system, and misters (both GPIO-controlled) to simulate rainfall (bare minimum for being away for a week, as I have external lights) working all through custom hardware, and I'm documenting everything along the way.

If I don't beat you to the initial version, I'm absolutely down to help flesh yours own.

theyosh, 2018-12-24:
Hi @stuartabrown 

My software is build in a general way, so that you can choose from the sensors and swiches on: https://github.com/theyosh/TerrariumPI/wiki/Hardware

The DHTXX sensors are not that stable. I have experienced a lot of faulty readings if you want to operate it 24/7. I suggest to use somekind of a I2C setup with for example:

https://taaralabs.eu/1-wire-humidity-temperature-sensor/ with https://taaralabs.eu/category/1-wire/ (I have not used this, but it is known to work.

Or look at: http://www.sheepwalkelectronics.co.uk/product_info.php?cPath=22&products_id=30 with http://www.sheepwalkelectronics.co.uk/product_info.php?cPath=23&products_id=55

Al the humidity sensors do have a temperature sensor.

The GPIO relay is fine and will work.

If you want to use this in a production system, it is wise to by every hardware at least twice or more so you can create a setup to test and have backup hardware when one is failing.

If you are able to make a small tutorial, that is great. I do not find the time to write it all out. As English is not my native language it takes me more time then I have...

Benjin, 2018-12-28:
> @stuartabrown <https://github.com/stuartabrown> how do you mean a
> tutorial branch? You will add that to my software? I was thinking of using
> something link https://medium.com/
>
> Else I would either fork my repository and create your own and add there a
> tutorial branch.
>
> And as latest, I can imagine you want access to this wiki, to add the
> tutorial there.
>
> Else I am not sure what you mean. What is your preference?

I think the will makes the most sense for a tutorial.

stuartabrown, 2018-12-29:
@theyosh I've cloned the wiki locally - it's actually a different repo to your main main codebase `https://github.com/theyosh/TerrariumPI.wiki.git` (you can se the details on a wiki page).  On my local I have a `tutorial` branch.  I was thinking that there might be a way for you to give me and @Benjin permission to commit to the wiki repo so that we can collaborate and then create pull requests for the tutorials to be pulled into the main/master wiki?

theyosh, 2018-12-29:
@stuartabrown that is not possible: https://help.github.com/articles/adding-and-editing-wiki-pages-locally/#cloning-wikis-locally-to-your-computer

> Once you have cloned the wiki, you can add new files, edit existing ones, and commit your changes. You and your collaborators can create branches when working on wikis, but only changes pushed to the master branch will be made live and available to your readers.

Only the master branch will/can be shown. Also, that means you have write rights to my repository. I am not a fan of that.

So, I still think a website called https://medium.com is more suited for it. Or https://www.instructables.com/howto/reptile/

I do appreciate that you want to make a tutorial. But it should not end up that you need full write rights to my repo. Normal code you can create pull requests and those I can import. But for the wiki, it does not work that way.

stuartabrown, 2018-12-29:
@theyosh no problem, happy with that.  I don't understand why the Github permissions don't allow owners to have separate permissions for the wiki and the main code - seems odd.  

If you don't mind, before going the Medium route I'll create a fork of your repo and then try making a pull request to your `documentation` branch with wiki tutorial stuff.  Should work as per https://help.github.com/articles/creating-a-pull-request-from-a-fork/

stuartabrown, 2018-12-29:
good grief, OK so that approach doesn't work for wikis!  They make collaborative documentation really tough!

I'll go the Medium approach.  Thanks for your patience!

stuartabrown, 2019-01-06:
Not really into the details yet but the first couple of tutorial posts are up at https://medium.com/@jinky32.  Waiting for some sensors to arrive before getting to the interesting stuff :)

theyosh, 2019-01-08:
Cool. I like your vivarium. Way cool below the stairs!

For the post on https://medium.com/@jinky32/setting-up-your-raspberrypi-and-installing-terrariumpi-97f2d2e8e6f I think it is handy to add something like:

> When the software is installed, you can access it in your browser at: http://raspberrypi:8090

The hostname 'raspberrypi' is the default and mostly it will register it self at your router with that name. I think adding the line above to your tutorial, is a nice closing of the installation, and you can also check if the installation did work well. Even without any sensors, it should give a working web interface.

stuartabrown, 2019-01-09:
@theyosh thanks :).  I've updated the post you mention and added a little bit extra too.

The sensors you suggested above 
https://taaralabs.eu/1-wire-humidity-temperature-sensor/ with https://taaralabs.eu/category/1-wire/  arrived today.  I can guess how these two items fit together, but could you point me in a direction where I can find out how https://taaralabs.eu/category/1-wire/ connects to my RPi3/TerrariumPI please?

Benjin, 2019-01-09:
Check this one out: https://tutorials-raspberrypi.com/raspberry-pi-temperature-sensor-1wire-ds18b20/

It shows how to wire it (you'll need a 4.7k ohm resistor) and how to confirm the pi can read it via the command line (no code necessary).

stuartabrown, 2019-01-10:
@Benjin thanks, will take a look.  The guys at TaaraLabs have suggested 
```
1-Wire board | Raspberry PI
GND          | 1 (GND)
5V           | 2 (5V)
3.3V         | 1 (3.3V)
D1           | 5 (SLC)
D2           | 3 (SDA)
```
for connecting the 1 Wire Shield https://taaralabs.eu/1-wire-shield-for-wemos-d1/ to the RPi3

Benjin, 2019-01-10:
Haven't tried using a shield yet.  Lemme know how well it works.

theyosh, 2019-01-12:
You should really try 1-wire: https://www.raspberrypi-spy.co.uk/2018/02/enable-1-wire-interface-raspberry-pi/ and https://www.raspberrypi-spy.co.uk/2013/03/raspberry-pi-1-wire-digital-thermometer-sensor/

The first url is enabling it for manual testing, my software does already that during installation. So you need only the second url for adding sensors.

It is really, really easy. Only 1 resistor needed and you can hook up X amount of temperature sensors. Will also by auto detected with my software

stuartabrown, 2019-01-13:
Thanks @theyosh I will look at those - I will certainly want to add multiple temperature sensors.

I've been looking at https://github.com/theyosh/TerrariumPI/issues/167 to see how to get a device (light for example) to turn on/off based on sensor readings.  Am I correct in thinking that my relay https://www.modmypi.com/raspberry-pi/relays-and-home-automation-1032/relay-boards-1033/8-channel-12v-relay-module should be connected to my RPi and then set as a switch in TerrariumPi and configured to turn devices on/off?

theyosh, 2019-01-13:
@stuartabrown yes. Those relay modules will work. The logic for switching based on sensors should be configured in the environment part at the heater or sprayer part

stuartabrown, 2019-01-13:
thanks @theyosh .  I hope you don't mind me asking all these questions, I appreciate they're all quite basic.

stuartabrown, 2019-01-15:
@theyosh haha no problem at all, small payback from me for all of your work!

stuartabrown, 2019-01-16:
Hello @theyosh . So I have my relay connected to my RPi and when I run this https://raw.githubusercontent.com/Mattrachwal/gpio-8channel-relay-test/master/gpio-relay-test.py it toggles through the inputs correctly and turns off the light that it connected to channel 1 (pin 26 in the code).
However when I add a switch and and set the GPIO address (I use 37 which I think is the physical address) without a timer the light doesn't turn on/off when I manually click it in GUI.
[image]
Any idea what I'm doing wrong here?

stuartabrown, 2019-01-17:
thanks @Benjin 
- I bought this relay https://www.amazon.co.uk/gp/product/B0794RF1ZW - 12v 8 channel
- Image attached.  Also a vide (crappy) at https://www.youtube.com/watch?v=o2wvCm8x_ls showing it running the script I mentioned above
[image]
- yes - hopefully you can also hear it in the video
- yes i do have a multimeter, not actually used one before (it came with a soldering kit I bought over christmas) but I can learn that.

theyosh, 2019-01-17:
Your GPIO address should be indeed around 40. So address of 37 should work. But it could be that there is a bug, that when you add it, it is not correctly registered. So add it, and just restart my software. If it then works, then we have found a software bug. If it still does not work after restart, then it is very strange and needs more investigation.

Also I have made a stupid code bug, and you should chose for `GPIO invert` to use this. As the example code does the opposite of my code. One day that will be fixed. But my GPIO does work that way. But still with normal GPIO it should make the clicking sounds when changing the state.

So add it, and restart and see what happens. Also you could try to debug a bit: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi and see if there are messages when you toggle the switch

AndrejWeber, 2019-01-17:
> Also I have made a stupid code bug, and you should chose for GPIO invert to use this. As the example code does the opposite of my code. One day that will be fixed.

There is no bug!
@stuartabrown you have to change the brown cable to lamp from nc1 to no1
it means "normaly open" and "normaly closed". Then the Lamp will also flasches up by using the test code and the switch in terrariumpi will also work as gpio (not inverted).

Can you also add a pic from above (raspi and relais board), so we can see, where you have connected the wires.

AndrejWeber, 2019-01-17:
It looks like green is connected from pin39 (GND) to IN1.
if it is, try the yellow wire to IN1, this is pin37 GPIO26

stuartabrown, 2019-01-17:
@theyosh @AndrejWeber many thanks for your help with this.

@AndrejWeber you are right - I am connected to nc1 rather than no1, but my RPi connection is from pin37.  It's not connected to the RPi atm so the below images won't show that but do show the nc/com connections.
[image]
[image]

Unfortunately my wife has just come home (obviously that is a wonderful thing ;-) ) so I have no more time to play tonight but I will set up asap tomorrow after work.

thanks again :)

AndrejWeber, 2019-01-17:
Now I am shure, ther is a connection problem!
wire green is connected to pin39 and IN1.
also you need to have a connection (5th wire) from pin39 (GND) to DC-
But you have to be shure, your 12V supply is connected right.
The video shows also a 5th wire, I belive to see a brown wire from pin39 to DC-

theyosh, 2019-01-17:
If I look at: https://www.amazon.co.uk/gp/product/B0794RF1ZW I do not see a ground connector. Only 12V in and out, and 8 inputs for the GPIO cables. No ground ..

Benjin, 2019-01-17:
`DC-` should be treated as ground, no?  In any case, I'm using this one, and have irrigation, misting, white, IR, and UV lights all being individually controlled: https://www.amazon.com/JBtek-Channel-Module-Arduino-Raspberry/dp/B00KTELP3I/

theyosh, 2019-01-17:
@Benjin Electronics is not my strongest part... but yours can be powered by the Pi itself. That makes it in my opinion easier to install. And yes, ground is ground... so the Pi needs also connected to the DC-

And not I also stop.... not clear thinking anymore.. ;)

AndrejWeber, 2019-01-17:
the relais board can also be supplied from raspi with 5V
so then it is pin2 or pin4 (5V) to DC+ and any GND on Raspi (like pin39) to DC-
when you use a second supply, like 12V supply, you have to connect both GNDs to DC-.
DC voltage does not exist with only one wire! you always need a positive voltage like 3,3V vrom GPIO37 and 0V/GND from pin39 to acting a switch.

stuartabrown, 2019-01-17:
ok I couldn't handle it and had to try the fix....it works!
1) Lamp now connected to com and nc. 
2) DC- on the board takes ground from external 12v supply but now also has connection back to RPi (pin 39)
3) In1 on relay still goes to pin 37.
vid at https://www.youtube.com/watch?v=jqYIoOvZkC8 and some images below - many thanks @AndrejWeber !

@theyosh do you mind me asking all kinds of questions here or would you prefer separate issues for specific questions?  e.g. i'm working on two Medium posts atm for this relay and another for webcam which i just set up. If / when i ask webcam hardware questions are they ok here or a new issue?
[image]
[image]
[image]

stuartabrown, 2019-01-19:
Hi,  I've found an odd thing when setting up environment to control heat lamp connected to a switch.  The logic works butt seems that the `Current` reading in the environment panel on the dashboard is slow to update / doesn't update
for e.g. the reading on the sensor may be 22.25 but the current temp shows as 20.4. see below. 
[image]

log shows that the correct temp is being read.  I'm assuming this is why my heat lamp wont switch off unless i change max temp manually.
[image]

stuartabrown, 2019-01-19:
new post! https://medium.com/@jinky32/connecting-a-12v-8-channel-relay-to-an-external-power-supply-and-raspberrypi-6fec119c112c.  @theyosh let me know if you want any modifications

theyosh, 2019-01-20:
> new post! https://medium.com/@jinky32/connecting-a-12v-8-channel-relay-to-an-external-power-supply-and-raspberrypi-6fec119c112c. @theyosh let me know if you want any modifications

Holds the amount of seconds that this environment part is powered on when the value is out of range: This is now only used for spraying. If this value is higher then zero, it will define the power on duration when switched on. For spraying I have a single spray time of 20 seconds. And then turn of. This way, you can trigger short power bursts. For heating and lighting it is not usable, and should enter zero to disable this behavior.

How much time must there be between two actions and after start up. Enter the amount of seconds in which this environment part can settle.: This is used for spraying, heating and cooling. This means that there must be at least X seconds between two actions. And this is only handy for dimmers. When the temperature is to low, the environment will dimm less and therefore more power to the heating system. This will heatup the temperature. But a the temp changing could take 2 minutes, and the environment will change the dimmer value every 15 seconds. So in the end, the heater will get to warm.

A second example with spraying is that after spraying the humidity will rise slowly. So I said wait 2 minutes before a second spray action. And mostly the humidity is within the valid range and the sprayer is not needed anymore.

I hope it is clear what I mean with settle time. Not sure if that is the right term.

theyosh, 2019-01-20:
Your temperature values are strange. I have no clue right now...

theyosh, 2019-02-09:
He, any updates? You should definitely look at 1Wire temperature sensors through GPIO. It is easy and very cheap. That way you have a lot of temperature measure points.

stuartabrown, 2019-02-09:
Cool! 
At the moment I'm waiting for new hardware to arrive from China (uv sensors and camera) and Romania (dimmer you recommended).
For temp sensors I'm thinking I will go with some of the the IR sensors mentioned in https://github.com/theyosh/TerrariumPI/issues/247 but I have to wait until I have some money!
At the moment I'm trying to learn basic python (eg  https://github.com/theyosh/TerrariumPI/issues/248#issuecomment-462033010) electrical stuff until my new things arrive.

stuartabrown, 2019-02-17:
Hi.  I'm writing my tutorial for the MPDMv4.1 dimmer and I'm seeing some strange behaviour that I think is user (my) error and I'm hoping someone can put me right!  I'm working on two scenarios 1) Ramping lights up/down based on timer (in the lighting section of environments) and 2) dimming a heat source based on the value of a sensor.
In both cases my PWM Switch is set up as below and in both cases I seem to have two rules coming into play, once reducing the voltage and the other increasing it 
[image]
**Lights in Environment**
Settings here are below
[image]
[image]

In the logs at times between the start and stop times in the day section (7.30 & 7.40) I see
```
2019-02-17 19:37:26,024 - INFO    - terrariumSwitch      - Power switch 'PWM Dimmer' at address '32' is done at value 90.0% in 15.0 seconds
2019-02-17 19:37:39,569 - INFO    - terrariumEnvironment - Environment light is turning on the alarm min powerswitches based on timer
2019-02-17 19:37:39,571 - INFO    - terrariumSwitch      - Changed power switch 'PWM Dimmer' of type 'pwm-dimmer' at address '32' from state '90.0' to state '100.0' (Forced:False)
2019-02-17 19:37:39,934 - INFO    - terrariumEnvironment - Environment light is turning off the alarm max powerswitches based on timer
2019-02-17 19:37:39,935 - INFO    - terrariumSwitch      - Changed power switch 'PWM Dimmer' of type 'pwm-dimmer' at address '32' from state '90.0' to state '80.0' (Forced:False)
2019-02-17 19:39:55,627 - INFO    - terrariumEnvironment - Environment light is turning on the alarm min powerswitches based on timer
2019-02-17 19:39:55,629 - INFO    - terrariumEnvironment - Environment light is turning off the alarm max powerswitches based on timer
2019-02-17 19:39:57,813 - INFO    - terrariumSwitch      - Power switch 'PWM Dimmer' at address '32' is done at value 100.0% in 15.0 seconds
```
You can see that something is turning the alarm `on` and then `off`
Looking at the logs after the stop time in the day section (7.40) the logs show the below:
```
2019-02-17 19:40:10,629 - INFO    - terrariumEnvironment - Environment light is turning off the alarm min powerswitches based on timer
2019-02-17 19:40:10,631 - INFO    - terrariumSwitch      - Changed power switch 'PWM Dimmer' of type 'pwm-dimmer' at address '32' from state '100.0' to state '90.0' (Forced:False)
2019-02-17 19:40:10,993 - INFO    - terrariumEnvironment - Environment light is turning off the alarm max powerswitches based on timer
2019-02-17 19:40:10,994 - INFO    - terrariumSwitch      - Changed power switch 'PWM Dimmer' of type 'pwm-dimmer' at address '32' from state '100.0' to state '90.0' (Forced:False)
```
There are still two activities but now they are both turning the alarm off.

**Temperature in Environment**
Settings below
[image]
[image]

In the logs you can see again there are two process going on, each working in opposite directions:

```
2019-02-17 19:52:10,857 - INFO    - terrariumEnvironment - Environment temperature is turning on the alarm min powerswitches based on sensor
2019-02-17 19:52:10,859 - INFO    - terrariumEnvironment - Environment temperature is turning off the alarm max powerswitches based on sensor
2019-02-17 19:52:14,332 - INFO    - terrariumSwitch      - Power switch 'PWM Dimmer' at address '32' is done at value 30.0% in 15.0 seconds
2019-02-17 19:52:24,394 - INFO    - terrariumSensor      - Updated owfs sensor 'Temperature Sensor' from 37.28C to 37.12C in 0.00011 seconds
2019-02-17 19:52:25,857 - INFO    - terrariumEnvironment - Environment temperature is turning on the alarm min powerswitches based on sensor
2019-02-17 19:52:25,861 - INFO    - terrariumSwitch      - Changed power switch 'PWM Dimmer' of type 'pwm-dimmer' at address '32' from state '30.0' to state '40.0' (Forced:False)
2019-02-17 19:52:26,470 - INFO    - terrariumEnvironment - Environment temperature is turning off the alarm max powerswitches based on sensor
2019-02-17 19:52:26,471 - INFO    - terrariumSwitch      - Changed power switch 'PWM Dimmer' of type 'pwm-dimmer' at address '32' from state '30.0' to state '20.0' (Forced:False)
2019-02-17 19:52:40,860 - INFO    - terrariumEnvironment - Environment temperature is turning on the alarm min powerswitches based on sensor
2019-02-17 19:52:40,862 - INFO    - terrariumEnvironment - Environment temperature is turning off the alarm max powerswitches based on sensor
2019-02-17 19:52:42,675 - INFO    - terrariumSwitch      - Power switch 'PWM Dimmer' at address '32' is done at value 40.0% in 15.0 seconds
2019-02-17 19:52:54,406 - INFO    - terrariumSensor      - Updated owfs sensor 'Temperature Sensor' from 37.12C to 37.12C in 0.00010 seconds
2019-02-17 19:52:55,863 - INFO    - terrariumEnvironment - Environment temperature is turning on the alarm min powerswitches based on sensor
2019-02-17 19:52:55,865 - INFO    - terrariumSwitch      - Changed power switch 'PWM Dimmer' of type 'pwm-dimmer' at address '32' from state '40.0' to state '50.0' (Forced:False)
2019-02-17 19:52:56,220 - INFO    - terrariumEnvironment - Environment temperature is turning off the alarm max powerswitches based on sensor
2019-02-17 19:52:56,221 - INFO    - terrariumSwitch      - Changed power switch 'PWM Dimmer' of type 'pwm-dimmer' at address '32' from state '40.0' to state '30.0' (Forced:False)
2019-02-17 19:53:10,865 - INFO    - terrariumEnvironment - Environment temperature is turning on the alarm min powerswitches based on sensor
2019-02-17 19:53:10,867 - INFO    - terrariumEnvironment - Environment temperature is turning off the alarm max powerswitches based on sensor
2019-02-17 19:53:10,870 - INFO    - terrariumEnvironment - Update done in 0.00821 seconds. Waiting for 14.99179 seconds for next update
2019-02-17 19:53:12,940 - INFO    - terrariumSwitch      - Power switch 'PWM Dimmer' at address '32' is done at value 50.0% in 15.0 seconds
```
Any advice here much appreciated!

theyosh, 2019-02-19:
Ok, this is a long story :P So we will do the lights first. Because that is discussed here: #237 Here is discussed how it is working now. It is not working like you would expect. So have fun with reading :D

You should use the dimmer in your case only for the low alarm. Never use a switch more than once.

stuartabrown, 2019-02-19:
thanks @theyosh , I'll try the settings suggested in https://github.com/theyosh/TerrariumPI/issues/237#issuecomment-453857454. I'm still not sure though what the `dimmer on` and `dimmer off` values are for.  How do they relate to the start / stop times if the lighting environment is set to timer?

BTW, new post is up for RPi camera tutorial.  Thanks for your help with that.  If you want any changes, let me know

stuartabrown, 2019-02-23:
@theyosh OK great. I have the light daytime ramp up and then ramp down working correctly.  The only issue is that once the dimmer has turned the light back down to 0% it then turns it on again.  Is this due to an incorrect `dimmer down duration`?  Should that contain the amount of minutes between day stop and day start or something?

Once I've got this sorted I'll move on to controlling temps (disabling lights first so switch isn't used twice)

stuartabrown, 2019-02-24:
OK so I can get temp control working reliably too :) If someone can help me understand the purpose of `dimmer on duration` and `dimmer off duration` then I can start writing up my tutorial.  And are these settings something to do with the light turning on against once it has reached 0% (as per above comment) when set as lighting environment?

AndrejWeber, 2019-02-24:
Do you have still the dimmer used twice in light environment?
You only need to set it in day. Like last pictures it will dimm on at night time again.

stuartabrown, 2019-02-24:
thanks @AndrejWeber. Settings are as per below
[image]
[image]
[image]

logs
```
2019-02-23 11:27:39,936 - INFO    - terrariumSwitch      - Changed power switch 'PWM Dimmer' of type 'pwm-dimmer' at address '32' from state '25.0' to state '10.0' (Forced:False)
2019-02-23 11:28:09,942 - INFO    - terrariumSwitch      - Changed power switch 'PWM Dimmer' of type 'pwm-dimmer' at address '32' from state '10.0' to state '0.0' (Forced:False)
2019-02-23 11:28:32,840 - INFO    - terrariumSwitch      - Power switch 'PWM Dimmer' at address '32' is done at value 0.0% in 20.0 seconds
```
That all works fine, but then when the lights diminished to 0, they were then turned on again.

stuartabrown, 2019-02-27:
OK I've been trying this again and I still can't get the light attached to the dimmer to stay off once it has dimmed to 0.  Once it hits zero the light comes back on after approx 15 seconds.  My assumption was that it should stay off from the after the dimming ends (set in the lights environment, day, `stop` field) and then start to ramp up when the time reached in the value set in the lights environment, day, `start` field is reached.  But this might be a misunderstanding on my part.

I've tried setting
- `timer on` and `timer off` in lights environment to 0 
- setting `dimmer on` and `dimmer off` duration to 0 in the switch settings 

but it makes no difference.  Is the only way to keep the light off once it has dimmed to zero to use the dimmer in partnership with a relay as @AndrejWeber does in https://github.com/theyosh/TerrariumPI/issues/178 ?

AndrejWeber, 2019-02-27:
Normally your dimmer should work without Relais. Because you can reach 0% light level. My dimmer can only reach 1%, so will never turn off completely. 

I wonder about your on and off duration in switch setting,
My settings are 5s, you have 600s.
I have set
Action duration: 20
Step: 1
On duration : 5
On percentage: 70
Off duration: 5
Off percentage: 0
For faster testing you can change step to 15
I do fade up and down very slow and also only up to 70%, because the t5 bulb with 54W is to much light.

stuartabrown, 2019-02-27:
@AndrejWeber many thanks for your help with this.  So I have used your settings (with step as 15 rather than 1)
[image]

[image]

and here is a video https://youtu.be/3epRDUmCEBY?t=404 showing the effect that I get once bulb has gone to 0.

AndrejWeber, 2019-02-27:
Is it a flickering, I’m not shure...
Then I have a solution

stuartabrown, 2019-02-27:
@AndrejWeber do you mean the state of the lamp in the last part of the video?

AndrejWeber, 2019-02-27:
Yes, it looks not like normal “on”

stuartabrown, 2019-02-27:
any idea what it might be?  Something to do with TerrariumPi settings?  The dimmer? the Bulb?

AndrejWeber, 2019-02-27:
Like it is written in the reviews of the dimmer, you should set a lower maxdim value. 
It is in the terrariumSwitch.py
Line 585
In the Moment it is set to a PWM Value of 895. 
Try some lower, for example 870
With a multimeter, you can measure from ground to pwm pin, you should measure something around 2.9V (DC)
It should not be more, then the Electronics get confused.

stuartabrown, 2019-02-27:
@AndrejWeber many thanks for that, gives me something to work on :)  I don't have a multimeter with me at the moment but will have at the weekend.  In case it is of use to you it looks like the default value has changed in `terrariumSwitch.py`
https://github.com/theyosh/TerrariumPI/blob/master/terrariumSwitch.py#L634

AndrejWeber, 2019-02-27:
you are right!
you shold try also without multimeter.
the developer says, the voltage should be about 2.85 to 2.90 V
so if we know, the raspi has 3.3V at 100% PWM ( @theyosh just invert this to 0% dim value )
we can do some mathwork.

maxdim = 2.85 / 3.3 * 1000
maxdim = 863

maxdim = 2.9 / 3.3 * 1000
maxdim = 878

try to use 870

stuartabrown, 2019-02-27:
@AndrejWeber I love you!  A value of 870 works perfectly!

BTW it's cool that you are dimming your T5s.  I have Arcadia D5 Dragon Lamp and am currently running them on an on/off timer; I didn't realise they could be dimmed.  I have daylight LEDs that mimic sunrise / sunset (https://zoomonster.com/epages/Shop_Econlux.sf/de_DE/?ObjectPath=/Shops/Shop_Econlux/Products/00011500) so I will eventually get the T5 to ramp up / down to the same schedule

AndrejWeber, 2019-02-27:
they are perfect to be dimmed, but only with a dimmable ecg!
[https://www.osram.com/ds/ecat/QUICKTRONIC%20INTELLIGENT%20DIM%20T5-ECG%20for%20FL%20and%20CFL,%20dimmable%20110%20V-ECG%20for%20fluorescent%20and%20compact%20fluorescent%20lamps-Electronic%20Control%20Gears%20for%20Lamps-Digital%20Systems-/com/en/GPS01_1027982/PP_EUROPE_Europe_eCat/ZMP_56666/](url)
There are also some dali dimmer, but then you also need a i2c to dali interface and anyone, who implement a new hardware. ;D
Dali-ecg also do not need any relais to switch off the Lamp.

theyosh, 2019-03-02:
So, because I expected it to work on 3.3Volts, I used a to high MAXDIM value? I will change it then a bit lower as mentioned here now.

stuartabrown, 2019-03-07:
Phew - ok a few new guides at https://medium.com/@jinky32, including the PWM switch.  As every feel free to request changes or additions.  Thanks for all the help @theyosh and @AndrejWeber :)

vacuumboots, 2019-03-26:
I just wanted to chime in and mention that it has been great to follow along with this open ticket / the medium articles! . 
Thank you so much for all of the work put into this project @theyosh. @stuartabrown it's been great to follow along with your progress in near real-time.

stuartabrown, 2019-03-29:
Thanks @vacuumboots .  I've been busy with some other stuff but back on it again now!

@theyosh re the UV sensors and specifically your comment https://github.com/theyosh/TerrariumPI/issues/90#issuecomment-405478940 about only supporting one because they have the same I2C address, would it be possible to use an I2C multiplexor e.g. https://learn.adafruit.com/adafruit-tca9548a-1-to-8-i2c-multiplexer-breakout/overview?

theyosh, 2019-03-30:
I would rather follow: https://www.instructables.com/id/Raspberry-PI-Multiple-I2c-Devices/ That will add more I2C buses and there fore you can have more of the same sensor. Every bus 1 sensor.
This is more stable than using a multiplexer board.

Also, using a multiplex board is harder to implement. Currently, multiple I2C buses is supported.

[5 more replies]

closed 2020-02-07 by theyosh