theyosh/TerrariumPI #231: Advanced scheduling module

jornobe, 2018-12-22:
This is a request to enhance the timer settings of switches. 

Currently it is possible to set just 2 values, amount of time to turn on a switch and the amount of time to turn it off again.

Would be nice to have a separate timer function/module that can be used to set different start and stop times (hh:mm:ss) for multiple switches. 

Currently I use separate timers and smart plugs to run things like aquarium filter, automated morio worm feeder, fish feeder, aquarium heater, Nest camera’s, rain system. Would be great to have all of these timer settings and triggers in 1 place. 

Especially the ability to schedule actions for just seconds instead of minutes would make this a great feature.

theyosh, 2018-12-23:
Hi, this will be possible somewhere in the future. As time scheduling is a bit hard to make with all the possibilities there will be. But this will be implemented with the option to support multiple environments.

Because then you can create your own triggers like the spray system in the environment part. There you can set all kind of timer settings. That will then be available for all types of sensors and switches.

Your heater should work with a sensor in the environment settings so that it does not need a timer on the switch. As I think that is the right way. But, look at the sprayer settings in the environment part and that will be available for all kind of setups.

jornobe, 2018-12-23:
Understood. Thanks for the feedback. Would it be possible to already have the timer for a switch trigger in seconds rather than minutes?

Currently the ceramic bulbs are operated using the temperature environment and sensor. Therefore the pool heater is on a separate timer. Thinking about running a second Pi to double the amount of environment settings for now.

theyosh, 2018-12-23:
Yes, your 'issue' with heating is exactly the reason why the new environment system should be more flexible. And the new system will allow multiple heating ways in 1 environment.

For now, the timer is only checked once every 30 seconds. So I have changed the code a bit, that it will accept 0.5 minutes (30 seconds) as input. But it will still only check once every 30 seconds.

This means a value of 0.1 = 6 seconds, could run up to 30 seconds due to the timer interval.

jornobe, 2018-12-23:
Great!

Benjin, 2018-12-24:
What's the limiting factor on how often the timer checks?  Just that it's a more involved change than the changes in 542b2cf?

Due to the too-small size of my fresh water reservoir, I'd like to be able to set the switch timing on my misters low enough so it won't run out when I go away for vacation.  Granularity of 15 seconds would be ideal for me, personally, as 30 seconds still means that I'm artificially increasing the period to make sure the water lasts the entire time.

theyosh, 2018-12-28:
For both... could you look at: #210 :) ?

theyosh, 2021-09-01:
Hi,

as version 4 is released, and there is now the option to have multiple areas of the same type (like temperature) in 1 enclosure. So you can now fine tune every relay using areas.

And it is possible to enter decimal values in the on, off, power on and settle timeouts. The smallest unit is still 0.1 which is 6 seconds.

I hope this will make it a bit easier to work with.

theyosh, 2021-10-09:
@jornobe or @Benjin did you had any change to test this? I think it is possible to have a better timing setup with TP4. And therefore I would like to close this issue.

Could you test and confirm you can now make the time schedules you want?

theyosh, 2021-11-17:
Can we close this issue? Is it working for everybody?

Benjin, 2021-11-21:
Okay, I've _finally_ had the time to try this out!  I have no idea if I've configured this correctly (I'm finding the combination of begin/end time, on/off duration, and relay on/off delay pretty confusing; perhaps I can help with the descriptions?)

With this configuration, I was expecting that the misters would toggle between off and on every 6 seconds (0.1 minutes), but when I timed it, it was actually toggling every 30 seconds.  I'm running this on a Raspberry Pi 3 B, if that makes a difference.

[image]

theyosh, 2021-12-01:
You should not use a `light` type area for misters. Use the `humidity` type. That gives more options. Does the descriptions at https://github.com/theyosh/TerrariumPI/issues/629#issuecomment-984059376 help a bit?

Basically a timer works with an operating window which is defined with the begin and end time. Lets say you create an operating window of 10 hours. 

In that 10 hours, you can say the relay should be on for `On duration` and off for `Off duration`. In the linked example, 15 min on and then 45 min off.

Then when you want you sprayer to work for 30 seconds, use that value in `Power on time` value. That means, during the 15 minutes `On duration` the relay will toggle on, and off after 30 seconds. And this will be repeated for 15 minutes. If you set the `Settle time`, the relays wait with toggling (on or off) for that value. So lets say 2 minutes. During this period, the humidity will rise, and sensors can pick up the new values. And when there is time left `On duration` - (`Power on time` + `Settle time`) > 0, it will repeat it again.

I hope this makes it a bit clear. And yes, documentation is lacking on this... :(

theyosh, 2021-12-22:
Any updates on this... In the meantime I have updated a lot of documentation again: https://theyosh.github.io/TerrariumPI/setup/#areas

Benjin, 2021-12-22:
I’ve been out of country all month, but I’ll check once I’m home after the holidays.

Good luck, be nice, and have fun,
Benjin Dubishar

theyosh, 2021-12-22:
No problem, as long you where not waiting on me :)

closed 2022-02-13 by theyosh