theyosh/TerrariumPI #232: more Display Chipsets - better OLED Driver

AndrejWeber, 2018-12-23:
Hey @theyosh,

I have bought a OLED-display. Of course, it was not supported by the driver, you are using :( 
Your code only allows to use SSD1306 chipset, but there are much more Displays, that do not work with Adafruit_SSD1306 driver. I have the newer and cheaper SSH1106.
So I found a driver, that let us use much more chipsets and resolutions and even color-displays:
[https://luma-oled.readthedocs.io/en/latest/intro.html](url)
There are some incompatible parts in your code like     self.__screen.begin()   , but easy to fix. ( I was able to fix by try and error! :D )
I do not know, how it is about Copyrights, so please have a look at your own.
BTW, where I can change, what the display will show me? so, where is rawdata generated?
I will try to change it by my own, trying to get more usefull stuff set to display.
thank you

Merry christmas!

theyosh, 2018-12-28:
Cool! I think that this module is indeed better. It supports more devices and I do not to have all kind of different modules. But it will be low on my priority list. As I do not use it myself actively.

AndrejWeber, 2018-12-28:
No Problem, I have done it work for me, so I can wait. i have to look, if I can update to new version.

theyosh, 2019-01-13:
Hi,

I have created new code for those Oleds. Yours should now also be supported. No colors at this time.

The code is in the `oled` branch. So if you can check that out and test it, that would be nice.

Rerun the installer before restarting TerrariumPI. It needs new python and os packages.

AndrejWeber, 2019-01-13:
Great job👍
I can set up my display, and it shows the starting picture. 
It also shows changes in switch and other changes, I can set. But now it doesn’t show the values of sensors, when no switch-changes appears. It will stay by the picture

theyosh, 2019-01-13:
Euh, it should show at least CPU stats after max 1 minute. Are there any errors in your screen session? And did you enable the title setting for the display?

AndrejWeber, 2019-01-13:
I have also expected this CPU and other stuff.
I had this values also by my own changes.
But I changed the original class terrariumOLED to start the driver working. 
You have changed a lot, so I am not able to compare it. My knowledge of Python is much to low to understand that changes. 

I have deactivated all messages for now, so only the time, cpu, ram, temperature, and humidity should be there. 
After about 1h there is only the picture on the screen.

AndrejWeber, 2019-01-13:
`[display]
address = 3c,1
resolution = 128x64
title = true
hardwaretype = SH1106
`
no changes, only the picture.

theyosh, 2019-01-13:
Remove the resolution line and restart. The resolution should not be there.... If you remove it, and restart, it should work better. In the mean time, I did found some small issues. So also a git pull will fix some small issues.

theyosh, 2019-01-13:
Hmm, not sure if the resolution part is really the problem. But the new code will output some raw debug in the screen session. Does it show there some CPU stats etc..?

AndrejWeber, 2019-01-13:
yes, now it shows everything,
even befor remooving resolution.
I love the debugging screen with new display stats.

theyosh, 2019-01-13:
The contents for the display are only/mostly warnings and errors. So it can for now only show messages that are available in the notification settings. So, not sensor temperatures for example. By hovering over the images on the right side next to the messages, you can specify the way of notification. The icons should match the notification part like display or twitter.

The idea was/is that the notifications will alert you when there are errors. And a display is one way to do it. But the display is not added with the option to make some kind of overview screen.

The debug that is now added will be removed. In order to see the status, you should really use the web interface. 

In order to test the display and to show something at least, I added code that will update the display every 30 seconds with the latest CPU stats and sensor averages. This is not editable.

AndrejWeber, 2019-01-13:
This test-screen is much more usable than you expected  ;)
for shure, it is always possible to look in the browser, I do, but my son does not, and he like to have a display with shown temperature and humidity, like the normal offline-sensors do.
I have to think about another way to display him the interesting values on a screen.

thank you for now for adding this screen.

theyosh, 2019-01-13:
Ah yes, I see. But that is also my problem with creating some kind of display editor. The screen is small and I need to make it a general system. That is a bit hard.

So, for you, I would suggest buy a second pi and add a screen to that. And use own simple code that read out the TerrariumPI with JSON. That way you have all the freedom to create your own screens.

As a somewhat compromise I do show the average sensor values on the display. That is enough for me know vs the amount of work that has to be done.

Then I have a question for you. Could you look at: https://github.com/theyosh/TerrariumPI/issues/210
I have a 'policy' now that every feature request require to add some pictures of your setup :)

AndrejWeber, 2019-01-13:
I will post some pics next month. The geckos are in winter rest at cellar, so the terrarium is very empty, no rocks, no wetbox.., no geckos 😅

theyosh, 2019-01-14:
Ok, then I think we close this issue? And I will wait for the photos

closed 2019-01-22 by theyosh