theyosh/TerrariumPI #237: Unreliable timing for DC Dimmer

Raerion, 2019-01-07:
Hey @theyosh 
as this is my first post here I would like to thank you for sharing this awesome project!
I just started using TerrariumPI a few days ago and was surprised with how easy it was to set up from scratch. Current hardware is 2x DS18B20 via 1W, EG-PM2, DC Dimmer via GPIO and planning to add MiFlora and Raspberry Camera.
The normal power switches are working just as aspected, however I have some issues with the DC Dimmer. 

**What I want to do is the following:**
Switch on 230 V power for the LED lights on a configured time and start fading up the LEDs for a configured time (-> sunrise). At the end of the day fade down the LEDs for a configured time (-> sunset) and then switch off the 230 V power. 
I have configured two switches in Terrarium PI with the following settings:
[image]
I'm using the timer module instead of the environment because I need two different stop times (power switch has to wait for the fading down to finish). 
As you can see I would like to fade up slowly over 1 hour and down over 2 hours. However I don't really understand the purpose of the two fields "Dimmer action duration" and "Dimmer environment step". Is this configuration faulty?

**The main issue is:**
- The dimmer does not start fandig up reliably. It is configured to start at 11:30 but in fact did start at 13:06 today. Yesterday it didn't start on time aswell but did immediately start fading up after I restarted TerrariumPI within the screen session (12:04), see .csv as reference. [csv_data.zip](https://github.com/theyosh/TerrariumPI/files/2734173/csv_data.zip)

**Other side issues (minor):**
- Commands to the dimmer-switch are repeated every time terrariumSwitch is called even if there is no change of value. This creates a lot of unnecessary data which is visible in the log and the data export (see .csv). The binary switch only logs data when changes occur (as expected).
- Fading down starts after the end time which imho is not very intuitive. It makes changing of times for my use case a bit complicated and prevents me from using the environment and changing sunset/sunrise time according to weather data, which would be quite cool. 
I would have expected the dimmer timer it to behave like this:
`Start fading time = Timer stop time - Dimmer off duration`
Of course I understand if the current behavior is created for different use cases and may be like this by design.

Hope you could help me with my issue! 
If helpful, here is todays logfile. [terrariumpi_log.zip](https://github.com/theyosh/TerrariumPI/files/2734265/terrariumpi_log.zip)

All the best
Jan

AndrejWeber, 2019-01-07:
Hi Jan,
Try to change to 1% and 360 action duration.
I had also problems by using less than 1%. 
In german: keine Kommazahlen
After changing to longer duration and 1% it works fine. 
I use the dimmer in switches setup and switch on/off by environment (timer).

Raerion, 2019-01-08:
Hi Andrej,
Thanks for your suggestion! 
I tried those settings this morning and restarted TerrariumPI however nothing changed with my issues. 
The switch still logs the value with each call and it didn't start fading up in time.
I waited for half an hour and restarted, then the dimmer immediately started to fade up.

My next try would be to use the timer from switch settings only for the 230 V power and the timer from the Environment part for the dimmer.

Raerion, 2019-01-09:
Thanks again.
Using the environment works much better: The commands are not repeated anymore and the fading up starts on time! 
However the times are still a little bit strange:
With 36 s duration and 100 steps (increment of 1) I would expect 36s*100=3600 s fading time, but actually it takes 75 minutes. There are two steps in the log, one wait of ~ 9 s and the configured 36 s wait for the next step ((36+9)*100=4500).
After changing the duration to 36 s - 9 s = 27 s the fading down takes 50 minutes with additinal 3 s. 

I don't quite understand why there are this additional times but I think I'll be able to tweak the numbers until it works perfectly. ;)

AndrejWeber, 2019-01-09:
Hey Jan,
Try to set up timer on duration to about 5 seconds. 
If I have understand it right, that will calculate the real increments. 
So at 25s by 1% you will have a more smooth dimming with 5 x 5s 0.2% Increments. 
I cannot check it now, because of winter rest. 
I have seen something in logging that I have not figured out completely. 
@theyosh, please tell us, if I am wrong.

theyosh, 2019-01-11:
Yesh, it is very confusing... I do know it, and this is part of some legacy logic.

For now to get a bit of a sunrise and set experience, use a dimmer in the light environment. I use the weather data for the on and off time. But you can also select your own times when the lights should go on and off.

Next, you have to setup the dimmer switch with the following fields:
- Dimmer action duration
- Dimmer environment step
- Dimmer on percentage
- Dimmer off percentage

To make it easy, use 100% for Dimmer on percentage and 0% for Dimmer off percentage

Then the field `Dimmer action duration` holds the value in seconds how long a single change will take. 
And the field `Dimmer environment step` holds the amount it should increase / decrease per single change.

So it should work like 25 sec on duration with 1 increment set from 0 to 100: it should take 25 * 100 = 2500 seconds = +/- 41 minutes.

At least, this how it should work.  Make sure you have disabled the timer functionality of the switch itself. I do also use it in this way. But I think that I also have some timing issues now that I take a closer look to my graphs. For me it was not that a big issue. But as I see more people have it, I will investigate it.

theyosh, 2019-01-11:
And for now, please use round number and not decimals. Not sure if that will work

theyosh, 2019-01-12:
Ah crap... Now I see it clearly.. The above steps are done once every 30 seconds. Due to the fact that a dimmer can also used for heating. With heating, you would like the dimmer to increase and decrease small steps in order to get a steady temperature. Therefore the code is now made that every 30 second the dimmer will go up or down for the specified values at the dimmer switch.

But when your single duration action is like 45 sec, it will end up in 60 sec. Because when the dimmer is running, it will not accept new values. So the dimmer will start for 45 sec. 30 sec later it get another request to change the dimmer. Due to already running and keep running for 15 sec, nothing will change. And the dimmer will then wait for 15 sec for the new 30 seconds round.

I hope this explains the time offset based on the values. This is made for better heating with dimmers, and I thought that a bit more or less delay in lights is not that a big of a deal.

Raerion, 2019-01-13:
Thanks for the explanation! Now that I understand whats happening, I have tweaked the numbers a bit and I'm happy witht he current setup.
I have set the action duration to 30 seconds which results in 30+15=45 seconds for each update.
Since I want the whole sunrise/sunset to take 90 minutes I can calculate the environment step as:
90*60 = 5400 seconds
100 * 45 / 5400 = 0,83333333...
Setting the step to 0.8333 is precise enough and actually there seems to be no issue with using decimal numbers. Fading up and down works perfectly as it should! =)

From my point of view this doesn't have to be changed but could be documented in the help section in the switch settings.

theyosh, 2019-01-14:
Yesh the magic word.. 'documentation' :P

The 'problem' is that the lights are handled the same as a heating or cooling device, which will go up and down every 30 seconds based on sensor data. For lighting this should not happen.

Therefore, with other feature requests around timing and multiple environments, I will change this in such a way, that when dimmer are used for lighting, you can just enter the duration for on and off. This will add some more logic and code, but makes it that more sense in how it works.

But cool that it is now working as wanted.

theyosh, 2019-02-09:
This dimming logic has to change to make it more easy to understand

theyosh, 2021-04-10:
[image]

Yesh!!! Party time... Finally after a few years.... it is here!! If you have time, could you have time to test this: #539