theyosh/TerrariumPI #239: Manual Mode for switches (override)

Raerion, 2019-01-14:
In the last days while setting up I missed a very basic (but maybe not so simple to implement) feature and would like to propose it for a future version:
I think it could be helpful to set a switch to manual operation. This manual mode would override any timer / environment action so you could set the switch to a fixed value (on, off, x percent) for as long as you desire. When you release the manual mode the normal program should take over again.

One example when I would use the feature:
If you reboot the Pi (update, new hardware, maintenance..) while a dimmed light is active the dimmer resets and starts again once TerrariumPI has started again. One could say that maybe the switch should "remember" that it has already faded up but as I'm actively working on the system I would rather just set the light immediately to a desired value.

Other examples:
- Testing purposes
- Turn on light temporary (e.g. in the night)
- Disable automatic actions temporarily (e.g. no heating or no sprayer while cleaning)
- Compensate malfunction, e.g. when my dimmer with switch-timer instead of environment-timer didn't start on time (software bug) or if a sensor that triggers the switch is faulty

Do you think this would be useful?

jornobe, 2019-01-16:
The “remember” switch status while restarting and being able to do a manual override on a switch are things I would also like and use frequently.

theyosh, 2019-01-17:
Hi,

I have to see what is possible. Not everything is that easy. Also, you are close with #204 .

- Setting to state before reboot can probably be done. Have to look into it
- Turn on light during the night is hard! It goes against all logic and is in my opinion the same as #204 . So that have to wait
- Temporary disable is an option. My idea is that you have to toggle a switch or option to disable and enable again. So that can be made
- Compensation is not possible if it happening due to a bug. Because there is a bug. Faulty sensors are covered already: https://github.com/theyosh/TerrariumPI/issues/206#issuecomment-440409698

Keep in mind that I cannot take care about broken hardware. Mostly broken hardware will not work and disable the sensor and the connected switches. So keeping a eye on your hardware and checking the web interface is still manual work.

theyosh, 2019-01-17:
But, the remember state will probably only switch back if the restart is within 30 minutes or so. Because when you have the software shutdown for more then 1 hour, it could be that the switch should be switched off because of timer logic. Then you get that the switch goes on during startup where it should be out. Therefore the remember code will only 'remember' for 30 - 60 minutes and not longer

Raerion, 2019-01-18:
Thanks for considering this wish!
Just for clarification, my idea was not a hardware button like #204 but a software option in the webinterface. Currently you can toggle a switch by clicking on it and it will keep that state as long as there is no environment/timer configured which will overwrite the state.
What I was thinking about is basically an option to toggle such switches that are "connected" to an environment/timer. There could be something like an popup if you click on the switch with a message  "Do you want to manually override this switch? [Yes] [No]" and if confirmed the manual state and value are saved. The script that handles the output would have to make a check against that manual state (something like this `if manual_state(switch_id)=true output(manual_value(switch_id)) else [current output logic]`).
Actual implementation would probably be way more complex of course. ;)

Remembering the state of a switch when TerrariumPI restarts is a different thing which would be nice to have, too. In my opinion this would only be applicable if you manually restart the software from the screen session or if you do a reboot via ssh, not for example if you make a hard reset by cutting the power. From my point of view a time check (like the 30 minutes you mentioned) would be fine as its the 30 seconds downtime when you just have to restart for any reason which should be worked around. At the moment in this case all switches stay on while the software is down and are reset to zero when the software restarts only to be turned on again 15 seconds later.

theyosh, 2019-01-28:
Just made an update where you can put the switch in 'manual overwrite'. Then you can toggle the power switch as normal. The environment will then give a warning in the logging that the power switch is in manual mode and ignore it. You can find the option on the 'settings' menu (wrench icon).

So, do a `git pull`, restart and **clear browser cache** to get the new option.

Could you test this and see if it is workable.

5point0, 2019-01-30:
I have updated and see this manual option. Will test and revert back.

Great feature add by the way!

AndrejWeber, 2019-01-30:
hey @theyosh 
great new function!
now you have added a manual mode, is it possible to add a new environment, called "manual mode" or even more than one for different uses? 
there we can add a push button or switch to enable manual mode for actions we can coose.
like now sensor mode (toggled by door). also someone wanted here to disable raining, when doors are opened.
my Idea was to switch on my dimmed lamp in the night (so a gpio-relais) for some minutes. 
some one will start raining manually.
An Error in the logfile is not bad by the way. so the display will show us, that we have enabled the manual mode.

theyosh, 2019-01-30:
@AndrejWeber thanks.

I am thinking of adding actuators as a new menu option. Where you can add push buttons #204 , but also door sensors and maybe IR motion sensors. Then you should add some kind of action to an actuator like powering light, stopping the sprayer etc... but that is currently on the wish list.

But at the moment it is already possible to disable the spray system when the door is open. Only when the sprayer is running, it will not stop when you open the door. This will also be looked at to see if this can be updated so that the sprayer will stop when the door is opened.

So this manual mode is the first part that is needed to make the actuators work...

theyosh, 2019-02-10:
Updated the code so that it will restart with the previous power switch settings. I have not tested it that much, so it could be that it is not working perfectly.

As I do not restart that often, and not to stress my animals, I will not do that much testing for this. Happy for some feedback

theyosh, 2019-02-16:
@rrgrow I see you scenario. But that will maybe done in a later time. This issue is just about manual actions.

Raerion, 2019-02-18:
Hey @theyosh, 
sorry for taking so long to test the feature and responding here - I didn't have much spare time and decided it would be best to not touch the running system for a while. ;)

Now I have finally updated to the current release and I do like the new manual mode! It's definitely working as expected while TerrariumPI is running and the switches keep their current state when the software is restarted from the screen session. 
I also tested rebooting the RPi in manual mode:
- GPIO outputs are reset by the RPi (I guess this can't be changed as the power state is probably reset after shutdown). When TerrariumPI starts light is directly set to the previous percentage (=> perfect! without the new mode it would dim up for an hour)
- Power switches with EG-PM2 stay at the prior value during shutdown/boot since its seperate hardware however when TerrariumPI starts it resets all EG-PM2 switches to off-state before they switch to the correct value (not as expected, but also not a big issue since reboots don't happen very often and could be scheduled for night time when everything is off).

Thanks a lot for the fast implementation! #210 has been updated. ;)

theyosh, 2019-07-07:
Ok, I found a small graphing bug regarding this.. but other then that, I think this is the best I can do, and I think this can be closed

theyosh, 2020-01-28:
Closing due to no updates and looks working fine....

closed 2020-01-28 by theyosh