theyosh/TerrariumPI #240: After the Initial Install, I can't access the web interface NOOB

ese23, 2019-01-17:
Hello,
first i was format card to fat32, then "mount" image of LITE rasbian...
than enable SSH, install terrarium... boot terrarium, changing password for pi user...
connect via putty via ssh.
TerrariumPI say server running on ip:xxxxxx
But no luck access it (port:8090, clear cache)

copy from ssh console
pi@raspberrypi:~/terrariumpi $ sudo ./start.sh
2019-01-17 16:07:01,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server running as user 'root' at location: http://192.168.1.58:8090 ...

And last NOObish question, what mean screen -r and these control + c (for restarts)  im trying in ssh putty session or with keyboard directly on PI. No luck :-)

PS i switch user from pi to root... because i read some same issues... im using rapsberryPI 1 +A

Thank you
Best regards

theyosh, 2019-01-17:
Hi,
> copy from ssh console
> pi@raspberrypi:~/terrariumpi $ sudo ./start.sh
> 2019-01-17 16:07:01,000 - INFO - terrariumWrapper - Starting TerrariumPI server running as user 'root' at location: http://192.168.1.58:8090 ...

Indicate that you have installed it with user root. That is not safe. Not sure if that will case a startup issue.

the command `screen -r` will give you the console of my software. So after `sudo ./start.sh` you should be able to see what is going on when entering `screen -r`. But in your case, as your software is running as user root, you need `sudo screen -r`

Then you could use crtl+c to restart my software.

But I have doubts if a Pi1 will work. As it is only tested on newer ones. So I would suggest to get a new one (latest model) and try again. That is why you it is not working with user `pi`

Then make sure you do not install the software as root. It should be installed as normal user. So you have probably run the installer as root user. So follow the installation instructions careful.

And as last, in order to use this software, some Linux knowledge is needed. And when the software is running, you need to add hardware which can be done in multiple ways. So that will also involve some technical knowledge.

stuartabrown, 2019-01-17:
@theyosh this is funny timing actually. I have just done a git pull into my RPi but the software is not coming back up.  screen -r for me is
`2019-01-17 20:22:23,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.1.35:8090 ...
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 17, in <module>
    import terrariumLogging
  File "/home/pi/TerrariumPI/terrariumLogging.py", line 17, in <module>
    from terrariumNotification import terrariumNotification
  File "/home/pi/TerrariumPI/terrariumNotification.py", line 32, in <module>
    from terrariumDisplay import terrariumDisplay, terrariumDisplaySourceException
  File "/home/pi/TerrariumPI/terrariumDisplay.py", line 30, in <module>
    from luma.core.interface.serial import i2c
ImportError: No module named luma.core.interface.serial
2019-01-17 20:22:25,000 - INFO    - terrariumWrapper     - Restart counter: 1/5
2019-01-17 20:22:25,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 2 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3 2 1 restart!
2019-01-17 20:22:36,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.1.35:8090 ...
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 17, in <module>
    import terrariumLogging
  File "/home/pi/TerrariumPI/terrariumLogging.py", line 17, in <module>
    from terrariumNotification import terrariumNotification
  File "/home/pi/TerrariumPI/terrariumNotification.py", line 32, in <module>
    from terrariumDisplay import terrariumDisplay, terrariumDisplaySourceException
  File "/home/pi/TerrariumPI/terrariumDisplay.py", line 30, in <module>
    from luma.core.interface.serial import i2c
ImportError: No module named luma.core.interface.serial`

stuartabrown, 2019-01-17:
yes, so if I comment out line 30 (actually I did lines 30-33)of https://github.com/theyosh/TerrariumPI/blob/master/terrariumDisplay.py TerrariumPI reboots fine for me

theyosh, 2019-01-17:
Ah yesh, crap... uh you have done a git pull after a new release. So the installer has to rerun. That will add the missing python modules.

So if you now just do again `sudo ./install.sh` and restart or reboot and it should work again

stuartabrown, 2019-01-17:
@theyosh awesome, works for me :)

res2cpu, 2019-01-19:
Good Evening.

I'm in the process of setting up 2 of these and i happen to have a raspberry-pi b (V1) and multiple Pi zero W's along with a Pi 3B+

I first tried on my Zero W, the install ended up taking about 2-3 hours and then failed (i tried twice) and the Pi (V1) also failed to install correctly.

Lastly i used the Pi 3B+, faultless install first time. So i think it might be the hardware.

Now i just need to get the 2 viv's setup for the new members of the family we are collecting next sunday. (i'll throw up some photo's to the thread once done).

theyosh, 2019-01-20:
I have not tested Raspberry 1. And I think there GPIO is different and I expect therefore mutliple issues and I call that 'not supported' . The Pi Zero is also a bit strange. It should/could work, but here you need to figure it out yourself. As I do not have one. Other people said they got it working, but it is never clear why.

So I can only confirm Raspberry 2 and 3 as working correctly. I have used them both.

theyosh, 2019-01-22:
But this issue can be closed? Conclusion: Raspberry Pi 1 is not supported.

ese23, 2019-01-24:
One try pls. Because not enought space :-D Soo tomorow i will install it on bigger card.
And new version of RP3 is on the way.

Old card is 2g. Soo i dont know if you have mininum reguiured, but for all noobs. Use bigger card :-)

theyosh, 2019-01-24:
I guess you also did a desktop install. As far as I know that will probably not work. I only support 'Raspbian Stretch Lite' https://downloads.raspberrypi.org/raspbian_lite_latest So make sure you use that image.

And that should fit on 2GB, but then installing all the needed software will not fit. So 4GB is indeed the minimum size.

theyosh, 2019-01-26:
Euh, this is still RPI1? That is not supported. And the error memory error is something that looks like a full disc.

closed 2019-01-28 by theyosh

theyosh, 2019-01-30:
@ese23 He, I almost forgot to ask. As an issue is fixed there is a small 'price to pay' :) Look at #210