theyosh/TerrariumPI #241: Humidity environment leaves Energenie switch on

jornobe, 2019-01-18:
Had an issue today with my rain system. It is controlled by the humidity environment settings and the switch was activated at some point and not switched off after the set 15sec...

In the log I can see this:
- humidity level drops slowly during every refresh cycle of the sensor
- at some point the 60% low limit is hit
- the environment settings activate the Energenie usb outlet for the rain system
- the humidity value raises again with every refresh cycle of the sensor
- the switch was left active and no indication of it being switched off (not in the log, not in the mail notifications)

Any idea why this can happen?

jornobe, 2019-01-19:
After my lights also failed to switch off I checked the console (screen -r) and there were no entries for the last few hours. Must have clocked up somehow.

After a complete restart things settled down again.

jornobe, 2019-01-21:
Problem was caused by a faulty SD card causing the engine to crash.

closed 2019-01-21 by jornobe

theyosh, 2019-01-21:
Ah crappy SD cards.... I have made some changes to write all temporary data to memory so that it will not wear down the SD card.

Also, I have SD cards that live 3 months, other 3 years ..... ? Strange..